MichaIng/DietPi #1160: DietPi-Software | minio

Fourdee, 2017-09-26:
> https://github.com/Fourdee/DietPi/pull/1152

Fourdee, 2017-09-26:
x86_64:
```
root@DietPi:~# systemctl status minio.service
● minio.service - Minio
   Loaded: loaded (/etc/systemd/system/minio.service; disabled; vendor preset: e
nabled)
   Active: failed (Result: exit-code) since Tue 2017-09-26 17:38:04 B
ST; 45s ago
     Docs: https://docs.minio.io
  Process: 4267 ExecStart=/usr/local/bin/minio server $MINIO_OPTS $MINIO_VOLUMES
 (code=exited, status=203/EXEC)
  Process: 4264 ExecStartPre=/bin/bash -c [ -n "${MINIO_VOLUMES}" ] || echo "Var
iable MINIO_VOLUMES not set in /etc/default/minio" (code=exited, status=0/SUCCES
S)
 Main PID: 4267 (code=exited, status=203/EXEC)

Sep 26 17:38:04 DietPi systemd[1]: Starting Minio...
Sep 26 17:38:04 DietPi systemd[1]: Started Minio.
Sep 26 17:38:04 DietPi systemd[4267]: minio.service: Failed at step EXEC
 spawning /usr/local/bin/minio: Exec format error
```

Looks like incorrect binary for arch.

techdabbler, 2017-09-26:
I'll have to check that out. Admittedly I did all most testing on ARM.  Worst case it just gets disabled for x86_64.

techdabbler, 2017-09-26:
are you testing on x64? I downloaded the x86 thinking it would run on both.  I bet I just need to break out the executables. I'll do that when I get home tonight.

Fourdee, 2017-09-26:
@techdabbler 

> are you testing on x64?

Yep, all our non-ARM images are x86_64 only.

> I bet I just need to break out the executables. I'll do that when I get home tonight.

Done:
https://github.com/Fourdee/DietPi/commit/efbcc824a708466dd1259bc59867d10df4f6c238

Fourdee, 2017-09-26:
And corrected HW_ARCH index for x86_64:
https://github.com/Fourdee/DietPi/commit/f3dde8efe2a5f24b1193148cead8991a477c488b

techdabbler, 2017-09-26:
Great. Thanks.  I didn't realize there were not x86 images at all.  What is the difference between  20,  21 and 10?

Fourdee, 2017-09-26:
@techdabbler 

List of HW_ARCH's contained here:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-obtain_hw_model#L68-L73

Fourdee, 2017-09-26:
@techdabbler 

Looks good 👍 :
```
root@DietPi:~# systemctl status minio.service -l
● minio.service - Minio
   Loaded: loaded (/etc/systemd/system/minio.service; disabled; vendor preset: e
nabled)
   Active: active (running) since Tue 2017-09-26 18:44:12 BST; 1min 5
5s ago
     Docs: https://docs.minio.io
  Process: 10924 ExecStartPre=/bin/bash -c [ -n "${MINIO_VOLUMES}" ] || echo "Va
riable MINIO_VOLUMES not set in /etc/default/minio" (code=exited, status=0/SUCCE
SS)
 Main PID: 10926 (minio)
    Tasks: 9 (limit: 4915)
   CGroup: /system.slice/minio.service
           └─10926 /usr/local/bin/minio server /mnt/dietpi_userdata/minio-data

Sep 26 18:44:13 DietPi minio[10926]:    http://192.168.0.211:9000  http://127.0.
0.1:9000
Sep 26 18:44:13 DietPi minio[10926]: Command-line Access: https://docs.minio.io/
docs/minio-client-quickstart-guide
Sep 26 18:44:13 DietPi minio[10926]:    $ mc config host add myminio http://192.
168.0.211:9000 H7XB8NQS28HCPU1D6PQG nKOYBRJW/kv6zgWolU2pOTtzkYQVwc/nlyPdqebz
Sep 26 18:44:13 DietPi minio[10926]: Object API (Amazon S3 compatible):
Sep 26 18:44:13 DietPi minio[10926]:    Go:         https://docs.minio.io/docs/g
olang-client-quickstart-guide
```

To Do:
- Add to ```dietpi-process_tool```

techdabbler, 2017-09-26:
@Fourdee 

RE: HW_ARCHs -- I see what happened I was using a guide on adding software that appears to have an old ARCH list since I can't find 20 and 21 anymore at all.  If this is still considered a current guide it might be good to update.  Not sure how that works.

[https://github.com/Fourdee/DietPi/issues/490](https://github.com/Fourdee/DietPi/issues/490)

Happy that fixed the x86_64 problem. I love easy stuff.

Fourdee, 2017-10-14:
> If this is still considered a current guide it might be good to update. Not sure how that works.

@techdabbler 

Apologies, i've updated the link to current indexs 👍

Fourdee, 2017-10-14:
@techdabbler 
Running on x86_64:
```
root@DietPi:~# systemctl status minio
● minio.service - Minio
   Loaded: loaded (/etc/systemd/system/minio.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-10-14 16:21:38 BST; 8min ago
     Docs: https://docs.minio.io
  Process: 4955 ExecStartPre=/bin/bash -c [ -n "${MINIO_VOLUMES}" ] || echo "Variable MINIO_VOLUMES not set in /etc/default/minio" (code=exited, status=0/SUC
CESS)
 Main PID: 4958 (minio)
    Tasks: 10 (limit: 4915)
   CGroup: /system.slice/minio.service
           └─4958 /usr/local/bin/minio server /mnt/dietpi_userdata/minio-data
```

Believe we can now mark this as completed?

techdabbler, 2017-10-14:
Yes we can.

Fourdee, 2017-10-14:
Completed by @techdabbler , available for install in v157 👍

closed 2017-10-14 by Fourdee