MichaIng/DietPi #1161: DietPi-Software | PiHole

Hestonian, 2017-09-29:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 156
- SBC Device RPi 2
- Power supply used 5v 2A Anker
- SD card used Kingston 32GB
- Distro Jessie | Linux Gateway 4.9.40+ armv6l GNU/Linux

#### Additional Information (if applicable):
- Software PiHole installation
Post upgrade from v155 to v156, pi-hole stopped working as DNS server resulting in URL requests from devices within network timing out when accessing external sites.  Tried uninstall and re-install of the Pi-Hole through DietPi-Software, resulting in the following error during install:
 #               processing | D i e t  P i             #

 [Info] Apt-get is processing, please wait...

Segmentation fault
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'bluetooth' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
Segmentation fault
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'bluetooth' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues

Ran "apt-get install -f" followed by "apt-get autoremove" to get the following error:

Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  libpci3
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 32 not upgraded.
root@gateway:~# apt-get autoremove
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  libpci3
0 upgraded, 0 newly installed, 1 to remove and 32 not upgraded.
After this operation, 68.6 kB disk space will be freed.
Do you want to continue? [Y/n]
Segmentation fault
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'bluetooth' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

 
- Can this issue be replicated on a fresh installation of DietPi?  - Not verified yet.

#### Expected behaviour:
The PiHole install should have completed without failure which is not what is happening.  How to revert back to v155 to test if this is a DietPi-Software incompatibility for the utility?

#### Actual behaviour:
Apt-get errors preventing PiHole from being re-installed

#### Steps to reproduce:
Uninstall PiHole and re-install in version v155.

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
None

Fourdee, 2017-09-30:
@Hestonian 

Thanks for the report 👍 

> Segmentation fault

Hardware failure, PSU instability, and/or SD card insufficient. Not a DietPi specific issue.

Please try 
- re-writing the DietPi image to a well branded SD card (EG: Sandisk)
- Using official RPi PSU (5v/2.5A)

Hestonian, 2017-10-01:
OK.  But do you have a process to revert to an earlier version of DietPi i.e. so I can revert to v 155 to test?

Fourdee, 2017-10-02:
@Hestonian 
> OK. But do you have a process to revert to an earlier version of DietPi i.e. so I can revert to v 155 to test?

We cannot replicate segmentation faults from v151 > v156. There is nothing wrong with the v155 > v156 upgrade. The issue lies with your hardware.

Please look at resolving your hardware setup, ensuring stable operation of the RPi, by using official PSU and well branded SD card (eg: sandisk).

I'll mark this as closed.

closed 2017-10-02 by Fourdee

Invictaz, 2017-10-02:
Unplugging the Pi with write actions still running (since the RAM is also on the disk) is a possible explanation.

Hestonian, 2017-10-02:
I had PiHole running on the same hardware/power supply since v148 so I had a strong basis for investigation.   SD card corruption during the upgrade from 155 to 156 write is a possibility.    

Appreciate that the issue is marked closed so I'll try re-install.   Thanks.