MichaIng/DietPi #1167: nodejs raspberrypi zero

dragouf, 2017-10-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 156
- SBC Device (EG: RPi 3)? RPI Zero W
- Power supply used (EG: 5v 1A RAVpower)? 5v 2A
- SD card used (EG: Sandisk ultra)? sandisk 16Go
- Distro (EG: Jessie) | ```uname -a```? Linux RaspberryZero 4.9.40+

#### Additional Information (if applicable):
- Software title? NodeJs
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:

nodejs should be install

#### Actual behaviour:

not install

#### Steps to reproduce:

just try install script for nodejs

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

latest version of nodejs 8 does not contains arm61 binary. 
(http://nodejs.org/dist/latest/)
consequently script failed for arm 6 board.

Bridouz, 2017-10-02:
You van try to use this script to install NodeJS on ARMv6 board

https://github.com/sdesalas/node-pi-zero

michaelkay, 2017-10-02:
Root cause is detailed here https://github.com/taaem/nodejs-linux-installer/issues/9

Fourdee, 2017-10-03:
@michaelkay 

Yep 👍 

arm6l binaries missing from http://nodejs.org/dist/latest/:
[image]

Fourdee, 2018-02-02:
Seems ARMv6 binaries are now available:
http://nodejs.org/dist/latest/node-v9.5.0-linux-armv6l.tar.gz

closed 2018-02-02 by Fourdee