MichaIng/DietPi #1171: DietPi Image | UEFI compatible  x86_64 image

Fourdee, 2017-10-04:
> https://twitter.com/DietPi_/status/915396664013312000

Fourdee, 2017-10-08:
Z83:
[image]

Fourdee, 2017-10-13:
### I've created a UEFI based image ready for testing:
- Based on @sal666 excellent Clonezilla install method 👍 https://github.com/Fourdee/DietPi/issues/1007#issuecomment-328102850
- Targeted for mmcblk0 devices (eg: x86_64 systems with onboard EMMC)

#### We are looking for users to assist with testing mmcblk0 devices. EG: Beelink AP32, Z83 etc
However, you can replace the following files, after **STEP 3**:
- Edit ```/EFI/boot/grub.cfg```, ```/syslinux/isolinux.cfg``` and ```/syslinux/syslinux.cfg``` and replace 'mmcblk0' with the name of your target hard disk, eg ```sda``` ```sdb``` etc.

#### Step 1:
- Download [clonezilla](http://clonezilla.org/downloads/download.php?branch=stable) and [rufus](https://rufus.akeo.ie/)

#### Step 2: Use Rufus to write the ISO
- Run Rufus
- select the Clonezilla iso image
- Use the following settings
[image]
- Click start to write the Clonezilla image

#### Step 3: Add DietPi image and @sal666 Clonezilla installer
- Download http://dietpi.com/downloads/testing/DietPi_NativePC-x86_64-(Stretch)_UEFI.7z
- Extract contents to the same USB pen drive used in Rufus

#### Step 4:
- Power on the device with USB pen drive attached
- Enter BIOS and select USB pen drive as the 1st boot device, then save and exit BIOS.
- Select "Install DietPi"

Fourdee, 2017-11-13:
Rufus install process confirmed working by one of our users:
http://dietpi.com/phpbb/viewtopic.php?f=12&t=2452&p=9514#p9501

--- 

Outstanding issues:
- User reports no audio on device.
- Implement GPU driver install?

georgecarstoiu, 2017-12-22:
Tried installing it on an  Intel NUC NUC6CAYH - no emmc, just one normal ssd.

Followed the steps described above, but it didn't work. Was unable to install it in the end. Still would be nice to have a normal ISO.

nwcatalyst, 2018-01-11:
Tried on  NUC NUC6CAYS (with 32GB eMMC) and the install process did not work the first time (odd error about not being able to find mmcblk0 though it had just returned messages saying it was located...

Tried again immediately and it seemed to work.

I did see a bunch of I/O errors for the eMMC during install, which was a major reason I came to dietpi from trying to run Mint on it. I will get it up and running and report back if I am having the same issues. 

Thanks for all the efforts...

Would love to have audio support.

Fourdee, 2018-01-11:
@nwcatalyst 

Thanks for testing 👍 

This may be a hardware/EMMC issue (may also be related to ```discard fstab```). Do you experience EMMC I/O errors with other OS?

It may also be worth trying our new PREP system, requires a debian installation on EMMC:
https://github.com/Fourdee/DietPi/issues/1285

nwcatalyst, 2018-01-11:
I had no issues with default W10 installation before trying various Linux distros.

I will try again.

-------- Original Message --------

Fourdee, 2018-01-21:
Notes to prep:
- Run prep
- LXDE live, resize partition
- zerofree
- Clonezilla

MichaIng, 2018-01-23:
v6.0 EFI image for testing is there btw: http://dietpi.com/downloads/images/

@Fourdee
Is it wanted that this is in releases instead of testing?

Fourdee, 2018-01-23:
@MichaIng 

> Is it wanted that this is in releases instead of testing?

Yep tested it personally and ready for use, however, dietpi.com links are not updated yet (still using old <v159 image). I'll do the dietpi.com links prior to v6.0 completion.

Fourdee, 2018-01-28:
Marking as completed, image is now available, and, PREP method also:
https://github.com/Fourdee/DietPi/issues/1171#issuecomment-336522021

Fourdee, 2018-02-08:
http://dietpi.com/phpbb/viewtopic.php?f=9&t=2809&p=10773#p10773

tomillr, 2018-04-05:
Tried to boot the x64 EFi-build on a Beelink AP34 (Apollo Lake, internal eMMC) but the machine crashes with a non-blinking cursor right after selecting the UEFI usb drive from the bootloader of the machine.

Could it be that there's a difference between the Atom and Apollo Lake-based CPUs?

In fact, the only Linux distro I've been able to get to boot on the machine is one made by [Linuxium](http://linuxiumcomau.blogspot.com/2018/03/ubuntu-16044-for-intel-atom-and-apollo.html). And he always provides two different versions: One for Atom and another one for Apollo Lake.

Has anyone been able to install DietPi on an AP34?

Fourdee, 2018-04-05:
@tomillr 

> Could it be that there's a difference between the Atom and Apollo Lake-based CPUs?

Yep, appears to be an issue with either the CPU and/or device (or BIOS limitation) itself. Modifications are required:
http://rglinuxtech.com/?p=2021

> I did look at the bios settings, and there would appear to be a solution, as there is an option for ‘Legacy’ boot instead of UEFI.   I tried this, and it made no difference – all you see is a blank screen, with a small flickering cursor in the top left corner..

tomillr, 2018-04-06:
Sounds like exactly the same behavior on my system.

Using the 'isorespin.sh' script I'm able to get an Ubuntu iso to boot properly on the Apollo Lake system. The process the author uses is also described on the website. [LINK](http://linuxiumcomau.blogspot.com/2017/06/customizing-ubuntu-isos-documentation.html)

Any chance of getting this fixed somehow? Maybe it's possible to adopt the method linked above to work with Debian and hence RetroPie?

Fourdee, 2018-04-07:
@tomillr 

Seems the AP34 BIOS is 32bit, hence the need for all these software changes.

Unable to find a 64bit BIOS update from beelink.

> Any chance of getting this fixed somehow? 

Unfortunately, Its not something I can start at the moment, and, I also lack the device for testing. The only thing I can suggest is to also try contacting Beelink for a 64bit BIOS update.

tomillr, 2018-04-08:
Already tried to convince Beelink a couple of times but it doesn't seem like they are interested in support their devices.

How about a donation of 100 bucks so that you can get your hands on an Apollo Lake device?

tomillr, 2018-04-19:
Won't work on my system.

However, the rescue disk from [AltLinux](https://en.altlinux.org/Rescue) boots right away. Maybe it is possible to take a look at how they handle boot and apply this to the x86 build of DietPi? Looks like there image can boot on pretty much everything...

Fourdee, 2018-04-20:
@tomillr 

> By selecting "64-bit install" from the initial boot menu, debian-installer will install a 64-bit (amd64) version of Debian. The system will automatically detect that the underlying UEFI firmware is 32-bit and will install the appropriate version of grub-efi to work with it.

Does a box standard Debian install provide the same boot issue? Reading the above, it should detect efi32 and install/configure for it automatically.

tomillr, 2018-04-20:
Yes, the standard Debian multiarch distro with UEFI support for mixed mode systems has the same issue. 

With AltLinux or the "respun" Ubuntu images from Linuxium I can just flash the iso to a usb strick with Etcher and boot right of the stick without any problems.

tomillr, 2018-04-20:
This script can be used to make Ubuntu boot on either Intel Atom or Apollo Lake system:
https://drive.google.com/file/d/0B99O3A0dDe67S053UE8zN3NwM2c/edit

Looks like it's using GRUB_EFI_IA32 and applying a custom set of values to get Linux to boot.

tomillr, 2018-04-24:
Is there a way to change the boot loader the DietPi install medium uses to [the rEFInd Boot Manager](http://www.rodsbooks.com/refind/)? 

While GRUB won't boot, I can use the rEFInd boot manager just fine on my Apollo Lake mini pc.

MichaIng, 2018-04-24:
@tomillr 
Hmm, so far within our code (not just preparation script), we assume grub being installed on x86_64 systems. Using a different bootloader will lead to unexpected errors when using dietpi-config and other scripts.
Beside that, it would be not too hard to just ignore bootloader (not install any) by commenting/removing the following lines: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L708-L718

Did you try to manually `apt install grub-efi-amd64` or if I understood right, that you need 32bit efi `grub-efi-ia32`?
Second try to flash the Debian installer iso via Rufus as MBR for BIOS, to let it install Debian with BIOS as well. Finally for my impression there is no real argument that marks BIOS as no-go in most cases.
€: Ah on the other hand, `dd image` should be right in every case and GPT/MBR should be decidable when formatting the system drive during installation.

For now I think it will should be easier to find fixes with grub on troubling machines, than implementing a new x86 bootloader with all the additional checks and code adjustments that going along with this.

But these are rough thoughts, as I did not read deeper into the details of Apollo Lake mini pc. Let me do that tomorrow, than I hopefully have a clearer image of what can be possible 😉.

tomillr, 2018-04-25:
> we assume grub being installed on x86_64 systems.
As far as I was able to glean from the documentation over at [Linuxium's blog](http://linuxiumcomau.blogspot.com/2017/06/customizing-ubuntu-isos-documentation.html), Apollo Lake chips can run a 64bit OS (like the preinstalled Windows 10) but the UEFI implementation is somehow still 32bit.

> try to flash the Debian installer iso via Rufus as MBR for BIOS, to let it install Debian with BIOS as well. Finally for my impression there is no real argument that marks BIOS as no-go in most cases.

I've already tried using MBR with Rufus. Unfortunately, the legacy bios option is not available on this system. So I don't really have the option to use anything but UEFI to boot.

Fourdee, 2018-05-05:
@tomillr 

To implement to support for this, i need to obtain a device with 32bit UEFI BIOS with no CSM/legacy support. I've looked at the AP34:
https://www.gearbest.com/tv-box-mini-pc/pp_617335.html?wid=1433363

£111 + delivery + import tax
Probably looking at £150~ to cover it.

Unless I can get a device to test locally, we are basically going in here blind. I'll need the device to implement, however, its not something we have the funds for currently.

MichaIng, 2018-05-11:
@Fourdee 
Is there some technical/design reason why EFI/GPT images can't just be created and flashed via rufus "DD image" like MBR/BIOS images, ready to run?

Fourdee, 2018-05-11:
@MichaIng 

> Is there some technical/design reason why EFI/GPT images can't just be created and flashed via rufus "DD image" like MBR/BIOS images, ready to run?

I'am not 100% clued up on UEFI, however, I know we cant simply DD write the image, as the UEFI bios does not register it as a UEFI device.

Best guess: Clonezilla enables the UEFI bios to see the new UEFI device. I believe it generates a unique UEFI code on the system, which changes on each UEFI device.

sal666, 2018-05-24:
Unfortunately, GRUB is not able to boot Apollo Lake devices.
isorespin's author Ian Morrison (AKA Linuxium), explains it well [here](http://linuxiumcomau.blogspot.com.tr/2017/06/customizing-ubuntu-isos-documentation.html).
In a nutshell, we need to replace GRUB with [rEFInd](https://www.rodsbooks.com/refind), both in the installation pen drive and in the target device.

To prepare an installation pen drive with rEFInd follow these steps:

1. Create DietPi's installation pen drive. Use 'Method 2' described above.

2. Install rEFInd to the pen drive (in a Linux machine and assuming it's mounted as /dev/sdb1):
>     sudo umount /dev/sdb*
>     wget "https://sourceforge.net/projects/refind/files/0.11.2/refind-bin-0.11.2.zip"
>     7z x refind-bin-0.11.2.zip
>     sudo refind-bin-0.11.2/refind-install --usedefault /dev/sdb1  # After execution /dev/sdb1 will remain mounted at /tmp/refind_install
>     mv refind-bin-0.11.2 /tmp/refind_install"                     # move to the pen drive so we can install it later on the target device

3. Configure rEFInd
     - Extract this zip to /tmp/refind_install/EFI/boot: [extrafiles.zip](https://github.com/Fourdee/DietPi/files/2035085/extrafiles.zip)
     - GRUB and it's config files can also be removed from the pen drive.
     - Eject pen drive.

Then Install DietPi onto target device:
   - Boot device with pen drive (on one of my test devices sometimes rEFInd reads a ghost keypress and runs the first boot option. In that case quickly press Ctrl+Alt+Del to reboot and try again).
   - Install DietPi to mmcblk0, mmcblk1 or sda.
   - Once completed, the system will power off.

To install rEFInd onto target device:
   - Boot again with pen drive and select "Boot Clonezilla live".
   - At the first menu, select "Enter_shell" and execute these commands:
>       sudo su
>       PART1=mmcblk0p1   # DietPi's first partition (usually mmcblk0p1, mmcblk1p1 or sda1)
>       PART2=mmcblk0p2   # DietPi's first partition (usually mmcblk0p2, mmcblk1p2 or sda2)
>       /lib/live/mount/medium/refind-bin-0.11.2/refind-install --usedefault /dev/$PART1
>       rm -rf /tmp/refind_install/EFI/arch_grub    # not needed (maybe DietPi team can verify), and rEFInd would show it as a second boot option
>       umount /tmp/refind_install
>       mount /dev/$PART2 /mnt
>       # Block grub and lilo packages, as they can interfere with rEFInd
>       printf "Package: grub*:* lilo*:*\nPin: release *\nPin-Priority: -1\n" > /mnt/etc/apt/preferences
>       umount /mnt
>       poweroff
   - Extract pen drive and boot DietPi.
   - If your device is not booting into rEFInd then boot to EFI settings and reset to default settings. That will reset EFI's NVRAM entries, as explained [here](http://www.rodsbooks.com/refind/bootcoup.html).
   - Log in as root and run these commands to get rid of GRUB:
>      apt purge grub*  # uninstall grub
>      apt autoremove   # uninstall grub dependencies. 'efibootmgr' will be uninstalled. It's not needed, but we can install it again if we want to.

Hope it helps you all ;)

tomillr, 2018-06-03:
sal666, thanks for the nice instruction.

I've tried to follow your procedure using an Ubuntu 18.04 live version but have run in to a couple of issues:

7zip is apparently not part of the regular Ubuntu distro and I wasn't able to apt-get it either. That's why I had to manually extract the refind-bin-0.11.2.zip.

I've also copied your files from the extrafiles.zip manually to /dev/sdb1/EFI/boot.

I was able to successfully boot rEFInd from the usb stick and start the the DietPi installer. Unfortunately, the installer halted with with a couple of error messages.

[image]

Any idea what might have gone wrong here?

MichaIng, 2018-06-03:
@tomillr
Hmm apt install `7zip-full` does not work?? On 17.10 at least it worked.
Otherwise since it's zip, `unzip refind-bin-0.11.2.zip` should do the same.

The screen does not show direct DietPi-PREP error messages. At what install stage this showed up exactly?

tomillr, 2018-06-03:
> The screen does not show direct DietPi-PREP error messages. At what install stage this showed up exactly?

After selecting the "Install DietPi" entry from the rEFInd boot menu the screen goes black and starts vmlinuz with a long list of loading parameters. Right after that the screen refreshes and lists the error message seen on the attached photo above.

Here's what the used usb stick looks like at the moment:

[image]
[image]

sal666, 2018-06-04:
Current DietPi installer is based on [Clonezilla Live](https://clonezilla.org/downloads.php), which is based on Debian.
I've prepared a new installer based on Clonezilla Live **alternative**, which is based on Ubuntu.
Can you test it?
Just delete all content from your pen drive (it must still be formated with GPT+FAT) and then extract there this [zip](https://files.fm/u/byeen4ku). rEFInd is already installed so you can proceed directly to DietPi installation.

sal666, 2018-06-04:
On the other hand, mce errors are caused by faulty hardware, so not sure if this will help.
http://www.advancedclustering.com/act_kb/what-are-machine-check-exceptions-or-mce/
http://mcelog.org/manpage.html

tomillr, 2018-06-04:
That version won't start either. But I'm getting a new error message alongside the one pictured above which says that the EFI table couldn't be read.

Going to try again and make a new screenshot tomorrow.

Regarding the possible faulty hardware: I've been able to run Windows 10 and Ubuntu on the device without any errors pointing to a hardware defect.

DarrenRainey, 2018-06-09:
Any chance of getting a 32bit version as many Windows tablets use a 32bit UEFI but are capable of running a 64bit OS.

sal666, 2018-06-11:
@tomillr What device are you using? Please, post make and model. Are you sure it's capable of booting 64bit OS's?

leviticus1, 2018-07-20:
i ve flashed dietpi img/iso im a noob and there is no gui no cusror nothing, etcher confirns success but nothing can anyone help me get this sorted thanks any advice will do i know how to use the terminal ect

leviticus1, 2018-07-20:
a fat arse pi needs compiling

Kung-Foo-Kamel, 2018-09-23:
In case anyone needs the info, this works perfectly on a Beelink BT3 with 64bit UEFI Bios, Specs:

CPU: Atom z8300
Ram: 2GB DDR3
eMMC: 32GB

lucario1001, 2018-10-02:
This will boot into the installer on the imtel compute stick STCK1A8LFC but the installer will fail because it's not detecting a mmcblk0 if your booting from a SD card but if I can find a way to edit the file to write it i can install it don't know if the wifi will work though because it's a Ubuntu only stick with custom realtek drivers for it

Specs:
Intel atom 1.33ghz turbo boost up to 1.83ghz
1gb ram
8gb emmc

MichaIng, 2018-10-02:
@lucario1001 
The problem is, that x86 images expect the rootfs the be on `/dev/sda[0-9]` and `fstab` + bootloader need to be adjusted to boot into the correct partition.

Adjusting `/etc/fstab` to point to the right boot+root partition is easy, but I am not sure if `grub` (bootloader) can be easily manually patched. Usually there are some steps on the running system needed, including chroot into the destination root partition, and using `update-grub` command 🤔.

lucario1001, 2018-10-02:
i mean it there is a emmc inside of it its only 8gb and there is only 1 usb port on it so the only way to install it is thru a micro sd card slot but it does boot into the clonezilla installer but it does not detect the mmcblk0 because its on another location (ie mmcblk1) and the sd card is emmcblk2

MichaIng, 2018-10-02:
@lucario1001 
Ahh, sorry, jep totally forgot that UEFI image is clonezilla installer 😅.

Hmm hope someone has more knowledge about Clonezilla and how to change the target drive, at least I couldn't find some related config file or such. I just see the iso `\DietPi_v6.0_NativePC-EFI-x86_64-Stretch.iso\home\partimag\DietPi\` containing image files that are named `mmcblk0`. Not sure if this mean that it can only be installed to this drive...
Also the problem with wrong fstab/grub would theoretically stay. Not sure if Clinezilla adjusts this automatically in case?

sal666, 2018-10-02:
@lucario1001
Clonezilla can be tuned so that it will ask for the target device.
Take a look at my final update to this comment:  https://github.com/Fourdee/DietPi/issues/1171#issuecomment-391682301

MichaIng, 2018-10-02:
@sal666 
> Clonezilla can be tuned so that it will ask for the target device.

This is great. We should have this by default in our UEFI image.

Where exactly this "ask_user" needs to be added? I couldn't find any cfg file with boot device issue in our iso. Also your `grub.cfg` does not contain any boot device info, at least I could not find it.

sal666, 2018-10-02:
"ask_user" replaces the name of the target device. 
If you use an installation media based on rEFInd as described above, the command to start Clonezilla is at refind.conf, but it's already patched to ask the use for the target device. Try following the described procedure to create a bootable pen drive and to perform the installation itself.
"grub.cfg" it's not used in this case. It's used in the classic installer based on Grub.

lucario1001, 2018-10-02:
ok so good news bad news so good news it booted and asked me to where to install it
bad news it wont install be the destination is too small 7.6 gb emmc
disk image size is 31gb 😭

sal666, 2018-10-02:
Try this:
http://www.clonezilla.org/clonezilla-live/doc/02_Restore_disk_image/advanced/09-advanced-param.php

lucario1001, 2018-10-02:
so it skips the error message but it still wont flash the image
`Failed to create a partition table on this disk: /dev/mmcblk1
Is this disk too small: /dev/mmcblk1?`

sal666, 2018-10-03:
Use -k1 to create partition table proportionally.
I've updated config files posted at the procedure shared above.

lucario1001, 2018-10-03:
still does not work after with the new config file gives me the same error i re-did the usb also

lucario1001, 2018-10-08:
any news yet? @sal666

MichaIng, 2018-10-08:
I reopen the issue:
- Maybe we can create a new official UEFI image, based on tweaked Clonezilla to allow target drive selection.
- Hopefully include a fix to allow install on at least 4GB drives. Should be possible, by reducing to included DietPi image first via e.g. zerofree and reducing partition size to minimum?
- However I have really no experience with Clonezilla sadly.

sal666, 2018-10-08:
Good idea.
I've managed to create an image that is restorable to a drive as small as 1GB. It's just a question of taking the image from a smaller drive. It seems that Clonezilla is able to resize target partitions, but limits this to target drives not smaller than one half the source drive.

I was trying to upload an updated v6.16 image, but I don't know how to sysprep DietPi after updating and before making a new image.

Anyhow, tomorrow I could upload a v6.0 version with rEFInd instead of grub and restorable to small drives.

sal666, 2018-10-09:
This is the image I've prepared:
[https://ufile.io/yp04k](https://ufile.io/yp04k)

Basically it's [this image](http://dietpi.com/downloads/images/DietPi_v6.0_NativePC-EFI-x86_64-Stretch.iso) with this changes/additions:
1. Restorable on small disks (tested successfully on 1GB pen drives).
2. Destination drive is chosen by the user.
3. Grub replaced by rEFInd, so it will boot on Apollo Lake and similar x64 devices with x32 EFI.

After booting and upgrading remember to uninstall grub:
>apt purge grub*
>apt autoremove

Fourdee, 2018-10-18:
@sal666 

> This is the image I've prepared:

Many thanks for this, and, apologies for the delay 👍 

> Grub replaced by rEFInd,

We need to update this image, and be able to replicate rEFInd install and configuration if we are to switch. Are you able to share how you converted this image from grub to rEFInd?

tomillr, 2018-10-20:
Could this lead to an official DietPi-Image that runs on Apollo Lake-based devices out of the box?

In that case I'd be willing to help out with testing on a Beelink AP34.

Fourdee, 2018-10-21:
@sal666 

> Clonezilla live have already been updated, so now we can use it out of the box. No need to put rEFInd on installation disk any more.

🥂 

> Can you provide an updated image so I can test it on my Apollo Lake device? Once this works fine preparing the Clonezilla installer is quite straightforward.

Yep, leave it with me, may be a few days until I can get round to it mind.

tomillr, 2018-10-24:
> @sal666
> 
> > Can you provide an updated image so I can test it on my Apollo Lake device? Once this works fine preparing the Clonezilla installer is quite straightforward.
> 
> Apologies for the delay, below is the latest DietPi (v6.16) native PC UEFI clone using Clonezilla:
> https://dietpi.com/downloads/testing/DietPi-UEFI-v6.16_clonezilla.7z

Should that image work on Apollo Lake devices right out of the box? Or you first have to manually edit the iso?

Fourdee, 2018-10-24:
@tomillr 

Should do, using the latest version of clonezilla. Although, might be best to wait for @sal666 so we can update our image write process, else, you'll need to use pure clonezilla.

[58 more replies]

closed 2019-09-02 by MichaIng