MichaIng/DietPi #1175: Multiple Pull Request Best Practices?

techdabbler, 2017-10-06:
I'm relatively new to git and I'm curious if I am adding multiple applications into dietpi-software what is the best way to do the pull requests? 

Right now I made an individual pull request for each software as if they were the only one being merged into the testing branch.  My thought was to keep them separate so they could be merged and tested individually.   This still makes sense to me but...

I tried merging on after each other on my own branch for testing and realized it gets messy after a certain point as some lines are the same and it split the code blocks into not terribly helpful chunks to resolve.  (minio update and fuguhub merged fine -- when i went to merge docker is where the conflicts weren't so obviously simple to merge).  

I could throw all 3-4 software into a single pull request but that seems like it could be unwieldy a times.

Any direction on preferences or best practices for this sort of edits would be appreciated.

MichaIng, 2017-10-06:
Other way would be to add one by one, but wait for one being merged, before you add the next. Takes more time but guarantees that everything can be tested well seperated and you can focus on potential fixes and improvements.

MichaIng, 2018-01-23:
Again another way is to create the PRs as branches based on each other. So you create the first PR with branch A. Then for 2nd PR you open a file from branch A and create a new branch B from there. Then the branches can be merged into each other for sure, if done in the right order. Of course single adjustments could break the line 😆.

No I think, just do intuitive. If you have several ideas, related to the same file lines (so that conflicts are possible) then they're most likely also related by topic and can be easier offered as single PR. If there is huge disagreement with one part, then it can be also reverted by another commit.

Will mark as closed, as there seem to be no open question for now. Reopen on demand.

closed 2018-01-23 by MichaIng