MichaIng/DietPi #1179: Image Request: Artila Matrix-700

lukecyca, 2017-10-07:
[Artila makes nice rugged SBCs](http://www.artila.com/en/p_matrix.html). The 7xx series uses ATMEL ATSAMA5D35. The older 5xx series uses ATMEL AT91SAM9G20.

* http://www.artila.com/docs/Matrix-700/Matrix-700_Datasheet.pdf
* http://www.artila.com/docs/Matrix-504/Matrix-504_Datasheet.pdf

I have no idea is it is feasible to make DietPi run on them, but I would sure like to use DietPi instead of the vendor's image!

Fourdee, 2017-10-08:
Hi,

Interesting device.

Unfortunately, this is probably not something we'll officially provide images for, due to limited market from our user base, and, unit cost to develop image is outside our reach.

If I can get my hands on one of the devices, we can probably do it, outside of that, you'd need to implement the image yourself:
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

Fourdee, 2018-02-22:
Closing due to lack of community interest in this device.

closed 2018-02-22 by Fourdee