MichaIng/DietPi #1183: DietPi-Software | DLNA render

Fourdee, 2017-10-09:
> http://blog.scphillips.com/posts/2014/05/playing-music-on-a-raspberry-pi-using-upnp-and-dlna-v3/ | https://github.com/hzeller/gmrender-resurrect/blob/master/INSTALL.md

Fourdee, 2017-10-09:
Binaries compiled and hosted on dietpi.com

Online doc:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9012#p9012

Fourdee, 2017-10-09:
Note to self:
- Ensure ALSA stream is free, when playback is not active.

mpember, 2017-10-29:
Why did the update install this software on a system which previously didn't have it?

Fourdee, 2017-10-29:
@mpember 

Hi,

GMrender will be installed on all Allo GUI systems, to match the current version of Allo GUI interface and its pre-req software.

You can disable the service in the web interface, if not required.