MichaIng/DietPi #1185: DietPi-Setup | Inital Setup via Wifi-Hotspot (dietpi-wifi)

k-plan, 2017-10-10:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=2310
> [image]

hoobatoo, 2018-04-02:
[DietPi Forum Post](http://dietpi.com/phpbb/viewtopic.php?f=12&t=2310#p10305)

I wonder if this wifi autohotspot could be incorporated somewhat easier into dietpi? http://www.raspberryconnect.com/network ... et-routing

The service calls a script that checks for presence of the SSID entered into wpa_supplicant.conf, and if not present then starts hostapd and dnsmasq and creates a hotspot. It presently requires mucking about in /etc/network/interfaces and replacing the entire contents with the following. 
` # interfaces(5) file used by ifup(8) and ifdown(8) 
# Please note that this file is written to be used with dhcpcd 
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf' 
# Include files from /etc/network/interfaces.d: 
source-directory /etc/network/interfaces.d`

It does not appear to work as is.

Fourdee, 2018-07-09:
Marking as closed.

Although i'd love to have this implemented, with lack of time and higher priority tickets, not possible currently with lack of interest.

closed 2018-07-09 by Fourdee