MichaIng/DietPi #1189: DietPi-Software | raspotify (Spotify connect client)

Fourdee, 2017-10-11:
> https://www.computeraudiophile.com/forums/topic/32132-allo-sparky-usbridge/?do=findComment&comment=728214

Fourdee, 2017-10-16:
Install method: http://dietpi.com/phpbb/viewtopic.php?f=9&t=2352#p9167

Fourdee, 2017-10-29:
Unable to test as I lack spotify premium account, and, trial has already been used previously.

Fourdee, 2017-10-29:
online doc:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9368#p9368

Fourdee, 2017-10-29:
Stretch repo not available, will use jessie for all distros:
```
Err https://dtcooper.github.io stretch/main armhf Packages
  HttpError404
```

Fourdee, 2017-10-29:
Marking as completed, however, unable to test audio playback at this time: https://github.com/Fourdee/DietPi/issues/1189#issuecomment-340273204

closed 2017-10-29 by Fourdee

georgegeorgetj, 2020-05-08:
Has this been abandoned for DietPi? I can't get to install Raspotify - both the automatic installer as well as the manual fail.

Joulinar, 2020-05-08:
Hi,

Many thanks for your report. What kind of device you are running? It should be still working as I did some test installation last week without issues.

georgegeorgetj, 2020-05-08:
Hi! It's a Nanopi board.I think it was you that we were talking over at the forum Joulinar. 
I'm still trying to get around with it and haven't found a working solution. 
Which method? The automatic installer or the manual? The automatic installer reports that the script is for the Raspberry Pi only. Shouldn't it be a more universal installation since it is through the DietPi image? I have compiled a DietPi image for the NeoPlus2.

Joulinar, 2020-05-08:
Yes it's me again. I guess there is a misunderstanding what DietPi is and how it's working. DietPi themselves isn't an own OS or Debian system. DietPi is a set of scripts on top of a Debian base image. Means for RPi systems the base image is Raspbian Lite. For other systems it's Armbian for example. And your device is not running Raspbian Lite as base image. Therefore Raspotify will not work in your device as long as you are not able to get Raspbian Lite running. Next to that, DietPi is not developing the Rasporify installation script. We simply use the stuff provided by Raspotify.

georgegeorgetj, 2020-05-08:
I see.It's pretty clear I have no luck with this. Thank you so much for clarifying!

Joulinar, 2020-05-08:
Well just get a RPi board and your issue is gone :wink:

georgegeorgetj, 2020-05-08:
I got all RPis except from RPi4. In my project there's space only for a RPiZero which Initially used but due to supporting armv6 set of instructions, I'm not able to run Roon-Bridge. :(