MichaIng/DietPi #1192: DietPi Software does not show MotionEye

fredjedimaster, 2017-10-14:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? v156
- SBC Device (EG: Orange Pi Zero Plus)?
- Power supply used (EG: 5v 2A Samsung original)?
- SD card used (EG: Sandisk 16gb class 10)?

#### Additional Information (if applicable):
- Software title? MotionEye
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:
 After type dietpi-software the MotionEye should be displayed on the list. 

#### Actual behaviour:
MotionEye does not appears in the lists

#### Steps to reproduce:
Install dietpi
type dietpi-software
Search for MotionEye in Software Optmized and Software Additional

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:

Fourdee, 2017-10-14:
@fredjedimaster 

Hi,

Its untested on this device, our main focus is RPi for this software as we support RPi camera in DietPi, hence disabled.

You can override this and force the install to test:
```
dietpi-software install 136
```

fredjedimaster, 2017-10-14:
Thanks for helping.

Did not work.

I got the following messages:

> 
> DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Install software
>  Please wait...
> 
>  MotionEye: web interface & surveillance for your camera
>  [Info] Please wait...
> 
>  #               processing | D i e t  P i             #
> 
>  [Info] Testing connection to http://github.com/Motion-Project/motion
>  [Info] Max duration of 20 seconds, please wait...
>  [Ok] Connection test | Completed
> 
>  [Info] Apt-get is processing, please wait...
> 
> --2017-10-14 10:36:08--  http://github.com/Motion-Project/motion/releases/download/release-4.0.1/pi_jessie_motion_4.0.1-1_armhf.deb
> Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
> Connecting to github.com (github.com)|192.30.253.112|:80... connected.
> HTTP request sent, awaiting response... 301 Moved Permanently
> Location: https://github.com/Motion-Project/motion/releases/download/release-4.0.1/pi_jessie_motion_4.0.1-1_armhf.deb [following]
> --2017-10-14 10:36:09--  https://github.com/Motion-Project/motion/releases/download/release-4.0.1/pi_jessie_motion_4.0.1-1_armhf.deb
> Connecting to github.com (github.com)|192.30.253.112|:443... connected.
> HTTP request sent, awaiting response... 302 Found
> Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/20822620/44aa1c18-9a2d-11e6-9750-cf4e6f150f31?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171014%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171014T143611Z&X-Amz-Expires=300&X-Amz-Signature=357171f3c7a69d429625b669b108bedf96fcd3245a2b753782c340598f52bd81&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dpi_jessie_motion_4.0.1-1_armhf.deb&response-content-type=application%2Foctet-stream [following]
> --2017-10-14 10:36:12--  https://github-production-release-asset-2e65be.s3.amazonaws.com/20822620/44aa1c18-9a2d-11e6-9750-cf4e6f150f31?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171014%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171014T143611Z&X-Amz-Expires=300&X-Amz-Signature=357171f3c7a69d429625b669b108bedf96fcd3245a2b753782c340598f52bd81&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dpi_jessie_motion_4.0.1-1_armhf.deb&response-content-type=application%2Foctet-stream
> Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.229.67
> Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.229.67|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 277688 (271K) [application/octet-stream]
> Saving to: ‘package.deb’
> 
> package.deb         100%[=====================>] 271.18K  21.2KB/s   in 13s
> 
> 2017-10-14 10:36:28 (21.2 KB/s) - ‘package.deb’ saved [277688/277688]
> 
> (Reading database ... 42034 files and directories currently installed.)
> Preparing to unpack package.deb ...
> Unpacking motion (4.0.1-1) over (4.0.1-1) ...
> Setting up motion (4.0.1-1) ...
> Processing triggers for systemd (215-17+deb8u7) ...
> Processing triggers for man-db (2.7.0.2-5) ...
> Collecting motioneye
> Collecting pillow (from motioneye)
>   Using cached Pillow-4.3.0.tar.gz
> Requirement already satisfied: tornado>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye)
> Requirement already satisfied: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
> Collecting pycurl (from motioneye)
>   Using cached pycurl-7.43.0.tar.gz
> Requirement already satisfied: olefile in /usr/local/lib/python2.7/dist-packages (from pillow->motioneye)
> Requirement already satisfied: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
> Requirement already satisfied: certifi in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
> Requirement already satisfied: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
> Requirement already satisfied: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye)
> Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from singledispatch->tornado>=3.1->motioneye)
> Building wheels for collected packages: pillow, pycurl
>   Running setup.py bdist_wheel for pillow ... error
>   Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-vGpEr4/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpEQ75Clpip-wheel- --python-tag cp27:
>   running bdist_wheel
>   running build
>   running build_py
>   creating build
>   creating build/lib.linux-armv7l-2.7
[... 307 lines cut ...]
> 
>  DietPi-Services
> ─────────────────────────────────────────────────────
>  Mode: stop
>  Please wait...
> 
>  [Ok] cron
>  [Ok] proftpd
> 
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Optimize and configure software
>  Please wait...
> 
>  [Info] Applying DietPi optimizations and configurations for OrangePi Zero (armv7l)
>  [Info] Please wait...
> 
>  #               processing | D i e t  P i             #
> 
> cp: cannot stat ‘/usr/local/share/motioneye/extra/motioneye.conf.sample’: No such file or directory
> sed: can't read /etc/motioneye/motioneye.conf: No such file or directory
> cp: cannot stat ‘/usr/local/share/motioneye/extra/motioneye.systemd-unit-local’: No such file or directory
> 
>

Fourdee, 2017-10-14:
@fredjedimaster 

> The headers or library files could not be found for zlib,

```
apt-get install -y v4l-utils python python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev
```

```
dietpi-software install 136
```

fredjedimaster, 2017-10-14:
Thanks...

Did not work. 

Got this message:

Failed building wheel for pillow

And this:

Failed building wheel for pycurl

The headers or library files could not be found for zlib,
a required dependency when compiling Pillow from source.

Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-If6HLW/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-tNL87D-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-If6HLW/pillow/

How to fix it?

fredjedimaster, 2017-10-18:
@Fourdee Thanks for the Help. Seems ccrisan does not get support for orange pi's board.

One last ask. What's the command for dietpi-cam? dietpi-software install ???

Where do i find a list with these commands for force installation?

closed 2017-10-18 by fredjedimaster

Fourdee, 2017-10-18:
@fredjedimaster 

```
dietpi-software list
```

Find all software with 'cam' in its name
```
dietpi-software list | grep cam
```