MichaIng/DietPi #1194: Uninstalling DietPiCAM while MotionEYE is installed removes required package `motion`

vilhelmp, 2017-10-16:
## Creating a bug report/issue:
DietPi v.156
RPi 1
5v/1A power supply
microSD card w. adapter.
Jessie

#### Additional Information (if applicable):
- MotionEYE and DietPiCAM

#### Expected behaviour:
Uninstalling DietPiCAM while MotionEYE is installed should not remove the `motion` package. 

#### Actual behaviour:
Uninstalling DietPiCAM removes the `motion` package when MotionEYE is installed.

#### Steps to reproduce:
- Mark and install DietPiCAM and MotionEYE in dietpi-software.
- Uninstall DietPiCAM
- Check if the `motion` package was removed.

#### Did you submit a ```dietpi-bugreport```?
NO

#### Extra details:
--

Fourdee, 2017-10-16:
Thanks for the report 👍 
Confirmed, we'll take a look.

Fourdee, 2017-10-29:
Resolved for v158

closed 2017-10-29 by Fourdee