MichaIng/DietPi #1197: No sound on LDXE or Kodi

Darwin-II, 2017-10-18:
## Creating a bug report/issue:

I recently installed DietPi and started to use it. 

My problem is a lack of audio output. When I try to play a video, or an .mp3 file there is no sound. The same applies for games or anything with sound. 

I have a Raspberry pi 3 model B running the latest install of DietPi (v150 Jessie) on a 32GB Sandisk ultra sd card.

Fourdee, 2017-10-19:
@Darwin-II 

Hi,

Thanks for the report.

Soundcards are disabled by default in DietPi to save on resources.

Please ensure the soundcard is enabled in:
```dietpi-config``` > Audio Options > Soundcard
[image]

---

Please also ensure you are running the latest version of DietPi (v156):
```
dietpi-update
```

---

We cant replicate this on RPi3 in our testing, however, If problems persist, please let us know.

Darwin-II, 2017-10-19:
Thank for the quick response. I now have sound working.

vp02884, 2020-10-23:
I have same problem

recently installed DietPi and started to use it.

My problem is a lack of audio output. When I try to play a video, or an .mp3 file there is no sound. The same applies for games or anything with sound.

I have a Raspberry pi 3 model B running the latest install of DietPi (23.10.2020) on a 32GB Sandisk ultra sd card.

vp02884, 2020-10-23:
Sound setting van not change to enable

vp02884, 2020-10-23:
Sound setting can not change to enable

MichaIng, 2020-10-24:
What happens if you use `dietpi-config` > `Audio Options` > `Enable` (if not yet done) > `Soundcard` > `rpi-bcm2835-auto`, then reboot and play sound?

vp02884, 2020-10-24:
Thenks for your quick responsd ...I manege define rpi-bcm2835-auto as sound card, but ehen i try to change setting yo enable Then it change the card setting back to noneAnd then after reboot no sound ....Best RegardsIlanנשלח מהסמארטפון שלי.
-------- הודעה מקורית --------מאת: MichaIng <notifications@github.com> תאריך: 24.10.2020  03:20  (GMT+02:00) אל: MichaIng/DietPi <DietPi@noreply.github.com> עותק: vp02884 <vp02884@gmail.com>, Comment <comment@noreply.github.com> נושא: Re: [MichaIng/DietPi] No sound on LDXE or Kodi (#1197) 
What happens if you use dietpi-config > Audio Options > Enable (if not yet done) > Soundcard > rpi-bcm2835-auto, then reboot and play sound?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/MichaIng/DietPi/issues/1197#issuecomment-715641359",
"url": "https://github.com/MichaIng/DietPi/issues/1197#issuecomment-715641359",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

MichaIng, 2020-10-24:
> but when i try to change setting to enable Then it change the card setting back to none

What do you mean by "change setting to enable"? When you enabled general audio capability once (basically this installs ALSA), this does not need to be done again and cannot be from the menu. The menu then only shows "Disable" then aside of sound card selection menu.

From the "Soundcard" menu you can select from a bunch of options, most are either a different ways for RPi onboard sound (3.5mm jack or HDMI or auto-detection) or drivers for additional sound cards which are enabled via device tree overlays (hardware features statically loaded on boot). Switching from onboard sound to a device tree overlay driver or the other way round requires a reboot, switching from one onboard method to a different one, e.g. `rpi-bcm2835-auto` to `rpi-bcm2835-3.5mm` to force 3.5mm jack sound, does not require a reboot.

So much for sound card changes, but since there is no "none" entry (which is only set when disabling sound), I think you need to explain more in detail what you did to have it "change the card setting back to none" 🤔.

vp02884, 2020-10-27:
I did it all
Still no sound

vp02884, 2020-10-27:
I dont have sound icon
It looks like sound is off and i cant activate it

MichaIng, 2020-10-27:
I can only repeat my question since I am not able to understand what exactly you did from the point of login:
> I think you need to explain more in detail what you did to have it "change the card setting back to none"

Also which sound icon do you mean? From the desktop panel?

Additionally please post a screenshot (or SSH copy&paste) of the dialogue that is shown when accessing `dietpi-config` > `Audio Options` and in case the same right after a reboot, if the content changes.

ashishsantikari, 2022-06-24:
I also faced a similar issue. When trying to change the value of sound card to either auto or hdmi, the value resets itself to none.

MichaIng, 2022-06-25:
Please open a new issue and go more into detail about where you change the value from auto to hdmi  (desktop or dietpi-config?) and how/where you see that it resets itself to none.