MichaIng/DietPi #1207: [Suggestion] Dietpi-software | Use IPs instead of domains

enolp, 2017-10-22:
## Creating a bug report/issue:
I propose using IPs instead URL domains when updating. Why? Currently I have installed pihole and dnsmasq as a DNS caching server, and when the update process begins, Dietpi-update stops all running services making dietpi.com unreachable because pihole and dnsmasq are deactivated. On the other hand, I have also dnscrypt installed but I verified it and I think it's not the problem.

Doing a dirty trick replacing dietpi.com with its IP address, I was able to update the system but the update zip calls again to dietpi.com (to download a new kernel for example) making an incomplete update.

#### Required Information:
- DietPi Version | 146
- SBC Device | RPi3
- Power supply used | RPi3 Official
- SD card used | Samsung EVO
- Distro (EG: Jessie) | Jessie

#### Additional Information (if applicable):
- Dietpi-update
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Expected behaviour:
Update Dietpi scripts using server IP address

#### Actual behaviour:
dietpi.com can't be reached

#### Steps to reproduce:
Having pihole installed, run dietpi-update and see results

#### Extra details:
Can anyone check it?

Fourdee, 2017-10-26:
@enolp 

> I propose using IPs instead URL domains when updating. Why? Currently I have installed pihole and dnsmasq as a DNS caching server, and when the update process begins, Dietpi-update stops all running services making dietpi.com unreachable because pihole and dnsmasq are deactivated. 

Hi, as of v156, ```dietpi-update``` will restart the dnsmasq service if PiHole is installed: https://github.com/Fourdee/DietPi/issues/1138

This doesn't help with your current version (v146).  A quick/simple solution would be to uninstall PiHole, update DietPi, then reinstall PiHole.

MichaIng, 2018-02-01:
I guess this is can be closed as several updates went on. Reopen on demand.

closed 2018-02-01 by MichaIng