MichaIng/DietPi #1208: Selecting 'Daemon + Drift' as ntp method does not set date

borpin, 2017-10-22:
## Selecting 'Daemon + Drift' as ntp method does not set date
#### Required Information:
- DietPi Version | 157
- SBC Device? OrangePi Zero
- Power supply used (EG: 5v 1A RAVpower)? Good!
- SD card used (EG: Sandisk ultra)? Kingston
- Distro (EG: Jessie) | ```uname -a```? Linux DietPi 3.4.113-sun8i #18 SMP PREEMPT Thu Jun 15 02:16:06 CEST 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behavior:
Selecting 'daemon + drift' as time syncronization method sets the correct date time

#### Actual behavior:
Date / Time is not being set correctly. Daemon does not seem to be running.

#### Steps to reproduce:
When creating a new image, in the dietpi setup configuration file, select 'Daemon + Drift' as ntp method. 

#### Did you submit a ```dietpi-bugreport```?
 31ce7a2b-3cd9-4754-a9b4-24ac5eba4b88-0

#### Extra details:
A message relating to ntp is seen during setup indicating it has been successful.

On a fresh install as tzdata is installed the following message is seen (indicating date/time has not been set)
```
Current default time zone: 'Europe/London'
Local time is now:      Sat Mar 11 12:16:44 GMT 2017.
Universal Time is now:  Sat Mar 11 12:16:44 UTC 2017.
Run 'dpkg-reconfigure tzdata' if you wish to change it.
```
Not sure what should be running as the daemon but there is no sign of ntpd or systemd-timesyncd

Package ntp is installed.

It seems reasonable to use systemd-timesyncd
```
root@DietPi:/lib/systemd/system# systemctl status systemd-timesyncd.service
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; disabled)
   Active: inactive (dead)
     Docs: man:systemd-timesyncd.service(8)
```

#### My Solution locally
- Edit /etc/systemd/timesyncd.conf and enabled the time servers line.
- Enable systemd-timesyncd `systemctl enable systemd-timesyncd.service`
- Start systemd-timesyncd `systemctl start systemd-timesyncd.service`
- Check timedatactl
```
root@DietPi:/usr/sbin# timedatectl status
      Local time: Sun 2017-10-22 09:33:52 BST
  Universal time: Sun 2017-10-22 08:33:52 UTC
        RTC time: Thu 1970-01-01 00:34:09
       Time zone: Europe/London (BST, +0100)
     NTP enabled: yes
NTP synchronized: no
 RTC in local TZ: no
      DST active: yes
 Last DST change: DST began at
                  Sun 2017-03-26 00:59:59 GMT
                  Sun 2017-03-26 02:00:00 BST
 Next DST change: DST ends (the clock jumps one hour backwards) at
                  Sun 2017-10-29 01:59:59 BST
                  Sun 2017-10-29 01:00:00 GMT
```
- Note *NTP synchronized: no*
- Enable syncronization `timedatectl set-ntp true`

Reference used [(https://feeding.cloud.geek.nz/posts/time-synchronization-with-ntp-and-systemd/)] although he intimates that ntp cannot be installed alongside systemd-timesyncd.

MichaIng, 2017-11-09:
Jep, choosing 'Daemon + Drift' doesn't seem to have any effect. No service/process of ntp can be found active.

Fourdee, 2017-11-09:
Quick test on Odroid C2, unable to replicate:
[image]

Could be specific to our OPi Zero image.

MichaIng, 2017-11-10:
- Okay, for the first, `dbus` need to be installed for systemd to initiate `systemd-timesyncd`
- On Jessie then the service is starting as expected.
- But on Stretch, it still fails if `ntp` is still installed and active: https://superuser.com/questions/938926/systemd-timesyncd-service-start-fails
This error really just shows up on Stretch. On Jessie everything works, even if the mentioned `/usr/sbin/ntpd` is present.

https://feeding.cloud.geek.nz/posts/time-synchronization-with-ntp-and-systemd/

In any case the ntp package is not necessary anymore, so should be purged if timesyncd is used, and pool server can be changed via `/etc/systemd/timesyncd`.

Fourdee, 2017-11-10:
@MichaIng 

Great find 👍 

> But on Stretch, it fails if ntp is still installed and active: 
 In any case the ntp package is not necessary anymore, so should be purged if timesyncd is used

So in theory, we could simply ```apt-get purge ntp``` when user selects ```daemon + drift```, and visa versa, reinstall as needed?

MichaIng, 2017-11-10:
Jep, this also works on Jessie and we can keep installation slim this way. ntp pools need to be set in /etc/systemd/timesyncd.conf then.

Fourdee, 2017-11-12:
@MichaIng 

👍 I'll add this to v159 milestone

borpin, 2017-11-12:
It has been suggested to me that timesyncd may not work on a Read Only filesystem https://community.openenergymonitor.org/t/ensuring-emonpi-emonbase-time-is-synchronised-with-an-ntp-server/5487/11

Not sure if this is useful information or not.

Fourdee, 2017-11-13:
Tested on:
- 🈯️ Jessie

Fourdee, 2017-11-13:
Tested after all commits on:
- 🈯️ Jessie

MichaIng, 2017-11-14:
Works great, also tested with Jessie and Stretch VMs back and forth 🙂.

closed 2017-11-15 by Fourdee