MichaIng/DietPi #1209: Nanopi Neo Air (AP6212A)

Dalto1, 2017-10-22:
See this post for info :
[http://dietpi.com/phpbb/viewtopic.php?f=11&t=2205&p=9271#p9271](url)

Fourdee, 2018-02-22:
Marking as closed due to:
- ARMbian based images dropped. 
- No official available Neo Air Debian image
- 0 installs via survey results

closed 2018-02-22 by Fourdee

troyane, 2018-09-06:
@Fourdee I'd like to mention that here are official images available:

* ARMbian -- https://www.armbian.com/nanopi-neo-air/
* Official Neo Air Debian image -- nanopi-neo-air_debian-nas-jessie_4.14.52_20180628.img.zip (Image File with Support for 1-bay NAS Dock, kernel:Linux-4.x.y) available from here -- https://drive.google.com/drive/folders/1_mPtgAVaRtYVfsYRgLJ15CAB_48ob_yQ

troyane, 2018-09-12:
@Fourdee shouldn't we open this issue?