MichaIng/DietPi #1211: Allo GUI | Inability to login after update

Fourdee, 2017-10-25:
> https://community.roonlabs.com/t/dietpi-allo-web-gui-now-available/32420/16?u=dan_knight

Fourdee, 2017-10-25:
If you experience this issue, please see the following post for more support information:
https://community.roonlabs.com/t/dietpi-allo-web-gui-now-available/32420/21?u=dan_knight

Fourdee, 2017-10-26:
Unable to replicate my end, waiting for user sent bug reports to investigate.

blogabe, 2017-11-26:
I'm having this issue and sent in a bug report: f4d3091c-b61c-4d4f-9539-54f8b48c935f-0

blogabe, 2017-11-26:
Forgot to mention, I'm logging into the DigiOne using SSH rather than directly plugging in a monitor/keyboard into the Pi.

Fourdee, 2017-11-26:
@blogabe 

Many thanks for sending the bug report 👍 

Can you please run the following commands to clear the MYSQL DB, and reinstall it:
```
systemctl start mysql
mysqladmin -u root -pdietpi drop allo_db -f
dietpi-software reinstall 160
```

After reboot, please try to login to the web interface.

blogabe, 2017-11-26:
Unfortunately that didn't work either.  I also tried doing it a second way where rather than reinstalling, i manually uninstall 160 and 159 then install 160.  Resent bug report.  That code is f4d3091c-b61c-4d4f-9539-54f8b48c935f-1

I changed the default password from "dietpi" to something else.  But I don't know if that impacts the mysql credentials so I used the second line as is w/ `-pdietpi`

Thanks.

Fourdee, 2017-11-27:
@blogabe 

> I changed the default password from "dietpi" to something else. But I don't know if that impacts the mysql credentials so I used the second line as is w/ -pdietpi

Currently the web interface uses the SSH2 PHP library, which requires hard coded password. Changing the root password will render the web interface dis-functional.

If you change the ```root``` password back to ```dietpi``` it should resolve the issue.
Lets us know if this resolves the issue, i'll then change this ticket accordingly, so we can investigate alternative PHP SSH library that does not require hard coded password.

blogabe, 2017-11-27:
Ok.  Appreciate you investigating other libraries that remove this restriction.

I changed the root password back to `dietpi` and reran the above three steps.  I can confirm that this does work and after the reboot.  I'm able to log into the Allo Web GUI.

Thank you!

Fourdee, 2018-04-05:
Marking as closed due to the following:
- https://github.com/Fourdee/DietPi/issues/1211#issuecomment-347245505
- Lack of time to invest/research in rewriting the web interface with a different SSH library, that allows for password changes.

MichaIng, 2018-06-07:
@blogabe 
We did not change something about Allow GUI within the passwords rework, so I think it will be not resolved. But let me check for v6.10

Maybe we can find a way to change the hard coded password within Allo GUI as well, when changing root/dietpi user pass via the new method.

Btw.: Actually mysql access via root user does not need any password any more with MariaDB. If you are logged in with unix root user, login into mysql root user works without any password need due to unix_socket authentication method. If you are logged in with another unix user, you will have no change to log into mysql root user 😉.

blogabe, 2018-06-08:
@MichaIng yeah, wouldn't have thought so as Dan alluded to moving away from the current configuration.  6.10 may be aggressive as well, but I think it's an important enough issue to remain open irrespective of when it gets implemented.  Anyhow, thank you for your help.

Fourdee, 2018-06-12:
I'll mark this as closed in favor of: https://github.com/Fourdee/DietPi/issues/1841

closed 2018-06-12 by Fourdee