MichaIng/DietPi #1213: dietpi-letsencrypt fails

TheRaspyDev, 2017-10-27:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 157
- SBC Device (EG: RPi 3)? RPi 2
- Power supply used (EG: 5v 1A RAVpower)? 5v 2.4A
- SD card used (EG: Sandisk ultra)? Kingston 16GB
- Distro (EG: Jessie) | ```uname -a```? Stretch --> Linux DietPi 4.9.52-v7+ #2 SMP Mon Oct 2 23:27:32 UTC 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title? dietpi-letsencrypt
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:
dietpi-letsencrypt should run and a certificate should be created

#### Actual behaviour:
Error when running dietpi-letsencrypt ->
/DietPi/dietpi/dietpi-letsencrypt: line 153: ./certbot-auto: No such file or directory

#### Steps to reproduce:
- Write image "DietPi_v156_RPi-armv6-(Stretch).img" to SD card
- Boot RPi
- Configure RPi with dietpi-config (only locale and timezone was configured)
- Reboot
- Install no-ip with dietpi-software -> works
- Reboot
- Install LASP with dietpi-software -> works
- Reboot
- Install certbot with dietpi-software -> no error message but doesn't seem to install correctly
- Run dietpi-letsencrypt gives the following error: /DietPi/dietpi/dietpi-letsencrypt: line 153: ./certbot-auto: No such file or directory

#### Did you submit a ```dietpi-bugreport```?
c801c4ce-964b-40bd-8108-276a6d706129-0

#### Extra details:
The file certbot-auto is missing on the file system. So my assumption is that the installation of certbot didn't work correctly.
I didn't install nginx webserver oon my own. This service was enabled/installed during installation of certbot.
I tried it several times from a fresh install but always get the same error.

Full output of dietpi-letsencrypt:

 DietPi-LetsEncrypt
─────────────────────────────────────────────────────
 Mode: Running cert
 Please wait...

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start
 Please wait...

 [Ok] cron
 [Ok] apache2
 [Failed] nginx
 [Ok] noip2

 DietPi-Process_Tool
─────────────────────────────────────────────────────
 Mode: Apply
 Please wait...

 Cron (948)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 DHCP Client (721)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Dropbear (401)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Dropbear (1967)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 NoIp (1057)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Webserver Apache2 (978)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Webserver Apache2 (1006)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Webserver Apache2 (1008)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Webserver Apache2 (1009)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0
 Webserver Apache2 (1011)
 [Ok] Nice      0
 [Ok] Affinity  0-3
 [Ok] Scheduler SCHED_OTHER 0

 [Ok] DietPi-Process Tool | Completed

 [Ok] Apache2 webserver detected
/DietPi/dietpi/dietpi-letsencrypt: line 153: ./certbot-auto: No such file or directory

Press any key to continue...

Any help will be appreciated.

Cheers
Maurus

Fourdee, 2017-10-27:
@TheRaspyDev 

Thanks for the report, we'll try to replicate.

> Configure RPi with dietpi-config (only locale and timezone was configured)

Can you confirm which locale you applied to the system?

Fourdee, 2017-10-27:
Reference testing branch changes: https://github.com/Fourdee/DietPi/pull/1206

TheRaspyDev, 2017-10-27:
Thanks for the fast reply. Just applied the default en-GB locales.

MichaIng, 2017-10-29:
Just found the reason for this error and corrected it by PR: https://github.com/Fourdee/DietPi/pull/1217#issuecomment-340265797

@TheRaspyDev 
New apt packages for certbot on stretch are installed since exactly v157, but it was forgotten to let `dietpi-letsencrypt` choose the new binary for every web server.

To fix this you could get the new versions of `dietpi-letsencrypt` and the related part of `dietpi-software` from the PR above. Or lunch the certbot binary with automatic apache configuration by yourself: `certbot --apache --rsa-key-size 4096`.
The next DietPi release will also have this fixed of course.

Fourdee, 2017-10-30:
@MichaIng

> Just found the reason for this error and corrected it by PR: #1217 (comment)

Excellent, thanks.

PR merged. As this resolves the initial issue in this ticket, i'll mark as closed.

TheRaspyDev, 2017-10-30:
Great stuff, thx a lot.
I'll try to rerun certbot again tomorrow.

TheRaspyDev, 2017-11-03:
Works now like a charm.
Thx for your work!

closed 2017-11-03 by TheRaspyDev