MichaIng/DietPi #1218: DietPi-Update installed unexpected software

mpember, 2017-10-30:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
Upgrading from 156 to 157

- SBC Device (EG: RPi 3)?
Raspberry Pi 3

- Power supply used (EG: 5v 1A RAVpower)?
Office Pi Foundation 5V 2.5A

- SD card used (EG: Sandisk ultra)?
Sandisk Ultra 8GB

- Distro (EG: Jessie) | ```uname -a```?
Linux google 4.9.52-v7+ #2 SMP Mon Oct 2 23:27:32 UTC 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
DietPi-Update

- Can this issue be replicated on a fresh installation of DietPi?
Not sure

#### Expected behaviour:

A clean upgrade of existing software

#### Actual behaviour:

A bunch of software packages that were previously not selected suddenly installed themselves from scratch.

#### Steps to reproduce:

run **dietpi-update**

#### Did you submit a ```dietpi-bugreport```?

N/A

#### Extra details:

[term.log](https://github.com/Fourdee/DietPi/files/1425673/term.log)
[dietpi-software_apt.log](https://github.com/Fourdee/DietPi/files/1425674/dietpi-software_apt.log)
[dpkg.log](https://github.com/Fourdee/DietPi/files/1425675/dpkg.log)
[history.log](https://github.com/Fourdee/DietPi/files/1425676/history.log)

Since I have the logs set to only remain for 1hr, I don't have any detailed logs.  Here are the logs that grabbed copies of when I first noticed the issue.

I have since run **dietpi-software** and deselected the packages.  Some of the lingering folders / files were manually removed when the output from **dpkg --purge** indicated that the folders were unable to be automatically removed.

Fourdee, 2017-10-30:
@mpember

Thanks for the report

Looks like a possible bug in our patcher, patch assumes your running the Allo GUI image. Can you please confirm this is the stock DietPi image, and not http://dietpi.com/phpbb/viewtopic.php?f=8&t=2317#p8868?

Fourdee, 2017-10-30:
Unable to replicate on Stock DietPi image. Assuming user is running Allo GUI, waiting for confirmation.
https://github.com/Fourdee/DietPi/blob/master/CHANGELOG.txt#L41

mpember, 2017-10-30:
> Can you please confirm this is the stock DietPi image

It is a stock install that started with v150 (downloaded in August) and has been upgraded since then.  While I have installed a few other software packages, The only custom software on these units is the Google Assistant code and a few small scripts to provide functionality required by the Google AIY software.

Since it only appears to have happened on one of the two AIY units, I can't be exactly sure what is different between the two.

Due to the fact that I was doing the update at about 1am, I was more focused on simply getting it cleaned up and not so much focused on documenting the process I went through to clean up the install.

Removing the apps (using **dietpi-software**) seemed to clean up all trace of the rogue installs.  I also ran `dpkg --list |grep "^rc" | cut -d " " -f 3 | xargs sudo dpkg --purge` to remove all trace of the additional packages.

mpember, 2017-10-30:
I don't recall installing any 'optimized' software after the initial install.  The programs installed via the dietpi.txt setup were:

AUTO_DietpiSoftware_Install_ID=21       #will install Vim-Tiny
AUTO_DietpiSoftware_Install_ID=5        #will install ALSA sound system
AUTO_DietpiSoftware_Install_ID=130      #will install Python Pip installer
AUTO_DietpiSoftware_Install_ID=69       #will install RPi.GPIO libraries
AUTO_DietpiSoftware_Install_ID=70       #will install WiringPi libraries

The rest of the software on the unit was installed using the regular **apt-get** method.  In the case of the Google AIY code, that was done with **git**.

Fourdee, 2017-10-30:
@mpember 

Thanks

I've been unable to replicate with:
```
AUTO_DietpiSoftware_Install_ID=21       #will install Vim-Tiny
AUTO_DietpiSoftware_Install_ID=5        #will install ALSA sound system
AUTO_DietpiSoftware_Install_ID=130      #will install Python Pip installer
AUTO_DietpiSoftware_Install_ID=69       #will install RPi.GPIO libraries
AUTO_DietpiSoftware_Install_ID=70       #will install WiringPi libraries
```

Strange, the only reason it would be installed if the following returns ```1```:
```
cat /DietPi/dietpi/.installed | grep -ci -m1 '^aSOFTWARE_INSTALL_STATE\[159\]=2'
```

**Can we get the dump of ```dietpi-update```?**
```
cat /etc/dietpi/logs/dietpi-update.log
```

mpember, 2017-10-31:
> cat /DietPi/dietpi/.installed | grep -ci -m1 '^aSOFTWARE_INSTALL_STATE\[159\]=2'

Returns 0

> cat /etc/dietpi/logs/dietpi-update.log

This file does not exist on either of my Google units.  The directory is empty.

Sync I have the ram logs set to be cleared every hour, all system logs have now been lost into the ether.

Fourdee, 2017-10-31:
@mpember 

👍 Thanks

Apologies, ```dietpi-update``` logging was only added recently (v157), updating from v156 would not have this feature:
https://github.com/Fourdee/DietPi/commit/cd1a3b59cd200d296b97019f9aff3ec56d930011#diff-c613a85da508fb885b67c34f9661e243

Following is not part of RAMlog:
```
/etc/dietpi/logs
```

Not sure where to go from here regarding this issue. As we lack the ```dietpi-update``` log (due to it not existing in v156 <=), we can't see the reason for these packages to be installed.
I've also been unable to replicate this issue currently.

I'll leave the ticket open for now, just in case we receive further reports from other users.

Fourdee, 2017-11-10:
Unable to replicate. I'll mark this as closed for now as we require the update logs (added to v157 https://github.com/Fourdee/DietPi/issues/1218#issuecomment-340780688)

Please reopen if required.

closed 2017-11-10 by Fourdee