MichaIng/DietPi #1222: Run `Dietpi-update` after reboot my waln0 can`t work!

1021683053, 2017-11-02:
Run `Dietpi-update` after reboot my waln0 can`t work.

i use :
```
1.Download rpi-source [1]
sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update
2. Install bc (required by rpi-source)
sudo apt-get install bc
3. Run rpi-source to get the required kernel headers
rpi-source
4. Download the driver
git clone https://github.com/astsam/rtl8812au.git
5. Make the driver
make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm
6. Install the driver
sudo make install
sudo modprobe 8812au
```
error with:
```
    modprobe: ERROR: could not insert '8812au': Invalid argument
```

Update before that  can use!

MichaIng, 2017-11-02:
Hmm, since newest DietPi also includes a kernel version bump to 4.9.52, your error might be related to that.

Just in case could you try to rpi-update to the newest edge version again? Do system backup in any case before updating kernel or DietPi (dietpi-backup).

rpi-update can be also done via dietpi-config inside advanced.

joaofl, 2017-11-15:
I had the same issue, but I think the issue is simpler than some kernel issue.

This happens since Dietpi included a service to monitor the wireless connections and reconnect if connection is lost (every 10 seconds). So, once you kill this service, the connection is lost.

So, whenever one tries to update, first thing Dietpi does it to stop all its services, including this one, which kills the connection and prevents the update/upgrade.

There should be some sort of whitelist, with the services that should not be stopped. I did it manually in mine, but it gets back to the original every time I upgrade...

Fourdee, 2017-11-19:
DietPi kernel does not provide headers/source in our binaries (reduces filesize/load on server).

To compile drivers with ```rpi-source```, you'll need to use ```rpi-update``` to revert to stock kernel as per @MichaIng suggestion.

I'll mark this as closed.

closed 2017-11-19 by Fourdee