MichaIng/DietPi #1223: DietPi-Software | moOde

Fourdee, 2017-11-02:
> http://moodeaudio.org/

Now with FOSS license 👍

sfb2, 2017-11-02:
_soo wonderful, can´t believe its true_ ;-)_
(and on moodeaudio.org is now only this FREE 4.0 version..absolute wonderful too)..
just in the minute i try to set up; somany thx again **_"Fourdee"_** for leaving this info here !!

Fourdee, 2017-11-04:
@sfb2 

> (btw.:got it here to work; took about 3 and half hours ;-), but really works, absolute fabulous "tut"/recipe with no errors/absolute straight-forward->fantastic work by Tim/Moodeaudio !! )

Yep, Tim has done an amazing job with the build-recipe 👍

Fourdee, 2017-11-05:
### Notes

#### Step 1:
- Get everything working at a basic (as expected) level

#### Step 2:
- Integrate into ```dietpi-software```

#### Would be nice:
- Support webserver selection (eg: lighttpd/apache2/nginx). Currently Nginx with .conf. Will require site.confs for all servers.

Fourdee, 2017-11-05:
### Notes:
- DietPi ```/etc/rc.local``` overwritten by moOde, integrate both.

Fourdee, 2017-11-05:
[image]

TheOldPresbyope, 2017-11-06:
Glad to see this push forward. You have an eagerly awaiting audience!

Count me as one willing to be a crashtest dummy for DietPi/moOde on RPi and Odroid not to mention other SBCs and VMs. I've been puttering with moOde on RPi since r2.6 and have used DietPi in several appliances. No problem with the Linux CLI.

Your Step 2: It would not trouble me if DietPi were limited to a "moOde only" install but I can understand the appeal of multi-app installs.

Your Optionals: it would not trouble me if DietPi were to support only MPD for the time being.

Your "Would be nice" list:  1) it isn't clear why supporting different http servers is desirable but that's your call.  2) Most moOde users would be ecstatic to see a pre-built binary but it is not an essential to a us early adopters.

rikardo1979, 2017-11-06:
nice to see you working on this. I am an long time MoOde user.
You can count me in for crash testing :P

sfb2, 2017-11-06:
PS.: I´m/was a Moode-fan too..right know i´m testing Moode V4B7 with a RPI3+an ES9028Q2M-board..and i must say if Moode is 85%, Dietpi is still 90%..;-)..for me Dietpi is more "nuancable"/has a different sound..and i guess one reason is because its so slim/lightweight, not so much background-tasks/processes and therefore the tune/sound is clearer, not soo much load on the CPU/PSU and so on..
and so the main reason, spoken as an "badass" audiophile, is why i like this lightweight-distros (as Dietpi) so much: it has a more cleaner/straighter sound..more "hardware-near"..clear, isnt it ? ;-)

moodeaudio, 2017-11-09:
Hi Dan,

Looking good!

-Tim

thechildofroth, 2017-11-10:
I'd love for the 'optionals' to be in there too. 

But I do think there ought to be some consideration about how it interacts with DP as a whole (many of the technologies are already in DP) - let's face it getting MPD up and running with DP was never all that hard anyway! 
So I think that if (for instance) one installs sharepoint sync using Moode, it should turn up ticked in the DP-software menu. Also, if it's then uninstalled from DP-software it should come out of the moode options too.

This may well be what's already in mind but just throwing some ideas in the pot! Looking forward to testing on my NanoPi Neo.

Fourdee, 2017-11-10:
@thechildofroth 

Thanks for the suggestions Roth, appreciate them as always. I will take them on board 👍 

Our initial goal is to provide a moOde install that stays true to the build-recipe Tim created, and, supports official updates. We don't want to deviate too far away from this, without ensuring full support across both DP and moOde.
Once thats completed and stable, we can look at tweaking the DP side.

Tim is coding support for our DietPi-RAMdisk (```/DietPi/config.txt```), once completed, should be a few days before we can open testing.

wojtekkutylacc, 2017-11-12:
I am eagerly awaiting this becoming available as an image. Great work, people, keep it coming :)

Fourdee, 2017-11-15:
## Automated image installation of moOde:
Currently in testing, under RPi Stretch.

#### Steps to test the DietPi + moOde installation:
- Download the DietPi **Stretch** image and write to SD card: http://dietpi.com/downloads/testing/DietPi_RPi-armv6-(Stretch).7z 
- Download http://dietpi.com/downloads/conf/moode/dietpi.txt and save it to the SD card ```/boot/dietpi.txt```
- Connect Ethernet.
- Connect HDMI if you want to see the installation process.
- Eject SD card, pop into RPi and power on

TheOldPresbyope, 2017-11-16:
Followed your steps to test...on an RPi3B with ethernet connection

Forgot to time the installation but Moode client has just popped up and I can ssh into the host as well. Sweet.

I won't have time to exercise the installation until tomorrow, but this is very encouraging. Well done.

TheOldPresbyope, 2017-11-16:
DietPi_v156_RPi-armv6-(Stretch).img failed to install on an RPi0W with ethernet connection. 

After waiting a time, the nginx server came up with the default page. The install was left to cook while I did other things. Hours later, no change. Powered off and mounted the uSD card in a Linux Mint system. The (root)/var/log directory is nearly bare. I found DietPi-Automation.log in the (root)/root directory:
--------------------
$ sudo cat root/DietPi-Automation.log
Error: Apt-get has failed:
 - Apt string = rpi-update mpc memcached php-memcache bs2b-ladspa libbs2b0 libasound2-plugin-equal telnet automake sysstat tcpdump shellinabox udisks-glue exfat-fuse in
otify-tools libav-tools
 - Error code = 100
 - Software Title = moOde: audiophile-quality music playback
 - Apt log = /var/log/dietpi-software_apt.log

Please report this issue to DietPi (INCLUDING THE APT LOG) so it can be investigated.
--------------------

There is no (root)/var/log/dietpi-software_apt.log file.

The apt string in the error message relates to STEP 3.2 of Tim's build script.

The apt error code 100 is often associated with dpkg being either missing or marked with the wrong permissions, but
--------------------
$ ls -l usr/bin/dpkg
-rwxr-xr-x 1 root root 227056 May 17  2017 usr/bin/dpkg
---------------------

Feature request: add more logging activity so we can more easily debug failing headless installs.

sfb2, 2017-11-16:
_Just for info/to confirm:_
WORKS !! here on a RPI3 + (ES9028Q2M)..
absolut straight-forward, no issues, about 35-45min (most part time for MPD-compiling); unbelievable, such a wonderful script..so many thx !! ;-)
(PS.: (with the gear here->soundwise: not so fine/good/accentuated as normal DietPi+MPD (still the "best"), but also with no clicks/pops while volume changing in the UI in opposite to the original MoodeV4Beta9 here->think its about the buffer size in MPD..(but that depends all on your I2S/USB-devices i guess)..
and another thing: i can´t get with this hifiberry-dac dtoverlay above 192khz (via cat/....)..it works in Dietpi 1.54v but afterwards not anymore..)
but thx and many greets !!

bobold, 2017-11-17:
Installed and working on a Pi-1B. 
Had a monitor attached to hdmi and confirmed start of installation. However, when I checked 3 hours later it had blanked and I had to assume the install had finished. 
How about a 'completed, please reboot' message ?
Yes, I could have ssh-in and checked but the 2 times I tried to install previously the install stopped once I had ssh-in... means you've just got to guess it's happened or being going long enough....

sfb2, 2017-11-17:
> How about a 'completed, please reboot' message ?
yes..thx bobold..that was an "issue" here too..thought also about if its "stucked" or not..something like "Installed successfully..please reboot->(Enter)" or similar..;-)

Fourdee, 2017-11-17:
Thanks guys for testing and the feedback.

> RPi0W | Error: Apt-get has failed: | There is no (root)/var/log/dietpi-software_apt.log file.

@TheOldPresbyope 

Most likely cleared by DietPi-RAMlog clear (1 hour cycle, and during system reboots http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68). I'll move the log file to ```/etc/dietpi/logs```.

I've not tested RPI0/1 yet with this install, looks limited to that device/ARMv6. I'll see if we can replicate the issue our end.

> How about a 'completed, please reboot' message ?

@bobold 

Great suggestion 👍 
I'll get that added in.

> (most part time for MPD-compiling)

We need to compile and offer pre-build .debs anyway for MPD, i'll include these in the next install. Should vastly reduce the install time: https://github.com/Fourdee/DietPi/issues/1236

Fourdee, 2017-11-17:
#### Notes:
- Offer prebuild .debs of MPD (reduce install time): https://github.com/Fourdee/DietPi/issues/1236
- Inform user with dietpi-banner, when system is ready for the 1st time
- 🈴 RPI Zero W / ARMv6: Fails during APT: https://github.com/Fourdee/DietPi/issues/1223#issuecomment-345063301

Fourdee, 2017-11-19:
Test install with:
- RPi Zero W 🈯️ 

Unable to replicate the APT error @TheOldPresbyope. Please can you re-test?

[image]

mho-ohm, 2017-11-19:
RPi 2B here
Following install, I'm presented with a login request. I login, reboot and am asked to fill in my default credentials again. The Moode ascii art screen flashes by and I'm presented with the command line, with options to run dietpi-launcher, dietpi-config, dietpi-software, htop and cpu. No option to start the moode client. Am I missing something? 

The /etc/dietpi/logs/dietpi-software_apt.log file is empty. 

Thanks for adopting moode audio on dietpi!

Fourdee, 2017-11-19:
@mho-ohm 

> No option to start the moode client. Am I missing something?

Should be running and accessible via web, try accessing it:
- http://dietpi.local
- http://192.168.0.100 (change ip as needed)

mho-ohm, 2017-11-19:
Thank you, sir! Accessed it via my web browser on another computer on the same LAN. (This former volumio user should have known better.)

TheOldPresbyope, 2017-11-19:
About to leave for a family party. I will retest with RPi0W in morning. Thanks for your efforts.

rikardo1979, 2017-11-20:
great stuff @Fourdee ;) 
Just installed with no problem on my RaspDAC(RPi2 I-Sabre 9023 TXCO) 
Can access the MoOde webgui and adjust some settings. Some parts does not work and looks like hanging when trying to adjust, like adding sources, trying to turn on UPnP or DLNA. 

Looking forward to new updates/improvements 

Thank you

rikardo1979, 2017-11-20:
just tested playback from NAS (LAN). Not smooth, a lot of skipping every few seconds. htop shows nothing unusual, very low usage 
[image]

Fourdee, 2017-11-20:
@rikardo1979 

Thanks for testing 👍 

> Can access the MoOde webgui and adjust some settings. Some parts does not work and looks like hanging when trying to adjust, like adding sources.

Thanks, i'll try to replicate.

> Trying to turn on UPnP or DLNA but no go. So understand is still not implemented?

Yep, not currently implemented, it is planned as we want to offer the full moOde experience. However, MPD only for now.

> just tested playback from NAS (LAN). Not smooth, a lot of skipping every few seconds. htop shows nothing unusual, very low usage

Yep, looks fine, possibly due to low audio/period buffers in MPD. What are the current settings?

TheOldPresbyope, 2017-11-20:
Yes, now builds on my RPi0W. Don't know what happened the first time. I think the uSD card was ok but I used a new one just in case.

Also, my apology for the "shouting style" of my previous message. I should have previewed it. Then I would have noticed the github formatter had gone to town on what I thought was plain text but which inadvertently contained markup.

Next order of business: be able to run via WiFi instead of ethernet interface. Is there a clash with DietPi which prevents switching to WiFi via the moOde config page? When I tried setting WiFi credentials via moOde I could see from the command line that /etc/wpa_supplicant/wpa_supplicant.conf had been correctly populated. However, I rebooted with ethernet disconnected and the dietpi host wasn't visible to my AP. No joy when I tried reconnecting ethernet whether or not I rebooted again. I may have to rig up a gpio-to-tty connection so I can see what the RPi0W is doing.

Fourdee, 2017-11-20:
@TheOldPresbyope 

> Also, my apology

No need, its all good 👍 

> Next order of business: be able to run via WiFi instead of ethernet interface.

WiFi is not functional on our Stretch image. We are trying to find the cause: https://github.com/Fourdee/DietPi/issues/475#issue-171250292

TheOldPresbyope, 2017-11-20:
> WiFi is not functional on our Stretch image. We are trying to find the cause: #475 (comment)

Curious. It works fine when we build moOde directly on 2017-09-07-raspbian-stretch-lite.

k-plan, 2017-11-20:
@Fourdee 

```

 ───────────────────────────────────────
 DietPi     | 19:33 | Mon 20/11/17
 ───────────────────────────────────────
 V158       | RPi B (armv6l)
 ───────────────────────────────────────

root@RPi-B:~# cat /etc/debian_version
9.1

root@RPi-B:~# /DietPi/dietpi/dietpi-software install 168
 [Info] Checking RootFS R/W access. Please wait...
 [Ok] RootFS R/W access.
 [Info] Initializing database, please wait...
 [Info] Reading database, please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install
 Please wait...

 [Ok] Installing moOde: audiophile-quality music playback
 [Info] Detecting drives, please wait...

.....
.....

 DietPi-Software
-----------------------------------------------------
 Mode: Install software
 Please wait...

 moOde: audiophile-quality music playback
 [Info] Please wait...

 #               processing | D i e t  P i             #

 [Info] Testing connection to http://moodeaudio.org/downloads/beta/r40b8/rel-stretch.zip
 [Info] Max duration of 20 seconds, please wait...
 [Failed] Connection test | An issue has occured

 DietPi-Software
-----------------------------------------------------
 Mode: Optimize and configure software
 Please wait...

 [Info] Applying DietPi optimizations and configurations for RPi B (armv6l)
 [Info] Please wait...

 #               processing | D i e t  P i             #

--2017-11-20 19:19:19--  https://raw.githubusercontent.com/rlerdorf/opcache-status/master/opcache.php
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.112.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.112.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 22915 (22K) [text/plain]
Saving to: ‘/var/www/opcache.php’

/var/www/opcache.php               100%[===============================================================>]  22.38K  --.-KB/s    in 0.04s

2017-11-20 19:19:21 (502 KB/s) - ‘/var/www/opcache.php’ saved [22915/22915]

--2017-11-20 19:19:21--  https://github.com/krakjoe/apcu/raw/master/apc.php
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/krakjoe/apcu/master/apc.php [following]
--2017-11-20 19:19:22--  https://raw.githubusercontent.com/krakjoe/apcu/master/apc.php
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.112.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.112.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 38125 (37K) [text/plain]
Saving to: ‘/var/www/apc.php’

/var/www/apc.php                   100%[===============================================================>]  37.23K  --.-KB/s    in 0.07s

2017-11-20 19:19:23 (555 KB/s) - ‘/var/www/apc.php’ saved [38125/38125]

Added user root.

```
Download moOde!

[DOWNLOAD MOODE](http://moodeaudio.org/downloads/beta/rel-stretch-r40b9.zipl)
MD5 = 4dd1ace90cf34f146615a0e292d597ba

Release = moOde 4.0 BETA 9

http://moodeaudio.org/downloads/beta/rel-stretch-r40b9.zip

moodeaudio, 2017-11-20:
Hi @k-plan,

Just to confirm that the download links were changed a while back.

Beta zips will be under 
http://moodeaudio.org/downloads/beta/rel-stretch-rNNbN.zip

Production release will be under
http://moodeaudio.org/downloads/prod/rel-stretch-rNNN.zip

-Tim

k-plan, 2017-11-20:
Hi Tim,

thanks for your hint. :+1: 

Is it possible, you will leave some older versions online in the future?
Difficult to get always the last release by shell scripting.

Or can @Fourdee  do his own hosting?

cu
k-plan

chirohito, 2017-11-20:
I've had some problems mounting NAS (moOde always pointing to /mnt/NAS/sharedfoldername and dietpi default samba mount is /mnt/samba) since while parsing mount string moOde uses text from "Source name" textbox instead "Remote directory". It played whole afternoon without problems or audiable (at least for me) skipping. I've bought another microSD for future playing with moOde on dietpi.

I would like to greet moOde on dietpi!

P.S. My hardware is Rpi3B + Hifiberry DAC+ pro

k-plan, 2017-11-20:
Last testing for today:

 ```
DietPi-Software
-----------------------------------------------------
 Mode: Install software
 Please wait...

 moOde: audiophile-quality music playback
 [Info] Please wait...

 #               processing | D i e t  P i             #

 [Info] Testing connection to http://moodeaudio.org/downloads/beta/rel-stretch-r40b9.zip
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

--2017-11-20 20:47:05--  http://moodeaudio.org/downloads/beta/rel-stretch-r40b9.zip
Resolving moodeaudio.org (moodeaudio.org)... 23.235.199.139
Connecting to moodeaudio.org (moodeaudio.org)|23.235.199.139|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16836295 (16M) [application/zip]
Saving to: ‘package.zip’

package.zip                        100%[===============================================================>]  16.06M   637KB/s    in 30s

2017-11-20 20:47:36 (544 KB/s) - ‘package.zip’ saved [16836295/16836295]

Archive:  package.zip
   creating: rel-stretch/

E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/m/mesa/libgl1-mesa-dri_13.0.6-1+rpi2_armhf.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/m/mesa/libgl1-mesa-dri_13.0.6-1+rpi2_armhf.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
 
```

 ```
root@RPi-B:~# apt-get update --fix-missing
OK:1 http://mirror.netcologne.de/raspbian/raspbian stretch InRelease
OK:2 https://archive.raspberrypi.org/debian stretch InRelease
Paketlisten werden gelesen... Fertig

```
 
```
root@RPi-B:~# apt-get install -f
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 4 nicht aktualisiert.

```

martintenor, 2017-11-20:
Hi All,

I'm also one for seeing moOde on DietPi.
Thanks all for the great work.

I've installed the image on an rPi3b with a Terra-berry DAC2 (hifiberry-dac driver).
For NAS settings, I usually go with NFS, so I installed it via the dietpi-config, then added the source on moOde GUI. No dropout issues I can see during playback, even DSD files with converting on the fly to PCM.

I went ahead with installing UPnP :) as I use Lumin app from an iPad or iPhone. Aside from a couple of libraries missing, installation is as per the COMPONENT 6 of Tim's recipe. Activates from the moOde GUI. No issues so far playing same files as above. Hope you can get this implemented officially soon!

Regards

Fourdee, 2017-11-21:
@k-plan 

> E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/m/mesa/libgl1-mesa-dri_13.0.6-1+rpi2_armhf.deb  GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

Looks like a possible issue with your APT mirror:
```
http://mirror.netcologne.de/raspbian/raspbian stretch InRelease
```

Might be worth trying the official/default mirror.

Fourdee, 2017-11-21:
@chirohito 

> I've had some problems mounting NAS (moOde always pointing to /mnt/NAS/sharedfoldername and dietpi default samba mount is /mnt/samba) since while parsing mount string moOde uses text from "Source name" textbox instead "Remote directory". 

Thanks i'll check the NAS 👍

Fourdee, 2017-11-21:
@moodeaudio 

> Probably best to host the file at DietPi. That way @Fourdee has control over the link. 

Yep no worries, we got the bandwidth 👍

Fourdee, 2017-11-21:
ToDo:
- Test NAS: https://github.com/Fourdee/DietPi/issues/1223#issuecomment-345810269
- Host moOde .zip on dietpi.com

k-plan, 2017-11-21:
@Fourdee 

> Looks like a possible issue with APT mirror/connection/server issue at the time? I'll try to replicate today.

Hmm, possible. Start today totally completely from scratch.

And now it works! :smiley: 
[image]

Think, I totally screwed it up by a lot of testing.
Nice work. :+1:

sfb2, 2017-11-21:
> E: Failed to fetch https://archive.raspberrypi.org/debian/pool/main/m/mesa/libgl1-mesa-dri_13.0.6-1+rpi2_armhf.deb GnuTLS recv error (-110): The TLS connection was non-properly terminated.
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Looks like a possible issue with APT mirror/connection/server issue at the time? I'll try to replicate today."
had the same issue here..but with the beautiful dietpi-script i just rebooted after a while and after the wonderful "installed"-checks through the dietpi-scripts i was able to continue till finish..

> @moodeaudio

Probably best to host the file at DietPi. That way @Fourdee has control over the link.
Yep no worries, we got the bandwidth 

this i always wanted to mention since my beginning here, i have a fast server here in Germany too (50-60MB/s up and down) unlimited bandwith and about 5GB ;-) free and wanted it to "offer" (sorry for my bad english) to the dietpi-"community" for absolute free as mirror or somethig too, because i experienced also a little "low" downlad-rates at some packages..
as said at absolute free proposal, that would be mine "contribution" for the wonderful dietpi-community, its absolute bullet-proofed here since years (i´m hosting about 20 websites btw. on it) and i would like to use it more..
as said 10000%-percent bullet proofed (had inbetween 10000 visitors per day, 100 online user simultaneously and so on; no prob in handling this/was capable)..
So i would feel very honored if dietpi would use it as a mirror or something ;-)..
if not, also okay, many thx from here again for now ;-)

Fourdee, 2017-11-21:
@sfb2 

> as said at absolute free proposal, that would be mine "contribution" for the wonderful dietpi-community, 

Thats very kind of you, really appreciate it 👍 

For the moment, we are lucky enough to have sponsored hosted by MVS https://www.myvirtualserver.com/en
- 5TB/Month bandwidth

Should things change in the future, and the offer is still valid, i'll be in touch 😄

rikardo1979, 2017-11-22:
> Yep, looks fine, possibly due to low audio/period buffers in MPD. What are the current settings?

@Fourdee I have to test more but not easy to find some spare time now. 
I have not changed any settings for playback. Was trying to stream FLAC from local NAS(SMB) and all on LAN as I dont use wifi anywhere in my music system.

rh1017, 2017-11-22:
Many thanks Daniel for your efforts with moOde automated install on DietPI. Much appreciated.

I installed DietPi + moOde on a 8 GB Sandisk Utra microSDHC Class 1  using a RPi3b. The installation went smoothly and took under 40 minutes.

Sound quality with a USB interface  (Focusrite Scarlett Solo  2nd Generation) is excellent. Media source is external USB Sata HDD. No dropouts or other artifacts during uptime.

Looking forward to WiFi implementation and final tuning for I2S DAC Hats.

saudade428, 2017-11-23:
I installed in RPI 1B,it's cost almost 1hour...(sd card is too slow..)XDD.
But, now it's can't rescan usb... 
In ssh "df -h", /dev/sdb1 was exist.

mho-ohm, 2017-11-23:
Installed on a RPi2b with 2TB WD USB external hard drive attached. I haven't been able update or rescan the source and "df -h /dev/sda3" returns:
/dev/sda3       1.9T  951G  912G  52% /mnt/9ba037ef-a4c9-3093-a212-a9856378bfa8

Fourdee, 2017-11-25:
@saudade428 
@mho-ohm 
Thanks for the reports with regards to USB drives. I'll investigate how moOde picks up USB drives, see if we can resolve.

moodeaudio, 2017-11-25:
Hi @Fourdee,

USB drives are auto-mounted by udisks-glue to /media directory using the DISK_LABLE

/media/DISK_LABEL

Also a samba share block will be created in /etc/samba/smb.conf

[DISK_LABEL]
path = /media/DISK_LABEL
read only = No
guest ok = Yes

-Tim

Fourdee, 2017-11-25:
@moodeaudio 
Thanks Tim 👍 

DietPi automatically mounts USB drives to ```/mnt/UUID_OF_PARTITION``` via ```/etc/fstab```. This would override ```udisks-glue ```.

Options:
- Disable ```dietpi-drive_manager``` auto mounting in ```/etc/fstab```
- User manually symlink (not ideal)
- Change ```mpd.conf``` to support ```/mnt``` (and all its sub-directories) as the music data location.

chirohito, 2017-11-30:
Volume control is not updated properly. For example when I turn volume up on M.A.L.P. app on smartphone ad open moode on PC in browser volume is still on last value updated through that same browser. Page refresh does not help.

BR!

TheOldPresbyope, 2017-12-03:
@chirohito 

The behavior you report doesn't concern the integration of moOde into DietPi. Rather it is the consequence of the design of moOde itself. You can take it up with the moOde developer, Tim Curtis, on the moOde thread on diyaudio.com

http://www.diyaudio.com/forums/pc-based/271811-moode-audio-player-raspberry-pi.html

Cheers

kaytata, 2017-12-07:
With rp3b + Allo Kali + Piano 2.1dac Moode works very well in Dual Mono mode. Could not make it work in subwoofer mode though. either 2.1 or 2.2. When setting is applied, it automatically switches back to dual mono mode. Any thoughts?

Major difference with volumio I find is that the discovery of my attached 2tb external hd is super quick, a couple of minutes compared to 4-5 hours on volumio.

moodeaudio, 2017-12-07:
Hi @kaytata,

The Allo Piano 2.1 firmware files are required to enable Subwoofer 2.1 and 2.2 modes to function. These files apparantly were not accepted into mainline Linux kernel but are available in an Allo Git repo. Below is the procedure I use to install them.

cd ~
wget https://github.com/allocom/piano-firmware/archive/master.zip
sudo unzip master.zip 
sudo rm ./master.zip
sudo cp -r ./piano-firmware-master/lib/firmware/allo /lib/firmware
sudo rm -rf ./piano-firmware-master
sudo reboot

-Tim

kaytata, 2017-12-07:
@moodeaudio 
Thanks Tim for the solution. Works like a charm. Cheers.

Fourdee, 2017-12-07:
@moodeaudio 
Thanks Tim,

This should of been in our RPi kernel package, i've added it back in, fixed for new installations.

kaytata, 2017-12-20:
Hi,
I had installed using fourdee's Automated image installation of moOde:
The moode version installed was beta 9. I noticed some hanging. 
Tried to update the version to beta12 using the interface. It says moode is up to date.

Is there a simple way to update the moode software to beta12 using SSH?

Cheers,

martintenor, 2017-12-21:
Hi,

Just installed moOde 4 beta 9 using the automated script.
I see that upmpdcli is included, however it fails to load due to below.
I exclusively use Lumin app to play songs (I don't like to mount my NAS) so I would appreciate if this could be solved.

Dec 21 08:53:49 DietPi systemd[2502]: upmpdcli.service: Failed at step EXEC spawning /usr/bin/upmpdcli: No such file or directory

youpilai, 2017-12-23:
Hi, 

First thanks for your effort to make it happens

I wonder, is this will work on Sparky one day ? i have an ALLO Piano DAC 2.1 + Kali on a Sparky board and wanted to try Moode
I also have one ALLO Piano 1 that i can try with a RPi2, but have only one Kali board for now
For testing purpose, i also have a RPiZero with IQAudio Pi-DACZero if needed (i'll have to buy more SD cards...)

[12 more replies]

closed 2018-06-30 by Fourdee