MichaIng/DietPi #1224: orange pi Zero 2+ H5 - Wifi broken after installing lxde

kurtjcu, 2017-11-04:
## Creating a bug report/issue:
Fresh Dietpi install
autoupgrades to 157 (cannot disable updates in dietpi.txt)
install lxde then wifi adapter is unable to be found (onboard)

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version``` = 157
- SBC Device (EG: RPi 3) = orangepi zero 2+ h5
- Power supply used (EG: 5v 1A RAVpower) = 5v 3amp and samsung charger tested
- SD card used (EG: Sandisk ultra) =transcend
- Distro (EG: Jessie) | ```uname -a``` = jesse

#### Additional Information (if applicable):
- Software title = lxde
- Can this issue be replicated on a fresh installation of DietPi = yes

#### Expected behaviour:
should keep the old wifi settings and continue to function.

#### Actual behaviour:
wifi adapter is unable to be found

#### Steps to reproduce:
select lxde to be installed from dietpi-software

#### Did you submit a ```dietpi-bugreport```?
no

#### Extra details:
The wifi driver seems a little buggy before installing lxde i.e. unable to scan( crashes and drops back to dietpi-config).

kurtjcu, 2017-11-04:
Just confirmed that it is not an issue with LXDE, wifi is broken in version 157 
"no supported wifi hardware found." when using dietpi-config after upgrading to 157 without installing any additional software on a fresh install.

kurtjcu, 2017-11-04:
Problem is not with the update 157 but occurs with apt-get update && upgrade.

kurtjcu, 2017-11-05:
Currently have a working version by commenting out all the apt-get upgrade calls from within dietpi-config and dietpi-software scripts

Fourdee, 2018-02-22:
Support and image for this device has been dropped.

closed 2018-02-22 by Fourdee