MichaIng/DietPi #1225: DietPi-Software | Owncloud URL 404

Fourdee, 2017-11-04:
>

molnarti, 2017-11-08:
i would rather suggest to drop owncloud support altogether, nextcloud is a better alternative in each way. rather focus efforts where they can bring more benefits.

MichaIng, 2017-11-09:
I just installed owncloud on VM Stretch with Lighttpy and faced several problems:

First of all about installation part: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L4005
- We use fix Debian 8 repository. Exactly since the last minor update (10.0.3), there is a Debian 9 repository available that should be used in case of Stretch: https://download.owncloud.org/download/repositories/stable/Debian_9.0/
- Also exactly since last minor update the repo does not contain the `owncloud` meta package anymore, which in the past installed the actual `owncloud-files` and `owncloud-deps` containing apache and mysql. Now just `owncloud-files` is available (Stretch AND Jessie), so we need to take this. This is actually the better solution to allow different databases and webservers. Okay it is also possible to set dependencies that one of several options is possible, but okay...
- After this changes owncloud should be installed correctly and available via browser.

Second the configuration: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L8943
- DietPi links the `owncloud/data` folder at the default place but symlinks it to DietPi userdata. This is okay as long as Apache is used and thus `.htaccess` is in place to block direct access to data folder.
- On Nginx the created owncloud config also blocks access: https://github.com/Fourdee/DietPi/blob/master/dietpi/conf/nginx.sites-dietpi.owncloud.config#L5
- But on Lighttpd, DietPis default webserver, nothing is done to block access to data folder.
- Thus I highly recommend to completely move data folder to DietPi userdata and of course tell owncloud, that it's there. I guess it was for comport to create this symlink solution where users can just leave the web ui installation mask as it is. BUt anyway they needed to create the database and user by themself.
- As well as for Nextcloud, also ownCloud supports the command line installation via occ command: https://doc.owncloud.org/server/latest/admin_manual/configuration/server/occ_command.html#command-line-installation
We can just use this to automatically create database, user and correct data folder location.

Otherwise ownCloud works fine. The fixes are not too hard, as we can copy much from Nextcloud installation. Also all the modules/webserver configs can be just copied. All still works the same for both solutions.

@molnarti 
I would still leave the choice for users. I will check if not everything can be handled the same for both, besides the archive installation itself.

Fourdee, 2017-11-09:
Confirmed, appears the ```.deb``` installation has changed, owncloud is now located in:
```
/usr/share/owncloud
```

Fourdee, 2017-11-09:
Ok, i'll send PR to fix the install, however, 
@MichaIng has made some good points, further investigation for improved installation is required, as a new ticket.

Fourdee, 2017-11-09:
Tested fix with Odroid C2. Install working correctly.

Will mark this as closed. Ticket for improved installation: https://github.com/Fourdee/DietPi/issues/1229

closed 2017-11-09 by Fourdee

Fourdee, 2017-11-10:
> i would rather suggest to drop owncloud support altogether, nextcloud is a better alternative in each way. rather focus efforts where they can bring more benefits.

@molnarti 

Whilst I agree NextCloud is generally better. We do like to offer choice in DietPi (where possible) and let the user decide.