MichaIng/DietPi #1227: [VM] DietPi-Config | Change terminal size/resolution on virtual machines

MichaIng, 2017-11-05:
I was annoyed by the tiny 640x480 terminal on VirtualBox images. Searching and trying around for a while I finally found a solution that works to adjust it: https://forums.virtualbox.org/viewtopic.php?f=3&t=33214#p152720
> echo "GRUB_GFXPAYLOAD_LINUX=1280x1024x8" >> /etc/default/grub
> update-grub
> reboot

as an example.

Using this we could allow display settings via `dietpi-config` also on VMs.

Fourdee, 2017-11-10:
@MichaIng 

Yep, its a little small. Personally I use solely SSH.

Been a while since I last ran our VM, if I remember correctly, in VMware, you can define the screen resolution in the VMware GUI, prior to running the VM.

I'll verify when I get round to https://github.com/Fourdee/DietPi/issues/1219

MichaIng, 2017-11-10:
Yeah, SSH is actually easier choice, at least for testing cases, as it enables copy/paste in headless terminal. Horrible that there is no solution on VirtualBox for this.

On VirtualBox I also didn't find any other possibility to adjust screen size, so there it is a way and does not hurt. Should be checked if there are interferes with vmware, if you choose screen size in the front.

MichaIng, 2018-01-26:
Sometimes it is still helpful to access the VM terminal directly. Priority low, but I would still offer this by times.

@Fourdee 
Did you check, if adjustment is possible in VMware? Would drop priority again 😆.

To enable this, we need to reliably find the supported resolutions of the "graphic card". The best solution I found here: https://wiki.ubuntuusers.de/GRUB_2/Aussehen/#Aufloesung
As we do not have access to GRUB console, we would need to go the second way:
```
G_AGI hwinfo
hwinfo --framebuffer | grep 'Mode ' | awk '{print $3"x"$5}'
```
This e.g. gives the strings we would need to add to `GRUB_GFXPAYLOAD_LINUX` and optional `GRUB_GFXMODE` (resolution of grub menu itself). Color depth ($5) is also optional, can be left out.

But on the other hand, at least all VirtualBox VMs will offer the same here. If VMware does not offer just other exotic resolutions, we could just go with a fixed list like 640x480, 1024x768, 1152x864, 1280x1024, 1600x1200. I guess all graphical desktops will have their own settings and change resolutions in case?

Fourdee, 2018-01-26:
@MichaIng 

> Did you check, if adjustment is possible in VMware? Would drop priority again 😆.

You can increase the max resolution size, but it has no effect on current size.

MichaIng, 2018-02-27:
Notes:
- I would prefer a usual set of supported screen resolutions to be not dependent on external programs, and we can assume, that VMs do support, as long as video memory is large enough.
- Need to check influence of VMware max resolution.
- Otherwise install hwinfo one time, read and write supported modes to disk, then purge again and use the written values? Optionally offer to revalidate.
- Nice would be to test immediately, but there doesn't seem to be a way on console.
- `GRUB_GFXPAYLOAD_LINUX=keep` will keep the grub resolution, set via `GRUB_GFXMODE`

MichaIng, 2018-03-08:
Further notes:
- If on WMware the within grub applied resolution is higher than the maximum allowed by WMware settings, it results just the WMware/machines default of 800x600 (even if higher maximum is allowed).
- Same for unsupported resolutions on WMware and VirtualBox.
- VirtualBox however seems to have a hard limit of 1600x1200, where higher or disabled maximum settings as well as other settings (gpu memory, chipset) do not change something. WMware allows higher resolutions (1920x1080 tested), based on maximum settings. But VMware does not support 640x480 on the other hand, at least it jumps to 800x600 again, if not 640x480 is chosen as maximum.
- `hwinfo --framebuffer` does not show any info on WMware and VirtualBox Buster btw. but the supported resolutions seem to be the same.
- Thus the only thing that makes sense to me is:
  - Allow user setting one of the higher default resolutions, that is supported by all VM software: 800x600, 1024x768, 1152x864, 1280x1024 and 1600x1200 and system default (comment resolutions settings).
  - As well user should be informed, that it may be necessary to increase maximum resolution with VM software settings.

MichaIng, 2018-03-14:
PR is up: https://github.com/Fourdee/DietPi/pull/1618

closed 2018-03-16 by MichaIng