MichaIng/DietPi #1228: Wikipedia page for DietPi

Fourdee, 2017-11-06:
> https://twitter.com/marcdraco63/status/927631556784648192
> https://en.wikipedia.org/wiki/Wikipedia:How_to_create_a_page

k-plan, 2017-11-10:
@Fourdee 

like to help you out, but it's a job for a  :gb:  native speaker

But some links for missing links:

https://en.wikipedia.org/wiki/Raspberry_Pi#Operating_systems

(here is a Wikipedia sub page link missing)

-------------------

and here is a lack of link to DietPi on some Wikipedia sub pages (I think so):

https://en.wikipedia.org/wiki/ODROID#Operating_systems

https://en.wikipedia.org/wiki/Banana_Pi#Available_operating_systems

https://en.wikipedia.org/wiki/Pine64

https://en.wikipedia.org/wiki/Asus_Tinker_Board#External_links

Hope it will help.

Fourdee, 2018-02-02:
Marking as closed, no progress and we offer online docs via our forum. I'am unable to take this on for the foreseeable future, due to lack of available time.

If someone is able to pick this one up, please reopen.

closed 2018-02-02 by Fourdee