MichaIng/DietPi #1229: DietPi-Software | Improve Owncloud installation

Fourdee, 2017-11-09:
> https://github.com/Fourdee/DietPi/issues/1225#issuecomment-343210555

MichaIng, 2017-11-10:
W8888, something went crazy with your fix:
- For Stretch the installation location did not change, still is `/var/www/owncloud`!
- For Jessie I just tested, a bid confused: Also for Jessie the package `owncloud` does not exist anymore on ownClouds own repo, but there is an ancient version 7.x (now 10.x) in debian repo (just Jessie, not Stretch) that will be installed with very old dependencies and to "old" location!
- So this definitely needs to change back!!! As alternative fix just install `owncloud-files` package instead of `owncloud`, this will already fix it for all repos.

Fourdee, 2017-11-12:
@MichaIng 

Good finds 👍 , i'll mark this as a bug and set v159 milestone.

MichaIng, 2017-11-17:
Installation fix applied, taking into account the different distros: https://github.com/Fourdee/DietPi/pull/1243

We could also directly use manual archive download for all distros, but I have no idea how well the integrated updater works.

Configuration including occ installation, webserver configs etc. coming soon. We can do it similar to Nextcloud here. Only adding `1 => '*',` to `config.php` to disable trusted domains does not work here, thus we need to reliably get hostname and sed it instead.

MichaIng, 2017-11-20:
Found complete installation (+database/config.php creation) and nginx+apache configs for ownCloud on Jessie and Stretch. Moreless similar to Nextcloud. Also recreated both configs from official docs. Not much changed, just some parent location directives added.

We can install both beside another, but then some redirections (acme+cal/carddav esprcially) and security headers are duplicated.

Security headers can be moved into ownCloud/Nextcloud locations, the other parent directory directives could be moved into an own "ownnextcloud.conf" file, so both installation will just overwrite/check for the same. It is just a question where to redirect default cal/carddav and user_webfinger requests to. Would be nice to let user decide, if both clouds are installed. Should be anyway nearly never the case 😆.

Only problem is still the trusted domain thing mentioned above. @Fourdee Any idea how to determine the users domain>staticIP>localIP?

Fourdee, 2017-11-20:
@MichaIng 

Current IP address:
```
sed -n 4p /DietPi/dietpi/.network
```

Domain/hostname:
```
cat /etc/hostname
```

MichaIng, 2017-11-23:
Implemented: https://github.com/Fourdee/DietPi/pull/1251
ownCloud will now also install ready to use, with database configured, ownCloud occ-install etc.

Further improvements will be done together with Nextcloud. I renamed the old issue accordingly. This one could be closed now in favour of https://github.com/Fourdee/DietPi/issues/1067

Fourdee, 2017-11-25:
@MichaIng 

Great work 👍 Making as completed.

closed 2017-11-25 by Fourdee