MichaIng/DietPi #1231: Allo GUI Images | Update to V3

Fourdee, 2017-11-10:
- Includes stability patch for Sparky SBC, and, a few web interface optimizations.

k-plan, 2017-11-10:
@Fourdee 

like to help you out, but it's a job for a  :gb:  native speaker

But some links for missing links:

https://en.wikipedia.org/wiki/Raspberry_Pi#Operating_systems

(here is a Wikipedia sub page link missing)

-------------------

and here is a lack of link to DietPi on some Wikipedia sub pages (I think so):

https://en.wikipedia.org/wiki/ODROID#Operating_systems

https://en.wikipedia.org/wiki/Banana_Pi#Available_operating_systems

https://en.wikipedia.org/wiki/Pine64

https://en.wikipedia.org/wiki/Asus_Tinker_Board#External_links

Hope it will help.

Fourdee, 2017-11-12:
Completed, images now live.
Releasing DietPi v158 now.

closed 2017-11-12 by Fourdee