MichaIng/DietPi #1234: Allo USBridge not playing using Mosaic UV DAC

paulstephane, 2017-11-14:
## Creating a bug report/issue:
Reference code: adc52cd0-5767-4c2a-a8e9-f8a3f5d86857-0 

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 158
- SBC Device (EG: RPi 3)? Sparky
- Power supply used (EG: 5v 1A RAVpower)? Allo 5V supply
- SD card used (EG: Sandisk ultra)? 
- Distro (EG: Jessie) | ```uname -a```? Linux DietPi 3.10.38 #96 SMP PREEMPT Tue Nov 7 17:10:13 IST 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

paulstephane, 2017-11-30:
Issue solved by Allo
Can be closes

closed 2017-11-30 by paulstephane

Fourdee, 2017-12-01:
@paulstephane 
> Issue solved by Allo
Can be closes

Thanks. Apologies for not replying to this ticket, got lost in the pile :frowning_face: