MichaIng/DietPi #1235: Syncthing Update via web GUI

twentyninehairs, 2017-11-15:
Syncthing currently appears to not have write permissions to run an automated updates due to the process located at /usr/bin, and with the user 'dietpi'.

What is the correct solution for this?  Do you think the bin install directory should be somewhere else that folder permissions can be finely tuned? 

I think I would like to change the syncthing user to root while this is being resolved. But I'm not sure where that setting is?

## Creating a bug report/issue:

#### Required Information:
- DietPi Version | V158
- SBC Device | RPi 2
- Power supply used | 5v 2A Samsung
- SD card used | Sandisk ultra 32Gb
- Distro (EG: Jessie) | Jessie; 4.9.52+ #1 Mon Oct 2 21:14:49 UTC 2017 armv6l GNU/Linux

#### Additional Information (if applicable):
- Software title | Syncthing
- Can this issue be replicated on a fresh installation of DietPi? | Yes

#### Expected behaviour:
Autoupdate via GUI interface downloads and updates Syncthing automatically.

#### Actual behaviour:
gives the following error:
2017-11-15 00:56:16: upgrading: open /usr/bin/syncthing217887563: permission denied

#### Steps to reproduce:
Install fresh copy of syncthing in DietPi version 158
Attempt to update syncthing using the web interface.

#### Did you submit a ```dietpi-bugreport```?
No. Should I?

#### Extra details:
Apparently the Syncthing update utility attempts to write a temporary bin file to the directory the syncthing bin file is located to copy over the running bin with the temp if the temp download is successful. Because the current bin file is located at /usr/bin and the syncthing user is dietpi, the Syncthing update cannot write the temp file, or replace the running bin, for that matter.

joaofl, 2017-11-15:
What I did to get it working was to move the binary to the dietpi-user folder, and change its path on the corresponding dietpi-service.

I should do it for new installations as well.

twentyninehairs, 2017-11-16:
Where is the script that loads the dietpi service? I looked, but haven't found it yet.

joaofl, 2017-11-16:
it is located at `/etc/systemd/system/syncthing.service`

then you have move the executable to the dietpi_userdata folder, and edit the service file to look like:

```
[Unit]
Description=Syncthing
After=network.target

[Service]
Type=simple
ExecStart=/mnt/dietpi_userdata/syncthing/syncthing -logfile=/var/log/syncthing/syncthing.log -logflags=3 -home=/mnt/dietpi_userdata/syncthing
User=dietpi

[Install]
WantedBy=multi-user.target
```

secpri, 2017-11-28:
I've solved it by:

Login via SSH as root, then just "sudo syncthing -upgrade",
restart Syncthing services (i've done over WebInterface)

[image]