MichaIng/DietPi #1238: Bootloop - Error line 302 0#NTPD: invalid number

tatoosh, 2017-11-17:
## Required Information:
- DietPi Version  158
- SBC Device  rpi zero
- Power supply used 5.1v 2a
- SD card used  Sandisk Ultra 8GB
- Distro (EG: Jessie) | 4.9.52+

## Info:
If i have not connected my USB LAN device i will end up in a boot loop that i can't break.
[image]

How to solve this?

Fourdee, 2017-11-17:
@oparoz 

Please change/verify the value of ```boot_wait_for_network=``` in ```/DietPi/dietpi.txt``` to:
```
boot_wait_for_network=1
```

oparoz, 2017-11-17:
I think you meant @tatoosh ?

tatoosh, 2017-11-18:
@Fourdee then my boot is delayed? I need a very fast boot for my music player project. Any other chance to solve this?

Fourdee, 2017-11-20:
@tatoosh 

0 will be fastest for boot, but will not wait for network.

Also, ensure new line after ```=0```, as per:
https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L159

tatoosh, 2017-11-21:
@Fourdee: solved by inserting blank after ```boot_wait_for_network=0```.

tatoosh, 2017-11-21:
@Fourdee: solved by inserting blank ```after boot_wait_for_network=0 ```
So you habe to check your script please.

closed 2017-11-21 by tatoosh