MichaIng/DietPi #1239: WLAN Error for wireless request "Set Encode" (8B2A) :

tatoosh, 2017-11-17:
## Required Information:
- DietPi Version  158
- SBC Device  rpi zero
- Power supply used 5.1v 2a
- SD card used  Sandisk Ultra 8GB
- Distro (EG: Jessie) | 4.9.52+

## Error

My wifi is not working anymore.

## Reproduction

```bash
root@uno:~#  iwconfig wlan0 essid x2 key s:DietPiTestPassword
Error for wireless request "Set Encode" (8B2A) :
    SET failed on device wlan0 ; Invalid argument.
```

MichaIng, 2017-11-22:
@tatoosh 
Did you try the solution of @TheOldPresbyope to manually create wpasupplicants execution folder and reconfigure wifi via dietpi-config?
`mkdir /var/run/wpa_supplicant`

k-plan, 2017-11-25:
TESTING: DietPi v156 on RPi-Zero W

------------------

```

───────────────────────────────────────
 DietPi     | Update available
 ───────────────────────────────────────
 V156       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 
root@RPi-Zero_W:~# uname -a
Linux RPi-Zero_W 4.9.52+ #1 Mon Oct 2 21:14:49 UTC 2017 armv6l GNU/Linux

root@RPi-Zero_W:~# cat /etc/debian_version
8.0

root@RPi-Zero_W:~# /DietPi/dietpi/misc/rpi_boardinfo
      - RPi Board Info -
 Revision         |     9000c1
 Released         |     Q1 2017
 Model            |     Zero W
 Memory           |     512 MB
 Manufacturer     |     Sony
 

root@RPi-Zero_W:~# iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"Test-NET"
          Mode:Managed  Frequency:2.412 GHz  Access Point: D6:CA:6D:xx:xx:xx
          Bit Rate=52 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=67/70  Signal level=-43 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
          
          
root@RPi-Zero_W:~# ls -la /var/run/ | grep wpa
drwxr-x---  2 root root     60 Nov 25 17:17 wpa_supplicant
-rw-r--r--  1 root root      4 Nov 25 17:17 wpa_supplicant.wlan0.pid  

root@RPi-Zero_W:~# dietpi-update
```
------------------

TESTING: DietPi v157 on RPi-Zero W

```
───────────────────────────────────────
 DietPi     | Update available
 ───────────────────────────────────────
 V157       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 
 
root@RPi-Zero_W:~# uname -a
Linux RPi-Zero_W 4.9.62+ #1 Fri Nov 17 22:37:52 GMT 2017 armv6l GNU/Linux

root@RPi-Zero_W:~# iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"Test-NET"
          Mode:Managed  Frequency:2.412 GHz  Access Point: D6:CA:6D:xx:xx:xx
          Bit Rate=65 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=70/70  Signal level=-39 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:5  Invalid misc:0   Missed beacon:0
          
          
root@RPi-Zero_W:~# ls -la /var/run/ | grep wpa
drwxr-x---  2 root root     60 Nov 25 17:44 wpa_supplicant
-rw-r--r--  1 root root      4 Nov 25 17:44 wpa_supplicant.wlan0.pid

root@RPi-Zero_W:~# dietpi-update
```
------------------

TESTING: DietPi v158 on RPi-Zero W

```

───────────────────────────────────────
 DietPi     | 17:49 | Sat 25/11/17
 ───────────────────────────────────────
 V158       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 

root@RPi-Zero_W:~# iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"Test-NET"
          Mode:Managed  Frequency:2.412 GHz  Access Point: D6:CA:6D:xx:xx:xx
          Bit Rate=52 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=69/70  Signal level=-41 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:8  Invalid misc:0   Missed beacon:0
          

root@RPi-Zero_W:~# ls -la /var/run/ | grep wpa
drwxr-x---  2 root root     60 Nov 25 17:49 wpa_supplicant
-rw-r--r--  1 root root      4 Nov 25 17:49 wpa_supplicant.wlan0.pid
 
 
root@RPi-Zero_W:~# ping dietpi.com -c 4
PING dietpi.com (185.101.92.145) 56(84) bytes of data.
64 bytes from 185.101.92.145: icmp_seq=1 ttl=57 time=14.6 ms
64 bytes from 185.101.92.145: icmp_seq=2 ttl=57 time=17.0 ms
64 bytes from 185.101.92.145: icmp_seq=3 ttl=57 time=16.9 ms
64 bytes from 185.101.92.145: icmp_seq=4 ttl=57 time=17.2 ms

--- dietpi.com ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3004ms
rtt min/avg/max/mdev = 14.611/16.480/17.243/1.086 ms

root@RPi-Zero_W:~# dmesg | grep brcmfmac
[    7.448824] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    7.467968] usbcore: registered new interface driver brcmfmac
[    7.781370] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    7.782557] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.41 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-08-07 00:37:47
[   11.385038] brcmfmac: power management disabled
[   11.736050] brcmfmac: power management disabled
[   31.979187] brcmfmac: power management disabled

root@RPi-Zero_W:~# dmesg | grep wifi
[    2.161769] systemd[1]: Cannot add dependency job for unit dietpi-wifi-monitor.service, ignoring: Unit dietpi-wifi-monitor.service failed to load: No such file or directory.

```
 ------------------

Hmm ... by my tests here I can not see or have any WiFi problems.

Have you:

- Auto Reconnect : Enabled  ?
- WiFi Country set correct ?
- more then the on-board wifi chip in use?

tatoosh, 2017-11-25:
Hi @k-plan.
- using pinzero with this USB WLAN: https://www.rasppishop.de/Offizieller-Raspberry-Pi-Mini-WLAN-USB-Stick-Dongle
- auto reconnect is enabled
- # in my wpa2 key

k-plan, 2017-11-25:
Hi @tatoosh,

>  - # in my wpa2 key

Possibly not the best idea. Can you please change this first and test again?

> - auto reconnect is enabled

For testing I would suggest to set this on Disable

> - using rpi zero (without w) with this USB WLAN

Can you please post:

`lsusb`

`dmesg | grep usbcore`

and when

`dmesg | grep xxx` -> xxx =  name of registered new interface driver

tatoosh, 2017-11-25:
- a \# in my wpa2 key

k-plan, 2017-11-25:
@tatoosh 

Is it solved without " # " in wifi-password?

tatoosh, 2017-11-25:
hi @k-plan : yes, it is working nice. If you want i can reproduce the error in a few days.

tatoosh, 2017-11-25:
@MichaIng 
Thx. Yes the folder exists.

k-plan, 2017-11-25:
> `Bus 001 Device 004: ID 0a5c:bd1e Broadcom Corp.`

>  `[   10.297753] usbcore: registered new interface driver brcmfmac`

brcmfmac is you wifi interface driver 

.. but anyway.

> i had problmes by installing with preconfigured dietpi.txt on my pi zero starting

If you edit `dietpi.txt` on a differnt device, please use a unix compatible editor like NotePad++, Notepad2, PS Pad, Nano, vim, ...

Think same is happend here: https://github.com/Fourdee/DietPi/issues/1238#issuecomment-345980758

tatoosh, 2017-11-25:
> brcmfmac is you wifi interface driver
Ok thx.
> If you edit dietpi.txt on a differnt device, please use a unix compatible editor like NotePad++, Notepad2, PS Pad, Nano, vim, ...

I used Notepadd++ as **UTF-8 without BOM**. But i will take a stranger look here next time. Good advice!

k-plan, 2017-11-25:
Will close this Issue. Please reopen if needed. Thanks.

closed 2017-11-25 by k-plan