MichaIng/DietPi #1240: [Question] DietPi-Software | Nginx why "xml-core"?

MichaIng, 2017-11-17:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3732-L3739

Is there a reason why the apt package `xml-core` is installed together with `nginx`?
Also I didn't find any recommendations for the nginx package on any repo, thus the `--no-install-recommends` option has no effect here 🙂.

Could create PR to clean this up.

Fourdee, 2017-11-17:
@MichaIng 

Unsure, cant honestly remember. If you do remove it, just do a quick test with nextcloud (or other web interface software) to ensure it doesn't break it.

MichaIng, 2017-11-17:
@Fourdee 
Nextcloud and ownCloud do work well without it. Of course I can't test all web service software now 😆.

Another question about apt packages I find after nginx/lighttpd stack installation: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3814
php-cli and php-xsl are installed for non-apache stacks, but Nginx and Lighttpd are both configured to use php-fpm. Thus the other two are not necessary in any case, or do I miss something?

Of course also tested with Nginx+Lighttpd ownCloud+Nextcloud installations.

MichaIng, 2017-11-20:
Made further investigation: 
- On Jessie xml-core is recursive recommendation for nginx > nginx-full/extras > libxml2. Thus it will be automatically installed together with nginx at the moment, as long as recommendations are not yet excluded from auto installation.
- On Stretch it has gone, as far as I was looking recursively into the deps.
- By looking into this, I recognized the [nginx-light](https://packages.debian.org/de/stretch/nginx-light) package, which is also an optional dependency of nginx, where nginx-full it the default.
- I would suggest to by default install this light version, which excludes some optional nginx modules, including mail, dav, geo, maps and stuff like that. Tested with Nextcloud + ownCloud on Stretch and Jessie.
- Additional modules can be installed together with software as necessary. Switching between nginx versions is also easy, they flawlessly replace each other on Stretch and Jessie.

Fourdee, 2017-12-01:
@MichaIng 

> I would suggest to by default install this light version, which excludes some optional nginx modules, including mail, dav, geo, maps and stuff like that. Tested with Nextcloud + ownCloud on Stretch and Jessie.
Additional modules can be installed together with software as necessary. Switching between nginx versions is also easy, they flawlessly replace each other on Stretch and Jessie.

My main concern is the sheer amount of web based software titles we offer. 
If we did switch to ```nginx-light```, we would need to, at the very least, test and ensure all web titles we offer, function with the lighter installation. This is the only way we will know if those additional default ```nginx-full``` modules are required.

I'am personally not keen relying on our user base to find the issues with non-major software titles. If we can get verify all our web software titles with ```nginx-light``` first, we can make the change.

MichaIng, 2017-12-01:
@Fourdee
Yes you are right. That would make a lot of work. Maybe I will start a separate issue with list to check and fill this step by step as far as I find time to check.
Anyway this is a total non priority issue, just of interest for perfectionists :).

Same for xml-core. There I guess on Stretch at least it is not necessary anymore as it was removed from recommends of nginx package Jessie -> Stretch.

MichaIng, 2018-01-23:
@Fourdee @k-plan
Maybe you know this better than me: Many of our software titles with web interface are reachable via non-standard port, where the webserver should actually not listen, eg 8080 8000 8443 etc. I heard that these 1k+ ports are anyway handled differently, but not sure how this works. Do the software titles themself answer this ports/redirect them to webserver or somehow directly call PHP, moreless skipping webserver? Sorry for maybe dump questing, I just want to know if they even can be affected by (missing) webserver modules e.g.

Fourdee, 2018-01-23:
@MichaIng 

If its not using port 80, the software title most likely has a mini webserver bundled into the program binary (eg: netdata), available on its specific port.

MichaIng, 2018-01-23:
@Fourdee 
Okay, hmm for Koel e.g. (which is reachable via port 8000) we also install a webserver. Somehow online guides also suggest installing mostly nginx. However on the official Koel installation guide I don't find any hint about needed webserver, just PHP of course. I will test, if it still works after manually uninstalling nginx 😄.

MichaIng, 2018-02-03:
@Fourdee @k-plan 
If you have some spare time and otherwise unused RPi, could you test, if `moOde` is working with just `nginx-light` installed (instead of `nginx`=`nginx-full`), and without `xml-core`, `php5-xsl` and `php(5)-cgi`?

All other webserver dependant software is tested now on Jessie + Stretch (see above). All of them don't depend on named packages packages, but `rtorrent`. At least installation + all quick testable web ui pages work fine. Just rtorrent depends on webserver SCGI package, which is not included in Jessie nginx-light (but Stretch nginx-light). But it is painless to just add `(( $G_DISTRO < 4 )) && G_AGI nginx-full` there, which automatically replaces the light package.

I know the intention of this issue is low prio. But I took the change to general verify function of our software with v6.0+, as also done with https://github.com/Fourdee/DietPi/pull/1418 😉.

k-plan, 2018-02-03:
@MichaIng 

> otherwise unused RPi, could you test, if moOde is working with just nginx-light installed

Sehe ich das richtig, moOde benötigt einen RPi2 oder RPi3 mit Audio Ausgang, also mit einem DAC?
HDMI geht hier nicht, da ich mittlerweile Familien Fernseher Herumspielverbot habe. 😄 
Dann wäre ich leider raus. Zu dem ganzen Audio Zeug habe ich eh' keinen richtigen Draht und somit kaum Erfahrung.

> But I took the change to general verify function of our software with v6.0+, as also done with

Alter Falter! Respekt, das wird ja richtig systematisch und strukturiert ausgetestet. 👍 
Hut ab, du hängst dich richtig rein. Das war eine Menge Arbeit. Hast du sehr gut gemacht und dokumentiert. ❤️ 

Langsam wird es Zeit, dass ich als ahnungsloser Bastler mich aus diesem Projekt verabschiede und das Feld den Profis überlasse.

MichaIng, 2018-02-03:
@k-plan 
RPi1 sollte auch gehen, jedenfalls haben wird moOde dort auch im DietPi-Software UI angezeigt. 
Ich habe nun auch nicht mehr getestet, als das fehlerlose Starten/Navigieren des web interfaces und vereinzeltem Ändern von Einstellungen. Wenn da keine Fehler auftauchen und der nginx log nix anzeigt, scheint webserver bezogen ja alles nötige vorhanden zu sein.
Und eben, trotz low prio des Themas, fand ich es insbesondere mal wertvoll bei den vielen Änderungen die fehlerlose Installation selbst zu testen. Da gab es ja gleich nach v6.0 release ein paar Kandidaten mit geänderten Abhängigkeiten/fehlenden Abhängigkeiten auf Stretch.

Insofern waren die Tests jetzt nicht soo aufwendig 😉. War auch mal ganz gut zu sehen, dass via DietPi auch viel Software auf einen rutsch installiert werden kann und vernünftig parallel läuft. War da echt überrascht, da gab es nur wenig gegenseitige Störung: Allo WebUI und Grashopper wollen beide über den webserver root aufgerufen werden, da überschreiben sich die index.php dann natürlich gegenseitig
 😄.

> Langsam wird es Zeit, dass ich als ahnungsloser Bastler mich aus diesem Projekt verabschiede und das Feld den Profis überlasse.

Um Himmels willen, nein! Nachher verrennen wir "coding Fachidioten" uns noch in irgendwelchen Details 😆. Ich meinte tue ich mich ja im Grunde genommen mit dem Thema schon etwas 🙈, also wenn es nur um das umsatteln auf nginx-light ginge 😉.

Fourdee, 2018-02-03:
@MichaIng 

> moOde is working with just nginx-light installed 

moOde is still in early dev, don't worry about it for now, i need to revisit the whole installation anyway.

MichaIng, 2018-02-03:
@Fourdee 
Okay, in this case I will open PR with suggested changes 🙂: https://github.com/Fourdee/DietPi/pull/1461

closed 2018-02-03 by MichaIng