MichaIng/DietPi #1244: DietPi-Config | Nvidia driver: nouveau disable required for 750Ti

Fourdee, 2017-11-19:
And possibly other cards:

> http://www.allaboutlinux.eu/remove-nouveau-and-install-nvidia-driver-in-debian-8/

Fourdee, 2017-11-19:
Fix applied, should be fine, however testing required to verify.

On a fresh installation:
- Edit ```/DietPi/dietpi.txt``` and set ```gitbranch=testing``` (bottom)
- Run ```dietpi-update```
- Install the driver and test

Fourdee, 2018-04-05:
Marking as closed due to:
- No activity.
- Inability to locally test 750Ti. I own a 1050Ti only, which is fully functional.

closed 2018-04-05 by Fourdee

NolandTech, 2025-11-26:
I seem to have this issue on a fresh install of the lastest UEFI iso. But I see this post is from 2017... Has this not been fixed yet? 

[image]