MichaIng/DietPi #1245: Request | Bundle pihole & pivpn

vmavromatis, 2017-11-19:
This is a feature request :)

With these two becoming a very popular option for RPi's (and the like) make it so that these two play nicely together. 
There is a nice guide to make them work together https://itchy.nl/raspberry-pi-3-with-openvpn-pihole-dnscrypt and dnscrypt is surely nice, but if it's too much work it can be skipped. The other two however seem to work very well together (remote network ad-free vpn is awesome - with your phone too!)

Just an idea, let me know what you think.

Fourdee, 2017-11-22:
@vmavromatis 

Thanks for the request 👍 

We've had a few users attempt this installation manually, would be great if we could automate this for them.

If noone else is able to take this on, i'll pick this up when I can (unsure at the moment)

vmavromatis, 2017-11-25:
@Fourdee many thanks for the response. If this ever does get implemented, it would make sense to tweak the iptables as well accordingly. (I'm still struggling with them, that's why I note this :P)

MichaIng, 2018-12-23:
Additional vote for this: https://dietpi.com/phpbb/viewtopic.php?f=12&t=5336&p=15828#p15828

miguipda, 2018-12-28:
Hi,

great to read this should be considered to be impemented in DietPi.
I will wait it soon ;-)

LexiconCode, 2018-12-31:
> 
> 
> Additional vote for this: [dietpi.com/phpbb/viewtopic.php?f=12&t=5336&p=15828#p15828](https://dietpi.com/phpbb/viewtopic.php?f=12&t=5336&p=15828#p15828)

I think you meant for a different Hyperlink/

Joulinar, 2021-02-07:
is this still a valid request? because during installation of PiVPN you could specify to use local DNS server

[image]

MichaIng, 2021-02-07:
I think it's not entirely the same aim, while the result could be the same: The PiVPN selection AFAIU is more about defining the upstream DNS that you trust, when using your VPN for other purpose, while the aim of this PR is to setup the VPN only to use Pi-hole remotely. But of course both might be wanted or overlap at least.

Also what I am not 100% sure about is:
- If you want to use the VPN for Pi-hole based ad blocking only, you'd setup WireGuard to tunnel requests to that Pi-hole/WireGuard only, while keeping all other requests bypassing the VPN and skip forwarding/NAT rules server-side. Our default WireGuard client configs e.g. contain a commented block which can be used to do so.
- But now, when defining a DNS in the client config as well, is this used for all requests or only for those tunnelled through the VPN? Ah should be the first case, since routing is based on IP addresses, not hostnames, so DNS is done before the client even knows whether to tunnel the request or not.

Does the PiVPN client setup script as well allow to define which requests to tunnel, or is everything tunnelled by default?
___________
Actually, since we recently implemented the Pi-hole + Unbound co-configuration and I just added OctoPrint + mjpg-streamer, and it is quite some doubled code and guessing involved to configure those to work together automatically, I think it would make sense to instead add dedicated bundled install options, so that when those are selected it is 100% clear that both shall be configured to work together, otherwise it's standalone installs. The same could be done then for:
- Pi-hole + PiVPN (even that it is only the "all origins" options for Pi-hole that needs to be set)
- Pi-hole + WireGuard (no IP forwarding required, only tunnel requests to WireGuard/Pi-hole server itself as DNS resolver)
- Pi-hole + OpenVPN (same as with WireGuard)

While PiVPN is great and now again fixed in regards to WireGuard on ARM installs, it's WireGuard support is limited compared to ours, and some might not need or want the additional scripts and features it provides.

Joulinar, 2021-02-07:
PiVPN is quite simple on this. There is exactly a single place where you can specify the upstream DNS. This is done during installation as shown above. Looks like PiVPN is storing this information somewhere and it's used during client configuration creation. During client creation, nothing is ask, except the client name. Nothing you can specify.

```
root@DietPi3:~# pivpn add
Enter a Name for the Client: demo
::: Client Keys generated
::: Client config generated
::: Updated server config
::: WireGuard reloaded
======================================================================
::: Done! demo.conf successfully created!
::: demo.conf was copied to /home/dietpi/configs for easy transfer.
::: Please use this profile only on one device and create additional
::: profiles for other devices. You can also use pivpn -qr
::: to generate a QR Code you can scan with the mobile app.
======================================================================
root@DietPi3:~#
```

By default entire traffic is routed to the VPN tunnel as `AllowedIPs = 0.0.0.0/0, ::0/0` will be set. If you really like to use your own config, better to use DietPi WireGuard 😃

MichaIng, 2021-02-07:
Okay, indeed then such a bundle still makes sense, where the VPN is used only for DNS purpose. Then PiVPN actually never was a great choice to achieve this.

> BTW: PiVPN did not detected the new kernel module for WireGuard since we are on kernel 5.10 with RPi OS.

Hmm, what's wrong there again? I'll have a look.

Joulinar, 2021-02-07:
probably 5.10 ist still to new and they are lacking behind to adopt.?

MichaIng, 2021-02-07:
Indeed, although they could have adapted already without breaking anything with older kernel versions.

On all systems it is checked whether WireGuard is builtin or not. But on Raspbian the result of this check is not used to decide whether to install kernel headers and DKMS or not: https://github.com/pivpn/pivpn/blob/master/auto_install/install.sh#L1269
See a few lines below where the WIREGUARD_BUILTIN variable is used intentionally.

I'm gonna fix that.

MichaIng, 2022-05-27:
PiVPN (in the meantime?) detects Pi-hole already and allows to configure it as VPN DNS 👍. So nothing to do on that end, but plain OpenVPN and WireGuard install options only.