MichaIng/DietPi #1246: Request: medusa instead of sickrage

DDA78, 2017-11-19:
Can I propose inclusion of medusa in the software list, it has much more features then sickrage and more actively developed

https://github.com/pymedusa/Medusa

https://github.com/pymedusa/Medusa/wiki/Installation-&-Configuration-Guides

DDA78, 2017-11-25:
Hello,

Is the inclusion of this possible, any ETA 

Thanks for the work you developers are doing.

sfb2, 2017-11-25:
thx a lot..according to their github theres also a Medusa-"Docker"-image  for armhf too..->[](https://hub.docker.com/r/lsioarmhf/medusa/) ; if anything goes wrong here ;-)

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

ezekini, 2018-10-14:
I would really like this addition. Sickrage is going down badly and a huge part of sickrage users are moving to Medusa or SickChill.
https://www.reddit.com/r/sickrage/comments/9ll427/echelon_attempting_hostile_takeover_again/

bonshates, 2018-11-10:
To start, thank you to all of the folks that support DietPi. It is greatly appreciated.

I have read that most of the SickRage devs have moved over to Medusa. Please consider adding it to DietPI.
https://www.reddit.com/r/sickrage/comments/9myqx0/can_some_give_me_a_run_down_on_what_is_going_on/