MichaIng/DietPi #1248: After update to v158 Wifi stopped working

OpNop, 2017-11-20:
#### Required Information:
- DietPi Version: v158
- SBC Device: Raspberry Pi 2 Model B
- Power supply used: 5v 3a
- SD card used: PNY Class 10 16GB
- Distro: Linux DietPi 4.9.62-v7+

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi?
 Yes
- Hardware in question:
 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]

#### Expected behavior:
USB wifi should work like it does before the update

#### Actual behaviour:
After updating from an earlier version (can not remember exact version, v154 possibly) the WiFI no longer connects to my internet. When setting up WiF in dietpi-config the following error is reported

```
 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: wifi (enable)
 Please wait...

 [Info] Please wait, enabling WiFi Modules...
 [Ok] wifi enable | Completed

 [Info] Restarting networking services, please wait...
ifup: interface eth0 already configured
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "MyWPA2PasswordIsShownHere".
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:25:f4:64
Sending on   LPF/wlan0/74:da:38:25:f4:64
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 3
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```
The adapter does pull a DHCP address on first boot when it is still on v150 but after auto updates it breaks again

#### Steps to reproduce:
- Insert the USB WiFi adapter
- Install fresh copy of v150 instraller
- Let auto updates do its thing
- Try to configure WiFi
- See error message

#### Did you submit a ```dietpi-bugreport```?
No

tatoosh, 2017-11-21:
I had same problems and opend issue #1239.
My problem is solved at the moment by the help of an USB LAN Adapter but i don't know why because i tested very much out ;)

MichaIng, 2017-11-21:
@tatoosh 
After which update step did you face the issue? 157 -> 158 or are you also comming from an earlier version?

I just tired so see where exactly Wifi was touched on the last updates and for first short investigation just found on v157 update `DietPi-Config | Resolved syntax issues with wifi_disable_powersave.service` where the service to disable wifi power management was corrected: https://github.com/Fourdee/DietPi/commits/master/dietpi/func/dietpi-set_hardware

Sadly I can't test this by myself, as my RPi2 has no integrated Wifi and VMs do not allow to simulate Wifi or pass-through my hosts Wifi hardware.

MichaIng, 2017-11-22:
@OpNop 
Did you try the solution of @TheOldPresbyope to manually create wpasupplicants execution folder and reconfigure wifi via dietpi-config?
`mkdir /var/run/wpa_supplicant`

Fourdee, 2018-02-22:
Outdated, < v6.0

closed 2018-02-22 by Fourdee