MichaIng/DietPi #1250: (RPi) V158 DietPi and DietPi Stretch - missing directory: /var/run/wpa_supplicant

TheOldPresbyope, 2017-11-22:
## Inoperative WiFi likely due to missing directory.

#### Required Information:
 ───────────────────────────────────────
 V158       | RPi 3 Model B (armv7l) also RPi0W (armv6l)
 ───────────────────────────────────────
 
- Power supply used: 5v/2a
- SD card used: MicroCenter housebrand class 10 8GB uSD
- Distro: Jessie | Stretch

#### Additional Information:
- Tried both DietPi_v150_RPi-armv6-(Jessie).img, DietPi_v156_RPi-armv6-(Stretch).img
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:
It should be possible to enable and configure the Wireless interface using dietpi-config.

#### Actual behaviour:
documented in several previous issues (see, e.g., #1248, #1239)

#### Steps to reproduce:
Just boot and let DietPi upgrade, then try to startup WiFi

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
While looking for clues in various V158 system directories, I noticed that the /var/run/wpa_supplicant directory is missing, although it is present in the .img files. As its name implies, this directory is needed by wpa_supplicant.

I created the directory and rebooted. Now I can startup the WiFi interface using dietpi-config. Huzzah, it works. Disconnecting the ethernet cable and rebooting, I can successfully ssh into the RPi via wlan0. Example, running the Stretch image (cuz I want to get Moode going with WiFi):

───────────────────────────────────────
 DietPi     | 04:13 | Wed 22/11/17 
 ───────────────────────────────────────
 V158       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 IP Address | 192.168.1.202
 ───────────────────────────────────────
 ...
root@DietPi:~# ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:0e:17:01  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 160  bytes 12960 (12.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 160  bytes 12960 (12.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.202  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ba27:ebff:fe5b:4254  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:5b:42:54  txqueuelen 1000  (Ethernet)
        RX packets 233  bytes 31125 (30.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 105  bytes 16579 (16.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

MichaIng, 2017-11-22:
Are you sure, this is connected to the other issues you linked? Did you otherwise see the same errors?
```
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
Error for wireless request "Set Encode" (8B2A) :
    invalid argument "MyWPA2PasswordIsShownHere".
```
I am wondering why wpasupplicant does not create the folder by itself, as it should. But found some other older topics on web, where the same issue occured with this folder missing.

Let's see if your solution also works for the other. Many thanks for sharing this in every case! 😃.

TheOldPresbyope, 2017-11-22:
Sorry, I was focused on the fact we couldn't bring up WiFi. Yes, the 8B2A error shows up once in journalctl output and displays the WiFi passphrase in clear text (I substituted x's for my 12-character passphrase), but the interface comes up and associates with my AP. 

```
Nov 22 04:12:22 DietPi kernel: brcmfmac: power management disabled
Nov 22 04:12:22 DietPi systemd[1]: Started Update UTMP about System Boot/Shutdown.
Nov 22 04:12:22 DietPi kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Nov 22 04:12:22 DietPi kernel: brcmfmac: power management disabled
Nov 22 04:12:22 DietPi systemd[1]: Reached target System Initialization.
Nov 22 04:12:22 DietPi sh[313]: Error for wireless request "Set Encode" (8B2A) :
Nov 22 04:12:22 DietPi sh[313]:     invalid argument "xxxxxxxxxxxx".
```
A web search turns up instances of people seeing these errors with various Linuxes over the years with no clear explanation. I'm beginning to get a spidey sense about it, but nothing definite.

Fourdee, 2017-11-22:
@TheOldPresbyope 

Thanks for the report 👍 

> I noticed that the /var/run/wpa_supplicant directory is missing, although it is present in the .img files.

Lets get this confirmed via testing. AFAIK, we do not specifically remove this folder in any of our scripts.

TheOldPresbyope, 2017-11-28:
Update:
I've been away for the last week.

Revisiting this issue, I've come to agree with @MichaIng. It seems to me the wpa_supplicant code ought to be creating this dynamic directory if it isn't present (/var/run is a symlink to /run, actually, so I probably should have said "/run/wpa_supplicant"). A running wpa_supplicant process writes its control socket there; user-space code like wpa_cli can communicate with the process via this control socket. I'm slowly poring over sources to try to figure out why the directory isn't being created in DietPi Stretch. My solution is just a hack in the interim.

Fourdee, 2017-11-28:
@TheOldPresbyope 

>  try to figure out why the directory isn't being created in DietPi Stretch

I'am currently working on Stretch and ```wpa_supplicant.conf``` network changes. I've not personally see any issues yet with ```/run/wpa_supplicant``` not existing, but i'll keep an eye on it.

TheOldPresbyope, 2017-11-29:
The work outlined in #1262 makes this issue moot. Closing.

Fourdee, 2017-12-02:
Issue with our default SSID/KEY, prevents ifup
```
root@DietPi:~# wpa_supplicant -Dnl80211 -iwlan0 -c/etc/wpa_supplicant/wpa_supplicant.conf
Successfully initialized wpa_supplicant
Line 8: Invalid passphrase length 7 (expected: 8..63) 'Not_Set"'.
Line 8: failed to parse psk '"Not_Set"'.
Line 9: failed to parse network block.
Failed to read or parse configuration '/etc/wpa_supplicant/wpa_supplicant.conf'.
```

Fourdee, 2017-12-03:
> https://github.com/Fourdee/DietPi/commit/992dd5044c6fecc45f0aaf31ff76cc3b0bf592d5

Resolved

closed 2017-12-03 by Fourdee