MichaIng/DietPi #1253: DietPi-Config | CPU performance benchmark

Fourdee, 2017-11-22:
Add ability to benchmark CPU performance, using ```sysbench``` and a MAX prime value.

This will give users a rough idea of CPU performance, when we compare the total time against other systems.

Fourdee, 2017-11-22:
#### Odroid C2 (possibly cheating this test?):
[image]

#### Z83-ii | Intel Atom Cherry Trail x5-Z8300
[image]

#### RPi 3:
[image]

Fourdee, 2017-11-22:
Completed.

closed 2017-11-22 by Fourdee

ThomasKaiser, 2017-11-24:
> Odroid C2 (clearly "cheating" this test?):

No, just you simply using exactly the wrong tool: `sysbench` can be used as a compiler benchmark but not to compare CPU performance of different systems/architectures.

An `arm64` Stretch binary running on RPi 3 will finish in less than 3.1 seconds (while upstream `arm64` binaries for Jessie take ~30% longer). With `sysbench` everything that matters are compiler switches and that's why you get different numbers depending on the GCC version the binary has been built with (that's the distro dependency) and especially allowing for certain features (that's why `arm64` binaries on ARMv8 SoCs outperform `armhf` binaries by magnitudes). BTW: if you choose `sysbench` 0.5 instead of the 0.4.12 Debian still uses numbers get also worse.

TL;DR: There's no other tool compared with `sysbench` that is less capable to show CPU performance comparing different platforms

Fourdee, 2017-11-24:
@ThomasKaiser 

> No, just you simply using exactly the wrong tool: sysbench

Whats the correct tool?

> With sysbench everything that matters are compiler switches and that's why you get different numbers depending on the GCC version the binary has been built with (that's the distro dependency) and [...]

Thanks for the info 👍 

So in theory, the only way to ensure a consistent benchmark across platforms:
- We can compile our own ```sysbench``` binaries, ensuring compile options and GCC versions match (where possible) for each CPU arch.
- Find an alternative (but still the GCC version and compile options will be a factor across CPU arch's)
- Code my own bash benchmark, integer based only as floating point would require ```bc```, again, causing compile options and GCC versions to be a factor.

Fourdee, 2017-11-24:
Multi-threaded bash integer benchmark:
```
			cat << _EOF_ > /tmp/dietpi-bench
#!/bin/bash
target_max_int=1000000
cores=\$(nproc --all)
int_split=\$((\$target_max_int / \$cores ))
aStart_Int=()
aEnd_Int=()

for (( i=0; i<\$cores; i++ ))
do

    aEnd_Int[\$i]=\$(( (\$i + 1) * \$int_split ))
    aStart_Int[\$i]=\$(( \${aEnd_Int[\$i]} - \$int_split ))

    echo \${aStart_Int[\$i]} \${aEnd_Int[\$i]}

done

Run_Bench()
{

    while (( \${aStart_Int[\$1]} < \${aEnd_Int[\$1]} ))
    do

        ((aStart_Int[\$1]++))

    done

}

for (( i=0; i<\$cores; i++ ))
do

    Run_Bench \$i &

done

for job in \`jobs -p\`
do
    echo \$job
    wait \$job
done

unset aStart_Int
unset aEnd_Int

_EOF_
		chmod +x /tmp/dietpi-bench

		time /tmp/dietpi-bench
```

ThomasKaiser, 2017-11-24:
> Whats the correct tool?

There is none since it's all about use case. :)

I normally rely on `7z b` for a rough estimate of integer performance (and also memory bandwidth which influences results slightly) but on boards with low memory conditions `7z` might get killed by the kernel since allocating too much memory and then some use cases aren't really affected by above 'integer performance'.

Or the other way around: on certain SoCs we deal with accelerators for this and that so taking this not into account would be misleading. If someone searches for full disk encryption or a board to be used as VPN endpoint then all those 64-bit SoCs that have ARMv8 AES crpyto extensions licensed are [magnitudes faster than those who don't](https://forum.armbian.com/topic/4583-rock64/?do=findComment&comment=37829) (the latter group consists of all 32-bit SoCs, Raspberry Pi 3 and ODROID-C2).