MichaIng/DietPi #1254: Image request Nanopi Fire2A , Fire3

MrHohenheim, 2017-11-22:
Hello, is it possible to get an Image from Dietpi for Nanopi Fire2A and Fire3

Fire2A
http://www.friendlyarm.com/index.php?route=product/product&path=69&product_id=205
Fire3
http://www.friendlyarm.com/index.php?route=product/product&path=69&product_id=206

and this 2 version have a LCD support
http://www.friendlyarm.com/index.php?route=product/product&path=81&product_id=167

Many thanks :)

MrHohenheim, 2018-01-05:
oh and i forget mentoin nanopi neo plus 2
http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Plus2

any news?

Fourdee, 2018-07-09:
> any news?

We require either:
- Funding (via donations) to purchase the device
- Dev sample board sent via manufacture

Before we can implement an image + support.

closed 2018-07-09 by Fourdee