MichaIng/DietPi #1255: Home Assistant Service not starting

GiffE, 2017-11-22:
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? v158
- SBC Device (EG: RPi 3)? Odroid C2
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)? 
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title? Home Assistant
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:
Home assistant service should be starting automatically on boot.

#### Actual behaviour:
Home assistant service not starting

#### Steps to reproduce:
Clean install of v158 with home assistant software checked.
Home assistant service doesn't start.

#### Extra details:

 [Failed] home-assistant	failed (Result: exit-code) since Wed 2017-11-22 17:37:18 EST; 1min 29s ago
● home-assistant.service - Home Assistant
   Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled)
   Active: failed (Result: exit-code) since Wed 2017-11-22 17:37:18 EST; 1min 29s ago
  Process: 1582 ExecStart=/srv/homeassistant/homeassistant-start.sh (code=exited, status=127)
 Main PID: 1582 (code=exited, status=127)

Nov 22 17:37:17 Morinth systemd[1]: Started Home Assistant.
Nov 22 17:37:18 Morinth homeassistant-start.sh[1582]: pyenv-virtualenv: version `homeassistant-3.6.3' is not a virtualenv
Nov 22 17:37:18 Morinth homeassistant-start.sh[1582]: /srv/homeassistant/homeassistant-start.sh: line 5: hass: command not found
Nov 22 17:37:18 Morinth systemd[1]: home-assistant.service: main process exited, code=exited, status=127/n/a
Nov 22 17:37:18 Morinth systemd[1]: Unit home-assistant.service entered failed state.

Fourdee, 2017-11-30:
Confirmed with C2.

Fourdee, 2017-11-30:
```
export PATH="/home/homeassistant/.pyenv/bin:$PATH"
eval "$(pyenv init -)"
eval "$(pyenv virtualenv-init -)"
Downloading Python-3.6.3.tar.xz...
-> https://www.python.org/ftp/python/3.6.3/Python-3.6.3.tar.xz
Installing Python-3.6.3...

BUILD FAILED (Debian GNU/Linux 9 using python-build 20160602)
```

Fourdee, 2017-11-30:
@GiffE 
Many thanks for reporting this issue 👍 

Resolved for v159, verified and tested on C2. Prior to v159 release, simply ```apt-get install build-essential```, prior to installing through DietPi-Software.

Completed.

closed 2017-11-30 by Fourdee