MichaIng/DietPi #1263: DietPi-System | Update images and shrink partition

k-plan, 2017-11-29:
@Fourdee 

> Extremely time consuming to update all our images.

Yes, will do my best to get it run for devices I have here. (e.g. NanoPi Neo, OPi One, Odroid C1, Pine64)
 Crying around doesn't change anything. So let's launch us into image update.

Have playing around a little bit with `finalise`. Nice script, so far so good.
_(Some little quirks, depending on tested devices, but let's move forward)_ 

Have some ideas, how to shrink ext4 partitions after running this.
But better to ask, how you want or will do this, before I go about?
Any tips or hints? 

Thanking you in anticipation.

cu
k-plan

Fourdee, 2017-11-29:
@k-plan 

You'll need to resize partition manually (gparted), then ```zerofree``` it (to 0 free space, reducing 7z size).

I use a desktop (LXDE), and created this bash script:
change ```IMAGE_FP``` and ```IMAGE_NAME``` accordingly.
```
#!/bin/bash
{

	#check packages are installed
	if (( ! $(dpkg -l | grep -ci -m1 'gparted') )); then
		apt-get install gparted zerofree -y
	fi

	#-------------------------------------------------------------------------------
	#MUST LEAVE 50MB+ space or automation autologin may fail due to 0 free space!!!
	#-------------------------------------------------------------------------------

	#-------------------------------------------------------------------------------
	#ARMbian recreating partitions. Boot failure due to UUID change. must also update /boot/armbianEnv.txt... Does not work....
	#-------------------------------------------------------------------------------

	#XU4
	IMAGE_FP='/mnt/samba/#Backups/_Daniel/Projects/DietPi'

	IMAGE_NAME='(NR_4.x)_DietPi_v158_OdroidXU4-armv7-(Jessie)'
	IMAGE_NAME+='.img'

	INDEX_BOOT_PART=1 #rock64 = 6
	INDEX_ROOTFS_PART=2 #rock64 = 7

	if [ ! -f "$IMAGE_FP/$IMAGE_NAME" ]; then

		read -p "No file found..."
		exit

	fi

	modprobe loop
	losetup -f
	losetup /dev/loop2 "$IMAGE_FP/$IMAGE_NAME"
	partprobe /dev/loop2

	e2fsck -f /dev/loop2p$INDEX_BOOT_PART
	e2fsck -f /dev/loop2p$INDEX_ROOTFS_PART

	#Mount loopback.
	# mkdir -p /mnt/loop2p1
	# mkdir -p /mnt/loop2p2
	# mount /dev/loop2p1 /mnt/loop2p1
	# mount /dev/loop2p2 /mnt/loop2p2

	# echo -e "\ncopying data to $HOME\n"
	# rm -R $HOME/loop2p1
	# rm -R $HOME/loop2p2

	# cp -R /mnt/loop2p1 $HOME/
	# cp -R /mnt/loop2p2 $HOME/
	# rm -R $HOME/loop2p1/lost+found
	# rm -R $HOME/loop2p2/lost+found

	# sync

	# umount /mnt/loop2p1
	# umount /mnt/loop2p2

	#Resize partitions
	echo -e "\Resize partitions\n"
	gparted /dev/loop2

	sync

	#transfer back
	mount /dev/loop2p$INDEX_BOOT_PART /mnt/loop2p1
	mount /dev/loop2p$INDEX_ROOTFS_PART /mnt/loop2p2

	# echo -e "\nCopying data back\n"
	# cp -R $HOME/loop2p1/* /mnt/loop2p1/
	# cp -R $HOME/loop2p2/* /mnt/loop2p2/

	read -p "Partitions mounted to /mnt/loop2px, make changes to files if required..."

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2

	#Resize partitions
	# echo -e "Resize partitions"
	# gparted /dev/loop2

	# sync

	read -p "If failed press CTRL+C to exit and prevent further action, else, press any key to continue..."

	zerofree -v /dev/loop2p$INDEX_BOOT_PART
	zerofree -v /dev/loop2p$INDEX_ROOTFS_PART

	losetup -d /dev/loop2

	END_PARTITION=$(( $(fdisk -l "$IMAGE_FP/$IMAGE_NAME" | grep ".img$INDEX_ROOTFS_PART" | awk '{print $3}') + 1 ))
	END_PARTITION=$(( $END_PARTITION * 512 ))

	truncate --size=$END_PARTITION "$IMAGE_FP/$IMAGE_NAME"

	echo -e "\nCleaning up $HOME/loop2px\n"
	rm -R $HOME/loop2p1
	rm -R $HOME/loop2p2

	read -p "Done, press any key to continue..."

	exit

}
```

MichaIng, 2017-11-30:
Just offered a PR and rearranged especially package dealing to actively use apt-mark to get rid of unnecessary packages: https://github.com/Fourdee/DietPi/pull/1266

Would be great to automate it by moving secure steps to finalise step by step. Indeed some steps need careful checks dependent on machine to prevent bootloops, network break in between etc. But should be possibe to solve.

Perhaps there is a way to automize debian base image first run setup also?

k-plan, 2017-11-30:
@Fourdee 

Fiddling around the hole day to shrink a PineA64 image as small as possible.
Use your script and tested as well a gparted-live-0.30.0-1-i686.iso live usb stick.
It's seem where is a "magic" boarder of 1.7 GB with gparted for ext4 partition. Can't get it close to your 1.1 GB in DietPi_v145_PineA64-arm64-(Jessie).img. :cry: 

Have tested here on  `Pine A64+ 1GB (aarch64)` the refresh image with the last longsleep 3.x kernel: ` Linux PineA64 3.10.105-0-pine64-longsleep #3 SMP PREEMPT Sat Mar 11 16:05:53 CET 2017 aarch64 GNU/Linux` .
 Tested some hardware stuff like, Ethernet, wifi, usb, hdmi. For me it works.
`red_led` didn't work and where is a error message on hdmi output, while booting up.
No big deal. Nothing that can't be fix with a future` dietpi-update ` 

Upload to your `dietpi-survey` directory is ready now.

Can you please have a look and give `##DietPi_PineA64-arm64-(Jessie)-by-k-plan.7z` a try?
If it is okay, I will try to do it for some other images from Feb/2017, as far as I own such a device.

Thanks.

k-plan, 2017-12-02:
Can you please have a look and give ##DietPi_PineA64-arm64-(Jessie)-by-k-plan.7z a try?

Will you please delete this. I will try to build a new version.
Thanks.

k-plan, 2017-12-02:
Close.

Please reopen if needed.

closed 2017-12-02 by k-plan

Fourdee, 2017-12-02:
@k-plan 

> Will you please delete this? I will try to build a new version.
Thanks.

Done 👍 . :
```
root@dietpi:~# ls -lha /home/dietpi-survey/ | grep 7z
-rw-r--r-- 1 dietpi-survey dietpi-survey  79M Dec  2 18:23 ##DietPi_OrangePiOne-armv7-(Jessie)_by-kplan.7z
-rw-r--r-- 1 dietpi-survey dietpi-survey 166M Nov 30 21:05 ##DietPi_PineA64-arm64-(Jessie)-by-k-plan.7z
-rw-r--r-- 1 dietpi-survey dietpi-survey 149M Dec  2 21:24 ##DietPi_PineA64-arm64-(Jessie)_by-kplan.7z

root@dietpi:~# rm "/home/dietpi-survey/##DietPi_PineA64-arm64-(Jessie)-by-k-plan.7z"

root@dietpi:~# df -h
Filesystem         Size  Used Avail Use% Mounted on
/dev/ploop11234p1   15G   12G  2.4G  83% /
```