MichaIng/DietPi #1267: DietPi Images | Update/Test RPi image with chroot?

MichaIng, 2017-12-01:
I already tried go get die DietPi RPi image running a virtual arm using qemu. They even support a RPi2 machine now, but it does not work with the image and testing is unbelievable without hardware acceleration.

I have a RPi(2) at home, but it is my running home server also, thus I don't want to shut it down. So I am thinking if it would be possible to mount a usb stick with the image and chroot inside, or create a testing partition on live sd and do the same? I never used chroot, so no idea what is possible. But reading a bid generally running a system via chroot seems possible. One thing that could make problems is the disk/partition naming (was also main problem with qemu I think), but maybe there is a solution also?

As said, I would like to use it for RPi image creation and testing, but production system should stay save.

Fourdee, 2017-12-10:
@MichaIng 

I have absolutely no experience with qemu. I've always preferred to create the image native to minimize potential issues (and inexperience) with qemu.

However, i'am always up for something new, if it shows the ability to speed up our image creation process.

MichaIng, 2018-01-26:
@Fourdee 
Yeah I was able to chroot into a DietPi RPi image on USB stick from SSH on "real" RPi system.
At least `/proc` and `sysfs` need to be remounted to the chroot system to give it some hw access. `/dev/pts` also highly recommended, as otherwise you'll get errors, that terminal is not accessible here and there. Whole `/dev` is necessary to give access to other drives etc.
However the issue somehow is, that I didn't find a way to "reboot" the chroot system without rebooting the whole system. Indeed it is no VM 😉.
So if anyway the base system need to reboot and it is then necessary to take care the drive ids/fstab afterwards, then it is just easier to use a separate sdcard for testing/image creation and shutdown the base server in between. So will close this issue as dead end.

Funny thing, as I left the stick just there: After reboot ~2 weeks ago, I suddenly found my system (via SSH) in kinda 1st run stage, but with otherwise all previously installed software processes running. I was shocked first, but then realized, that the RPi booted into the USB stick (chroot system), with moreless fresh DietPi image inside, but /proc and /sys symlinked to other sdcard system. After shutdown&plugoff it booted luckily into the unmodified sd card system again 😅.

closed 2018-01-26 by MichaIng

WaaromZoMoeilijk, 2022-02-15:
Hey, have you guys found another way to quickly test with chroot?
Just tried aarch64 on proxmox but that was hell...

Thanks for the great project! Let me know if you need any related to hosting (EU) to support the project.

Joulinar, 2022-02-15:
There is an older written how-to proxmox guide on our forum. It might gonna help. But might be to get VMs running without a special architecture user. Like in your case `aarch64`.

https://dietpi.com/phpbb/viewtopic.php?t=8612

But I know @MichaIng using some scripts to be able to spin up container quite easily.

MichaIng, 2022-02-15:
Here is the script I use when I want to boot a raw image as container: https://github.com/MichaIng/hacks/blob/main/boot_img.bash
- It uses `systemd-nspawn` as minimal container platform, more than a `chroot` but less than Docker or similar, of course.
- It also makes use of `qemu-user-static` and `binfmt-support` for automatic emulation when required, i.e. you can boot `aarch64` images on `x86_64` hosts and the other way round, but performance is naturally bad, of course.
- It automatically raises file, partition and filesystem size to 2 GiB, but you can define the size via e.g. `./boot_img.bash /path/to/image.img 4` for 4 GiB.
- It always runs fsck on the contained filesystems.
- Tailored to my needs for testing images or software options, but you may use it as starting point for yours.