MichaIng/DietPi #1270: DietPi-System | Quirk noticed by image update

k-plan, 2017-12-03:
----------
### General

- [ ]  - a typo in `dietpi.txt`
```
# >> Misc Options -----------------------------------
#Size of dphys-swapfile to generate (MB)
#   0=Disabled | 1=auto (1GB-RAM = size) | 2+=manual
```
since v155 size is now set to "2048 - total memory"
so it have to be:
```
# >> Misc Options -----------------------------------
#Size of dphys-swapfile to generate (MB)
#   0=Disabled | 1=auto (2GB-RAM = size) | 2+=manual
```

- [ ]  - missing feature in `dietpi-drive_manager` : exfat support
easy it fix, if you want: `apt-get install exfat-fuse exfat-utils`
(_tested and work on updated images_)

.....

......

------------------

### Dependent on device

- [ ]  - Patching 154 leave package which is no longer required: (OrangePi One, Odroide C1, PineA64)
```
[Info] Patching 153 to 154
 
 The following packages will be REMOVED:
  pciutils*
```

```
apt-get upgrade

Reading state information... Done
Calculating upgrade... The following package was automatically installed and is no longer required:
  libpci3
```
- Fixed in updated images, but a patch will be helpfull

- [ ]  - `dietpi-config`  green_led options make no sense for Allwinner H3 devices:
see: https://github.com/Fourdee/DietPi/issues/1257#issuecomment-347012900

- [ ]  - can't change WiFi country code in dietpi-config
- Odroid C1 => not working with USB Ralink Technology, Corp. RT5370 Wireless Adapter
- PineA64 => working with addon WiFi board
[image]

......

.......

--------------

Fourdee, 2017-12-03:
Thanks @k-plan,

I'll take a look at this after v159. Wrapping up v159 + Stretch currently.

k-plan, 2017-12-03:
@Fourdee 

Okay. 
If you willing to work on it, you can hack down the task list. 👍

MichaIng, 2017-12-17:
@k-plan @Fourdee 
Started to go through the steps, as this goes often hand in hand with preparation script automation. Need some time with my wife now, will go on tomorrow.
https://github.com/Fourdee/DietPi/pull/1312

Fourdee, 2017-12-19:
> Started to go through the steps, as this goes often hand in hand with preparation script automation.

@MichaIng 

Great work, legend 👍 

> Need some time with my wife now

Yep, same here, gotta keep the wife happy 😉 😉

Fourdee, 2018-01-02:
@k-plan 

> dphys-swapfile generation during inital installation process:

👍 Moved:  https://github.com/Fourdee/DietPi/commit/40c2a2ffb6d47f83d623222cbad0fe5f02443d8b

Fourdee, 2018-02-22:
I'll mark this as closed as it appears to apply to <v6.x. Please reopen if required.

closed 2018-02-22 by Fourdee