MichaIng/DietPi #1271: DietPi-System | ARMbian sources.list change

k-plan, 2017-12-03:
@Fourdee 

While updating the last image (NanoPi-Neo), this error occur:

```
:~# apt-get update
...
W: Failed to fetch http://apt.armbian.com/dists/jessie/InRelease  Unable to find expected entry 'utils/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

```
:~# cat /etc/apt/sources.list.d/armbian.list
deb http://apt.armbian.com jessie main utils jessie-desktop

```

http://apt.armbian.com/dists/jessie/InRelease

```
Origin: Armbian
Label: Armbian
Suite: jessie
Codename: jessie
Date: Sun, 3 Dec 2017 12:13:49 UTC
Architectures: arm64 armhf
Components: main
Description: Generated by aptly
...
```

What shall/can we do? Remove `utils jessie-desktop` from the list?
Cross-check with last night build image on OrangePi One which was updated without any error:

Now it's the same error:

```
root@OrangePi-One:~# apt-get update
...
Fetched 845 kB in 23s (36.2 kB/s)
W: Failed to fetch http://apt.armbian.com/dists/jessie/InRelease  Unable to find expected entry 'utils/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.

```

😢   😠   💩   👎 

`k-plan@DietPi:~# ... stop building update images for now! Please help ....`

Fourdee, 2017-12-03:
@k-plan 

Try:
```
deb http://apt.armbian.com jessie main jessie-utils jessie-desktop
```

Failing that, we can remove ```utils``` and ```jessie-desktop``` via patch.

k-plan, 2017-12-03:
@Fourdee 
> `deb http://apt.armbian.com jessie main jessie-utils jessie-desktop`

```
root@OrangePi-One:~# cat /etc/apt/sources.list.d/armbian.list
deb http://apt.armbian.com jessie main jessie-utils jessie-desktop
```

```
W: Failed to fetch http://apt.armbian.com/dists/jessie/InRelease  Unable to find expected entry 'jessie-utils/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

For now, I only see main/... in [InRelease](http://apt.armbian.com/dists/jessie/InRelease)

```
main/Contents-arm64.gz
main/Contents-armhf.gz
main/binary-arm64/Packages
main/binary-arm64/Packages.bz2
main/binary-arm64/Packages.gz
main/binary-arm64/Release
main/binary-armhf/Packages
main/binary-armhf/Packages.bz2
main/binary-armhf/Packages.gz
main/binary-armhf/Release
```

> Failing that, we can remove utils and jessie-desktop via patch.

👍  Nice! 
Shall/can I do this in updated images by hand?

k-plan, 2017-12-03:
@Fourdee 

do some reseach: https://forum.armbian.com/topic/5355-unable-to-find-expected-entry-jessie-utilsbinary-armhfpackages-in-release-file/

I think, they have f... up on of their repository server.

https://apt.armbian.com/dists/jessie/

[image]

http://apt.uk.armbian.com/dists/jessie/

[image]

Have done a quick look into their last `Armbian_5.35_Orangepione_Debian_jessie_default_3.4.113`

```
root@OrangePi-One:~# cat /etc/apt/sources.list.d/armbian.list
deb http://apt.armbian.com jessie main jessie-utils jessie-desktop

```
should be fine.

👍

k-plan, 2017-12-04:
Nice,  repository server is fixed. :smiley: 

https://apt.armbian.com/dists/jessie/

```
Origin: Armbian
Label: Armbian
Suite: jessie
Codename: jessie
Date: Mon, 4 Dec 2017 12:08:18 UTC
Architectures: arm64 armhf
Components: jessie-desktop jessie-utils main utils
Description: Generated by aptly
```
Will test it tomorrow with:  `deb http://apt.armbian.com jessie main jessie-utils jessie-desktop`

:+1:

k-plan, 2017-12-07:
> Will test it tomorrow with: `deb http://apt.armbian.com jessie main jessie-utils jessie-desktop`

- Done
- Works
- building updated ARMbian images now

Close.

closed 2017-12-07 by k-plan