MichaIng/DietPi #1273: DietPi-Software | FFMPEG E: Unable to correct problems, you have held broken packages.

Beaujr, 2017-12-06:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```158```?
- Odroid XU 4
- 5V 6A
- PNY Elite-X U3 32GB
- Distro (EG: Jessie) | ```Linux DietPi 3.10.105+ #1 SMP PREEMPT Mon Aug 7 10:09:24 CEST 2017 armv7l GNU/Linux```?

#### Additional Information (if applicable):
- FFMPEG
- Can this issue be replicated on a fresh installation of DietPi? YES.

Unable to install FFMPEG from dietpi-software

#### Expected behaviour:
ffmpeg installed
#### Actual behaviour:
apt-get error
The following packages have unmet dependencies:
 ffmpeg : Depends: libavcodec57 (>= 7:3.4) but it is not going to be installed or
                   libavcodec-extra57 (>= 7:3.4) but it is not going to be installed
          Depends: libavdevice57 (>= 7:3.4) but it is not going to be installed
          Depends: libavfilter6 (>= 7:3.4) but it is not going to be installed or
                   libavfilter-extra6 (>= 7:3.4) but it is not going to be installed
          Depends: libavformat57 (>= 7:3.4) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
#### Steps to reproduce:
Install ffmpeg from dietpi-software
#### Did you submit a ```dietpi-bugreport```?

c67ca81e-3042-456c-af0a-127222a9df27-0
#### Extra details:

MichaIng, 2017-12-06:
The reason is that `ffmpeg` just exists in Stretch and Jessie backports: https://packages.debian.org/stretch/ffmpeg
As backports are by default added to apt sources, but handled with low priorities, apt requests recognize it to install, but finds too low versions of its dependencies inside the (stable) Jessie repo.

So you need to force that apt also pulls the dependencies from jessie-backports repo, or let's say DietPi need to ensure that, when offering backport packages.

Use: `apt -t jessie-backports install ffmpeg`

Fourdee, 2017-12-07:
@MichaIng 

>  Hmm, seems to be handles different in cases. On my VirtualBox Jessie image, the dependencies from jessie-backport got pullled automatically even withou -t jessie-backports

Confirmed on my XU4 👍

Fourdee, 2017-12-07:
I'll apply the fix for XU4 Jessie for now. If we see this error on other devices/distros, we'll need to investigate a wider fix.

Fourdee, 2017-12-07:
Tested on XU4, completed.

MichaIng, 2017-12-07:
@Fourdee 
> Package: ffmpeg
> Pin: release a=jessie-backports
> Pin-Priority: 600

Does this also ensure that dependencies are pulled from backports? I remember that for pulling redis-server from stretch-backports on RPi, I needed to add it's dependency redis-tools also to the preferences package list.

Fourdee, 2017-12-07:
@MichaIng 

> Does this also ensure that dependencies are pulled from backports? I remember that for pulling redis-server from stretch-backports on RPi, I needed to add it's dependency redis-tools also to the preferences package list.

Yep, 600 is a higher priority:
```
root@DietPi:~# apt-cache policy redis-server
redis-server:
  Installed: (none)
  Candidate: 3:3.2.8-2~bpo8+1
  Version table:
     3:3.2.8-2~bpo8+1 0
        600 http://ftp.debian.org/debian/ jessie-backports/main armhf Packages
     2:2.8.17-1+deb8u5 0
        500 http://ftp.debian.org/debian/ jessie/main armhf Packages
        500 http://security.debian.org/ jessie/updates/main armhf Packages
```

Before hand:
```
root@DietPi:~# apt-cache policy redis-server
redis-server:
  Installed: (none)
  Candidate: 2:2.8.17-1+deb8u5
  Version table:
     3:3.2.8-2~bpo8+1 0
        100 http://ftp.debian.org/debian/ jessie-backports/main armhf Packages
     2:2.8.17-1+deb8u5 0
        500 http://ftp.debian.org/debian/ jessie/main armhf Packages
        500 http://security.debian.org/ jessie/updates/main armhf Packages
```

So its probably best I lower the Meveric repo, instead of increasing the backports.

Fourdee, 2017-12-07:
🈯️ Tested:
```
root@DietPi:~# apt-cache policy ffmpeg
ffmpeg:
  Installed: (none)
  Candidate: 7:3.2.5-1~bpo8+1
  Version table:
     7:3.4-3~bpo8 0
         99 http://fuzon.co.uk/meveric/ jessie/backports armhf Packages
     7:3.2.5-1~bpo8+1 0
        100 http://ftp.debian.org/debian/ jessie-backports/main armhf Packages
```

closed 2017-12-07 by Fourdee

MichaIng, 2017-12-07:
Ah, meveric repo explains different behaviours.

But I am wondering, as now I find all dependencies including right versions on meveric: http://fuzon.co.uk/meveric/dists/jessie/backports/binary-armhf/Packages

Maybe this was just a temporary issue during updating ffmpeg and all related dependency packages?

Fourdee, 2017-12-09:
@MichaIng 

> Maybe this was just a temporary issue during updating ffmpeg and all related dependency packages on repo?

Appears that way, hopefully. We'll soon find out and be able to resolve if other software titles are effected by this change.