MichaIng/DietPi #1277: NZBDrone Error 100 with Sonarr on Odriod C2

sunnythaper, 2017-12-08:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
158

- SBC Device (EG: RPi 3)?
Odroid C2

- Power supply used (EG: 5v 1A RAVpower)?
Manufacturer supplied power supply

- SD card used (EG: Sandisk ultra)?
Sandisk Ultra 32gig

- Distro (EG: Jessie) | ```uname -a```?
Linux PlexServer 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Additional Information (if applicable):
- Software title?
Sonarr

- Can this issue be replicated on a fresh installation of DietPi?
Unknown, wasn't willing to try. I do have other Dietpi services that do currently operate quite well for me.

#### Expected behaviour:
Upon "dietpi-software install 144" I should be able to complete the install.

#### Actual behaviour:
Error code 100 saying NZBdrone cannot be found.

#### Steps to reproduce:
dietpi-software install 144

#### Did you submit a ```dietpi-bugreport```?
24ca1211-e327-4b84-852d-050db039b816-0  

#### Extra details:
Other than standard Dietpi-software installs, I have left this unit alone completely. I was looking for the easiest no fuss solution so I didn't exactly go out of my way to do anything non-standard.

opnwrt4lede, 2017-12-08:
Just to confirm. Have the exactly same Error message (Error 100). My SBC Device is a OPiPC2.

Tried two times - same result:

                                                           │ Apt-get has failed:                                                │                                                           
                                                          │  - Apt string = nzbdrone                                           │                                                           
                                                          │  - Error code = 100                                                │                                                           
                                                          │  - Software Title = Sonarr: automatically download TV shows        │                                                           
                                                          │  - Apt log = /var/log/dietpi-software_apt.log                      │                                                           
                                                          │                                                                    │                                                           
                                                          │ Please report this issue to DietPi (INCLUDING THE APT LOG) so it   │                                                           
                                                          │ can be investigated.                                               │                                                           
                                                          │                                                                    │                                                           
                                                          │ DietPi-Software will now exit.    

```
E: Unable to locate package nzbdrone

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:  - https://github.com/Fourdee/DietPi/issues
```

It tried the suggestions but no luck.

sunnythaper, 2017-12-08:
@opnwrt4lede thanks for confirming! I've tried the suggestions as well and still have the issue :/

Fourdee, 2017-12-08:
@sunnythaper 

Thanks for the report.

Just ran a quick test on:
- 🈯️ Odroid C2 + Stretch, working fine. Must be limited to Jessie.

sunnythaper, 2017-12-08:
Thank you, @Fourdee !

Fourdee, 2017-12-30:
@sunnythaper 

Apologies for the delay. Can you try running:
```
dpkg --add-architecture armhf
dietpi-software install 144
```

sunnythaper, 2017-12-31:
@Fourdee apologize but since this post, I have switched away from DietPi and am just running the individual apps I needed as Docker containers on my oDroid. I guess I wanted Sonarr that badly :P

Fourdee, 2018-01-01:
@sunnythaper 

No worries.

Confident ```dpkg --add-architecture armhf``` should resolve this. Passed testing. Marking as fixed in v6.0.

closed 2018-01-01 by Fourdee