MichaIng/DietPi #1281: Restructure of supported images

Fourdee, 2017-12-09:
> https://github.com/Fourdee/DietPi/issues/1257#issuecomment-350415601

For discussion/consideration.

MichaIng, 2017-12-09:
I just created a single image so far, so have limited experience about how much work it does to support that much devices.
Indeed it took some time to create the image, but also I needed to learn how to deal with Debian installation questions and other things, one would have known, creating images more often.

But my impression is, that one special strength about DietPi is, that it does support so much different devices. The main developer of [NextcloudPi](https://github.com/nextcloud/nextcloudpi) already shared a thought about switching to DietPi as a grounding platform, because of that. Obviously it is not easy to provide software well working on different hardware, especially if it's about Linux, ARM etc. and involving sometimes many software working together. But we gathered now many experience here about that, and my impression is that it worked not too bad. The most issues occurred not because of different devices, but because of issues "outside" of DietPi, problems with PPAs, apt package issues, changed software commands etc. Those issues will occur also, if we support less devices and are solvable mostly for all together at one.

Of course image creation is a topic. If we focus on that, trying to automate the process as much as possible, this would be great. Also it would be nice to find other people, owning a certain device, willing to regularly create an test the related image. But that's another topic again...

**Long story short, supporting many different devices is a strength and not a weakness, and we first should thing about how to reduce the workload, this produces (mainly image creation), before dropping it completely.**

Fourdee, 2017-12-10:
@MichaIng 

👍 

Maybe a solution would be to class these images as **lower priority**, and make user aware of?:
- These images may experience issues, related to kernel bugs/drivers outside of our control, and their stability is not guaranteed
- No GPU/I2S/I2C supported
- Limited support for these images and specific issues related to the device.
- These Images will not be updated regularly, and may be 10-20 versions outside of current DietPi. However, 1st run will update to latest version (may take a while).

k-plan, 2017-12-10:
@MichaIng 

> I just created a single image so far, so have limited experience about how much work it does to support that much devices.

Do this for a VM is relatively fast, if you don't own ten years old hardware like me.
And if I remenber correct, VM are appears relative late in DietPi timeline and was in bottom line a helping tool for developer, not for enduser.

Messing around with several SBC, SD card, old images versions be a mess, a very slow and time-consuming porcess.
I can send you a OrangePi One and some SD card, so you can get an idea of it.
If something goes wrong or you slip up, you have so start over again and again.
And if you are into the last period of life like me, you don't have the time, patience and nerves to do this very often.

> If we focus on that, trying to automate the process as much as possible, this would be great.

Yea! 💯  👍 
Waiting for your scripts, program, system, ... or what ever it will be.

> Also it would be nice to find other people, owning a certain device, willing to regularly create an test the related image

Yes. 
Trust in me, we have try this in the past but without consistent success.

Und, damit das nicht falsch rüber kommt, ich persönlich bin mehr als froh, dass es im Moment so gut läuft, etwas Bewegung in die Sache kommt, ein Überdenkungsprozess begonnen hat und du Dich in dem Gesamtprojekt mit so großen persönlichen Einsatz einbringst. Danke dafür.

k-plan, 2017-12-10:
The main problem in my eyes is things have change over the time so many times:

- from all RPi's plus all Odroids over to support every low-cost china rubbish one-man-show SBC and now new player on the market like Sparky, Asus Tinkerboard, Pine, Allo, Just-Boom,  plus Native PC ( + UEFI) , varying VM's

- changes on software focus from time to time, 
over form basic stuff and very cool project specifically SBC (eg. pihole, VPN's, Hotspots, DietPi-Cloudshell, ...) , 
to media suite like Kodi, Plex, ... , 
to Desktop and Remote like MATE, Chromium, NoMachine, ... ,
to Gaming like AmiBerry, Retro-Pi, Cuberite, Steam, ...
to File Sever like Next/OwnCloud, Tonido, Pydio, OpenMediaVault, Syncthing, ... ,
to Torrent / Download Tools like rTorrent, SickRage, Sonarr, CouchPotato, ... ,
and lately to Audio stuff like Roon, Allo, JustBoom, moOde, ....

... but by all changes, we never discontinue support for historical stuff and devices.

At the beginning it was not a big problem, because @Fourdee has enough time and energy to keep hold all the balls rolling but nowadays this has change as well.

MichaIng, 2017-12-10:
@Fourdee
Do you have download statitics of images?
I am personally not a friend of data collection, but it would be interesting to know which software is actually used how much? Any other way we can derive that?

k-plan, 2017-12-10:
@MichaIng 

> I am personally not a friend of data collection, but it would be interesting to know which software is actually used how much? Any other way we can derive that?

Yes we, I or you can do this by [DietPi Survey](http://dietpi.com/phpbb/viewtopic.php?f=8&t=20&sid). But, as always, it's very time consuming to get a statistic out of this.
Shall I do this, if I get some spare time?

MichaIng, 2017-12-10:
"Shall I do this, if I get some spare time?"
Ah great, yes, if we already thinking about dropping ancient software, it should be a helpful guideline.

Generally, I like the idea to mark certain devices and software as low priority/no active development and imform users about this (instead of completely remove it). That way we can also ask for maintainers, people who use the device/software and are willing to at least help maintaining it. If someone is found, we can unmark the device/software as long as maintainer is there.

I e.g. will check in for Nextcloud+ownCloud, RPi (at the moment just testing, figuring out a way I can create image without disturbing production system) and VirtualBox Image creation+testing.
I like to contribute where I can and where I find time on everything, but reliably only about stuff I actively use for now :wink:.

k-plan, 2017-12-10:
> Do you have download statitics of images?

I like as well to see this.

And please as well a traffic statistic for the past 12 month `dietpi.com` generate.

But only @Fourdee  can share this information with us.

k-plan, 2017-12-10:
@MichaIng 

First device statistic by `DietPi Survey` 
But please keep in mind, only devices will be count which:

- users have decide to share this information with us
- devices must be online and have a working internet connection

--------------

cute-of date: 10-12-2017

-------------

### Debian Releases

**Debian 8  "Jessie":**
56.740

**Debian 9  "Stretch":**
1.162

### DietPi Version:

**Used on devices:** 
from v102 to v159

- "DietPi Version : 100 - 109" :
1

- "DietPi Version : 110 - 119" :
5

- "DietPi Version : 120 -129" :
100

- "DietPi Version : 130 -139" :
267

- "DietPi Version : 140 -149" :
13.779

- "DietPi Version : 150 -159" :
43.836

- "DietPi Version : 150" :
3.671

- "DietPi Version : 151" :
3.520

- "DietPi Version : 152" :
3.601

- "DietPi Version : 153" :
3435

"DietPi Version : 154" :
8.095

- "DietPi Version : 155" :
4.041

- "DietPi Version : 156" :
5.811

- "DietPi Version : 157" : _(current)_
5.523

- "DietPi Version : 158" : _(testing)_
6.106

- "DietPi Version : 159" :
33

--------------

### All Devices: 
**58.113**

----------------

### All "Raspberry Pi"  Devices:
**36.191**

Device: "RPi "  
3.5875

Device: "RPi A "  
35

Device: "RPi A+ "  
63

Device: "RPi B "  
4.650

Device: "RPi B+ "  
1.717

Device: "RPi 2 "  
6.306

Device: "RPi 3 "  
19.425

Device: "RPi Zero W"  
2.607

Device: "RPi Zero "  
866

--------------

### ALL "oDroid " Devices: 
**4.220**

Device: "oDroid XU3/4 "  
1.018

Device: "oDroid C1 "  
560

Device: "oDroid C2 "  
2.639

Device: "oDroid U3 "  
3

------------------

### All "OrangePi "  Devices: 
**9.012**

Device: "OPi PC2 "  
10

Device: "OPi Prime "  
90

Device: "OPi Win "  
179

Device: "OPi Zero 2 "  
285

Device: "OrangePi Plus "  
557

Device: "OrangePi Lite "  
289

Device: "OrangePi Zero "  
4.842

Device: "OrangePi One "  
853

Device: "OrangePi PC "  
1.904

---------

### All "BananaPi " Devices:
**579**

Device: "BananaPi M2+ "  
77

Device: "BananaPi Pro "  
502

------------

### All "Pine A64" Devices: 
**1.234**

Device: "Pine A64 512MB"  
57

Device: "Pine A64+ 1GB"  
278

Device: "Pine A64+ 2GB"  
899

------------

### All exotic or new devices:
**884**

Device: "**RoseapplePi** "  
10

Device: "**Asus Tinker Board** "  
665

Device: "**A20-OLinuXino-MICRO** "  
0

Device: "**Cubieboard 3** "  
0

Device: "**Beagle Bone Black** "  
4 + my device 👍 

Device: "**Sparky SBC** "  
205

-----------

k-plan, 2017-12-10:
@MichaIng 
> but reliably only about stuff I actively use for now

```
id 114 | =2 | nextcloud:
43.396

id 47 | =2 | owncloud:
2.216

```
wow ... :flushed:

Hmmm ... :pensive: 

```
id 93 | =2 | pi-hole:
7.337
```
😃 

😉

Fourdee, 2017-12-11:
@k-plan 
NB: Survey files are from March 2017 - Present day :)

Fourdee, 2017-12-11:
@k-plan 
Webserver report, from 10th Dec - Present Day:
http://dietpi.com/report.html

k-plan, 2017-12-11:
@Fourdee 
> NB: Survey files are from March 2017

yes, correct 21-03-2017

> Present Day (11th):

no, 10-12-2017 18:31 , because I have to download them to get it seached ... no right to do this on your server. And please let it as it is. It's okay.

Thanks for the report! 👍

Fourdee, 2017-12-11:
@k-plan 

I changed ```.``` in your post, hope you do not mind? 11.890 users is 11 users lol (from my eyes) 😄 

> no right to do this on your server.

You have every right 👍

k-plan, 2017-12-11:
> 11.890 users is 11 users lol

https://de.wikipedia.org/wiki/Wikipedia:Schreibweise_von_Zahlen#Dezimaltrennzeichen

We don't live on a island, but is okay! 😄 

Was only decimal point to get it more readable here. But for sure :gb next time. Promised.

MichaIng, 2017-12-11:
Do I interpret this right? Average DietPi user is RPi owner, running Nextcloud? Hmm I am soo totally average it seems 😭 😆.

Note to myself: Opt out survey with my testing VMs 😉.

Fourdee, 2018-01-09:
Marking as closed, in place of: https://github.com/Fourdee/DietPi/issues/1385

closed 2018-01-09 by Fourdee