MichaIng/DietPi #1282: DietPi-Software | Pi-Hole - Lost connection to API

Fourdee, 2017-12-09:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=2545&p=9849#p9847

- Revisit installation. Appears alot of changed with PiHole and its install.
```
/DietPi/dietpi/dietpi-software: line 10239: pihole: command not found
```
- Install after webservers, to avoid PiHole pre-installing deps?
- Loop install script, request user confirmed competition. Prevents accidental exit of PiHole script assuming its ok.

k-plan, 2017-12-09:
@Fourdee 

> - Revisit installation. Appears alot has changed with PiHole and its install.

Yes, correct.

Sorry, habe wenig Zeit im Moment. Daher auf Deutsch. Musst halt ein wenig nach Übersetzung googlen. Geht mir auch meist so. 😃 

Was ich bis dato feststellen konnte, ist das Problem ein wenig vielschichtiger.
Bis Pi-Hole v3.1x hat das erneute anstoßen der pihole Installation über `dietpi-software` immer funktioniert.
- `pihole -r` funktionierte nie.
- `pihole -up` funktionierte nur, wenn die FTL Version alleine zu aktualisieren war. 
Sonst wurde die gesamte Installation zerschossen

Mit der neusten Pi-Hole v3.2 scheint es mir, das weitere Probleme hinzu gekommen sind.

> `/DietPi/dietpi/dietpi-software: line 10239: pihole: command not found`

Ich habe da noch einige weitere Meldungen:

`/DietPi/dietpi/dietpi-software: line 10084: /usr/local/bin/pihole: No such file or directory`

Ich deute das so, dass dies nicht funktionieren kann, da pihole nicht vollständig installiert ist.

```
root@NanoPi-Neo:~# /etc/.pihole/pihole -r
/etc/.pihole/pihole: Zeile 15: /opt/pihole/COL_TABLE: Datei oder Verzeichnis nicht gefunden
```

Mein bis jetziger Verdacht ist, es könnte etwas mit `php5-sqlite` zu tun haben und den Optimierungen die DietPi bei der Web Server Installation an php5 durchgeführt hat.
Aber mir fehlte im Moment die Zeit, da tiefer zu graben und die pihole v3.2 ist noch brandneu.

```
  [✓] Root user check
  [✓] Disk space check

  [✓] Update local cache of available packages

  [✓] Checking apt-get for upgraded packages... up to date!

  [i] Installer Dependency checks...
  [✓] Checking for apt-utils
  [✓] Checking for dialog
  [✓] Checking for debconf
  [✓] Checking for dhcpcd5
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for whiptail

  [✓] Stopping dnsmasq service...

  [✓] Stopping lighttpd service...
  [i] Using   [i] Static IP already configured
  [i] Found IPv6 GUA address, using it for blocking IPv6 ads
  [i] IPv4 address: 192.xx.xx.100/24
  [i] IPv6 address: 2001:xx:xx:xx:c4ff:fe0a:415b
  [i] Web Interface On
  [i] Logging On.
  [i] Performing reconfiguration, skipping download of local repos
  [✓] Resetting repository within /etc/.pihole...
  [✓] Resetting repository within /var/www/html/admin...
  [i] Main Dependency checks...
  [✓] Checking for bc
  [✓] Checking for cron
  [✓] Checking for curl
  [✓] Checking for dnsmasq
  [✓] Checking for dnsutils
  [✓] Checking for iputils-ping
  [✓] Checking for lsof
  [✓] Checking for netcat
  [✓] Checking for sudo
  [✓] Checking for unzip
  [✓] Checking for wget
  [✓] Checking for idn2
  [✓] Checking for lighttpd
  [✓] Checking for php5-common
  [✓] Checking for php5-cgi
  [✗] Checking for php5-sqlite (will be installed)

```

- Testing:

```
root@NanoPi-Neo:~# apt-get install php5-sqlite
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
php5-sqlite ist schon die neueste Version.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
1 nicht vollständig installiert oder entfernt.
Nach dieser Operation werden 0 B Plattenplatz zusätzlich benutzt.

```

- und ein weitere Test auf einem RPi:

```
DietPi-Software
───────────────────
 Mode: Update & upgrade APT
 Please wait...

Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libraspberrypi-bin libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up php5-sqlite (5.6.30+dfsg-0+deb8u1) ...

Creating config file /etc/php5/mods-available/sqlite3.ini with new version
php5_invoke: Enable module sqlite3 for cli SAPI
php5_invoke: Enable module sqlite3 for fpm SAPI
php5_invoke: Enable module sqlite3 for cgi SAPI

Creating config file /etc/php5/mods-available/pdo_sqlite.ini with new version
php5_invoke: Enable module pdo_sqlite for cli SAPI
php5_invoke: Enable module pdo_sqlite for fpm SAPI
php5_invoke: Enable module pdo_sqlite for cgi SAPI
Processing triggers for php5-fpm (5.6.30+dfsg-0+deb8u1) ...
insserv: warning: current start runlevel(s) (empty) of script `php5-fpm' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `php5-fpm' overrides LSB defaults (0 1 6).
insserv: warning: current start runlevel(s) (empty) of script `php5-fpm' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `php5-fpm' overrides LSB defaults (0 1 6).
 [Info] Pi-hole: Requires user input and can not be automated.
 [Failed] Pi-hole: Will not be installed. Please run 'dietpi-software' to install.
```

MichaIng, 2017-12-09:
@Fourdee 
@k-plan 
Jep fails, because apt php-cgi and php-sqlite3 fails somehow to install.
```
  [✗] Checking for php-cgi (will be installed)
  [✗] Checking for php-sqlite3 (will be installed)
```
I just did manual apt install php-cgi php-sqlite3 and after that it worked, including working web interface etc:
```
./install.sh

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [✓] Root user check
  [✓] Disk space check

  [✓] Update local cache of available packages

  [✓] Checking apt-get for upgraded packages... up to date!

  [i] Installer Dependency checks...
  [✓] Checking for apt-utils
  [✓] Checking for dialog
  [✓] Checking for debconf
  [✓] Checking for dhcpcd5
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for whiptail

  [✓] Stopping dnsmasq service...

  [✓] Stopping lighttpd service...
  [i] Using   [i] Static IP already configured
  [i] IPv4 address: 192.168.178.29/24
  [i] IPv6 address:
  [i] Web Interface On
  [i] Logging On.
  [✓] Check for existing repository in /etc/.pihole
  [✓] Update repo in /etc/.pihole

  [✓] Check for existing repository in /var/www/html/admin
  [✓] Update repo in /var/www/html/admin

  [i] Main Dependency checks...
  [✓] Checking for bc
  [✓] Checking for cron
  [✓] Checking for curl
  [✓] Checking for dnsmasq
  [✓] Checking for dnsutils
  [✓] Checking for iputils-ping
  [✓] Checking for lsof
  [✓] Checking for netcat
  [✓] Checking for sudo
  [✓] Checking for unzip
  [✓] Checking for wget
  [✓] Checking for idn2
  [✓] Checking for lighttpd
  [✓] Checking for php-common
  [✓] Checking for php-cgi
  [✓] Checking for php-sqlite3

  [✓] Enabling lighttpd service to start on reboot
  [✓] Installing scripts from /etc/.pihole

  [i] Installing configs from /etc/.pihole...
  [i] Existing dnsmasq.conf found... it is not a Pi-hole file, leaving alone!
  [✓] Copying 01-pihole.conf to /etc/dnsmasq.d/01-pihole.conf

  [i] Creating log and changing owner to dnsmasq... log already exists!

  [i] Installing blocking page...
  [✓] Creating directory for blocking page, and copying files
  [✓] Backing up index.lighttpd.html

  [✓] Installing sudoer file

  [✓] Installing latest Cron script

  [✓] Installing latest logrotate script

  [i] FTL Checks...
  [✓] Detected x86_64 architecture
  [i] Checking for existing FTL binary...
  [✓] Downloading and Installing FTL

  [i] No active firewall detected.. skipping firewall configuration
  [i] Restarting services...

  [✓] Starting dnsmasq service

  [✓] Enabling dnsmasq service to start on reboot

  [✓] Starting lighttpd service

  [✓] Enabling lighttpd service to start on reboot

  [✓] Starting pihole-FTL service

  [✓] Enabling pihole-FTL service to start on reboot

  [i] Preparing to run gravity.sh to refresh hosts...
  [i] Running gravity.sh
  [i] Neutrino emissions detected...
  [✓] Pulling blocklist source list into range

  [i] Target: raw.githubusercontent.com (hosts)
  [✓] Status: Retrieval successful

  [i] Target: mirror1.malwaredomains.com (justdomains)
  [✓] Status: Retrieval successful

  [i] Target: sysctl.org (hosts)
  [✓] Status: Retrieval successful

  [i] Target: zeustracker.abuse.ch (blocklist.php?download=domainblocklist)
  [✓] Status: Retrieval successful

  [i] Target: s3.amazonaws.com (simple_tracking.txt)
  [✓] Status: Retrieval successful

  [i] Target: s3.amazonaws.com (simple_ad.txt)
  [✓] Status: Retrieval successful

  [i] Target: hosts-file.net (ad_servers.txt)
  [✓] Status: Retrieval successful

  [✓] Consolidating blocklists
  [✓] Extracting domains from blocklists
  [i] 129,154 domains being pulled in by gravity
  [✓] Removing duplicate domains
  [i] 106,012 unique domains trapped in the Event Horizon

  [✓] Adding 6 blocklist source domains to the whitelist
  [✓] Whitelisting 0 domains
  [✓] Parsing domains into hosts format
  [✓] Cleaning up stray matter

  [✓] Force-reloading DNS service
  [✓] DNS service is running
  [✓] Pi-hole blocking is Enabled
  [i] Web Interface password: AUOdjv_M
      This can be changed using 'pihole -a -p'

  View the web interface at http://pi.hole/admin or http://192.168.178.29/admin

  You may now configure your devices to use the Pi-hole as their DNS server
  [i] Pi-hole DNS (IPv4): 192.168.178.29
  If you set a new IP address, please restart the server running the Pi-hole

  [i] The install log is located at: /etc/pihole/install.log
  Installation Complete!
```
So solution is to simply install those two packages (like above on Stretch, php5-cgi + php5-sqlite3 on Jessie) before running `./install.sh`

Fourdee, 2017-12-09:
notes:
```
E: Package 'php5-sqlite3' has no installation candidate

root@DietPi:~# apt-cache search sqlite | grep php
php-db - PHP PEAR Database Abstraction Layer
php5-sqlite - SQLite module for php5 # <<?

```

Fourdee, 2017-12-09:
@MichaIng 

Legend :+1: https://github.com/Fourdee/DietPi/issues/1282#issuecomment-350490524

Completed.

closed 2017-12-09 by Fourdee