MichaIng/DietPi #1283: Transmission not stable / crashes with 100+ torrents

oakbrad, 2017-12-09:
## Creating a bug report/issue:

Transmission runs fine until I add over 100+ torrents, and then becomes unstable and crashes constantly. These torrents are pretty much all inactive, not actively downloading or uploading (long term seeding). 

#### Required Information:
- DietPi Version | 158, although same behavior on 157 with the older version of Transmission
- SBC Device (EG: RPi 3)? Odroid XU4
- Power supply used (EG: 5v 1A RAVpower)? hardkernel provided
- SD card used (EG: Sandisk ultra)? EMMC, didn't work running off SD either
- Distro (EG: Jessie) | Debian 8.9 (Linux odroid 3.10.105+ #1 SMP PREEMPT Mon Aug 7 10:09:24 CEST 2017 armv7l GNU/Linux) 
#### Additional Information (if applicable):
- Software title? Transmission
- Can this issue be replicated on a fresh installation of DietPi? yes, did a fresh install twice trying to fix this

#### Expected behaviour:
It should just seed, I've never had any problems with Transmission on any platform regardless of number of torrents seeding
#### Actual behaviour:
Crashes with nothing in the logs

root@plex:/mnt# service transmission-daemon status
● transmission-daemon.service - Barebones transmission-daemon service
   Loaded: loaded (/etc/systemd/system/transmission-daemon.service; disabled)
   Active: active (exited) since Thu 2017-12-07 22:56:40 PST; 20h ago
  Process: 26657 ExecStop=/usr/bin/killall -w transmission-daemon (code=exited, status=1/FAILURE)
  Process: 26659 ExecStart=/usr/bin/transmission-daemon --config-dir /var/lib/transmission-daemon/info (code=exited, status=0/SUCCESS)
 Main PID: 26659 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/transmission-daemon.service

#### Steps to reproduce:
Install transmission, add more than a few torrents for seeding. Works as expected with small number of torrents.

#### Did you submit a ```dietpi-bugreport```?
Nope

#### Extra details:
I've tried reducing the number of peers etc in the settings.json file but I'm obviously not finding the magic combo here. Any help greatly appreciated.

Fourdee, 2018-01-28:
@oakbrad 

Apologies for the delay. Its most likely memory limitation, although we do set a max value for cache size, 100+ items is asking alot 😃 

As this is an old ticket, i'll mark as closed, however, if you are still experiencing the issue, please reopen and check ```htop``` for memory usage during the issue, paste screenshot here. We can then investigate.

closed 2018-01-28 by Fourdee

oakbrad, 2018-01-28:
Thanks for the response. I have been tinkering with this for a while and have been able to get it a lot more stable by just turning the maximum connections overall and per-torrent in Transmission way, way down from the defaults. This is fine because I'm just seeding stuff on private trackers that have few peers to begin with. I have about 400 torrents seeding now.

from my settings.json:

    "max-peers-global": 15,
    "peer-limit-global": 200,
    "peer-limit-per-torrent": 10,
    "upload-slots-per-torrent": 4,

Not sure which of those is having the desired effect but it's only going down very infrequently now.

Zac610, 2018-01-29:
This seems a software issue and not dietpi related, I bet 100% of other software wolud crash if user input is above a certain sane level.