MichaIng/DietPi #1285: PREP_SYSTEM_FOR_DIETPI.sh | Automate

Fourdee, 2017-12-10:


MichaIng, 2017-12-10:
Just in theory: If we really get this automated secure for at least some devices, this means we do not necessarily need to create images for those. We could just provide the preparation script to upgrade any Debian system into DietPi. By use of apt-mark, as suggested here: https://github.com/Fourdee/DietPi/pull/1266, we would be sure that possibly disturbing software is removed and provide our own config files/adjustments for our dependencies. Kernel/bootleader apt packages are generally excluded from autoremove on every device I checked.
As running systems are, yeah running, we could extract boot critical information (especially fstab/boot.ini/network details) from the running system in case and create our configs around those.

Hehe just a little optimistic view into the feature 😆.

Fourdee, 2017-12-10:
@MichaIng 

> As running systems are, yeah running, we could extract boot critical information (especially fstab/boot.ini/network details) from the running system in case and create our configs around those.
Hehe just a little optimistic view into the feature

I like where this is going 👍

Fourdee, 2017-12-12:
@MichaIng 

I've not looked into manual holding packages yet. I'am simply converting the old script into automated. Then we can work on it.

MichaIng, 2017-12-12:
Great work so far, like the idea of setting options by whiptail, first steps into end user direction 😄.

Just consider: https://github.com/Fourdee/DietPi/pull/1266

If you generally like the idea, I will recreate the PR for the automated script. Other small adjustments based on https://github.com/Fourdee/DietPi/issues/1219#issuecomment-347734540 and following as well.

Fourdee, 2017-12-13:
Needs more work, but new system will compare our required packages, to currently installed and their deps. All other installed packages will be removed.
```
root@DietPi:~# /DietPi/dietpi/PREP_SYSTEM_FOR_DIETPI.sh

 [Ok] Generating list of minimal packages for installation

 [Ok] Obtaining list of currently installed packages

 [Info] (0): Passed

 [Ok] Generating a list of deps, required for the DietPi packages
This may take some time, please wait...

Checking deps: acl
Checking deps: adduser
Adding deps: passwd
Adding deps: debconf
Checking deps: apt
Adding deps: adduser
Adding deps: gpgv
Adding deps: debian-archive-keyring
Adding deps: init-system-helpers
Checking deps: apt-transport-https
Checking deps: apt-utils
Adding deps: apt
Checking deps: base-files
Checking deps: base-passwd
Checking deps: bash
Adding deps: dash
Adding deps: base-files
Adding deps: debianutils
Checking deps: bash-completion
Adding deps: dpkg
Adding deps: bash
Checking deps: bc
Checking deps: bsdmainutils
Adding deps: bsdutils
Checking deps: bsdutils
Checking deps: bzip2
Checking deps: ca-certificates
Adding deps: openssl
Checking deps: console-common
Adding deps: console-data
Adding deps: kbd
Adding deps: lsb-base
Checking deps: console-data
Checking deps: console-setup
Adding deps: console-setup-linux
Adding deps: console-setup-freebsd
Adding deps: xkb-data
Adding deps: keyboard-configuration
Checking deps: console-setup-linux
Adding deps: initscripts
Checking deps: coreutils
Checking deps: cpio
Checking deps: crda
Adding deps: wireless-regdb
Adding deps: iw
Checking deps: cron
Checking deps: curl
Adding deps: zlib1g
Checking deps: dash
Checking deps: dbus
Checking deps: dc
Checking deps: debconf
Adding deps: perl-base
Checking deps: debian-archive-keyring
Checking deps: debianutils
Adding deps: sensible-utils
Checking deps: diffutils
[... 92 lines cut ...]
Adding deps: systemd
Checking deps: sysv-rc
Adding deps: insserv
Adding deps: startpar
Checking deps: sysvinit-utils
Checking deps: tar
Checking deps: tzdata
Checking deps: udev
Checking deps: unzip
Checking deps: usbutils
Checking deps: util-linux
Checking deps: wget
Checking deps: whiptail
Checking deps: wireless-regdb
Checking deps: wireless-tools
Checking deps: wpasupplicant
Checking deps: wput
Checking deps: zip

 [Ok] Generating a list of packages, not required by DietPi, to be removed from system.
This may take some time, please wait...

 [Info] The following packages will be removed
adwaita-icon-theme alsa-utils amd64-microcode bind9-host binutils build-essential checkinstall cifs-utils cmake cmake-data cpp cpp-6 cryptsetup cryptsetup-bin dh-python dhcpcd5 dialog discover discover-data distro-info-data dnsmasq dnsmasq-base dnsutils dropbear dropbear-bin dropbear-initramfs dropbear-run efibootmgr fakeroot file fontconfig fontconfig-config fonts-dejavu-core g++ g++-6 galculator gcc gettext-base git git-man glib-networking:amd64 glib-networking-common glib-networking-services gparted gpicview grub2-common gsettings-desktop-schemas gtk-update-icon-cache hicolor-icon-theme installation-report intel-microcode iptables irqbalance iucode-tool laptop-detect leafpad light-locker lightdm lightdm-gtk-greeter lighttpd linux-libc-dev:amd64 lsb-release lsof lxappearance lxappearance-obconf lxde lxde-common lxde-core lxde-icon-theme lxhotkey-core lxhotkey-gtk lxinput lxlock lxmenu-data lxpanel lxpanel-data lxpolkit lxrandr lxsession lxsession-data lxsession-edit lxsession-logout lxterminal make manpages manpages-dev mime-support mysql-common netcat netcat-traditional openbox openbox-lxde-session os-prober patch pciutils pcmanfm perl perl-modules-5.24 php-apcu php-apcu-bc php-cgi php-common php-curl php-fpm php-gd php-mbstring php-mcrypt php-xml php-zip php7.0-cgi php7.0-cli php7.0-common php7.0-curl php7.0-fpm php7.0-gd php7.0-json php7.0-mbstring php7.0-mcrypt php7.0-opcache php7.0-readline php7.0-xml php7.0-zip pinentry-curses policykit-1 python python-minimal python-talloc python2.7 python2.7-minimal python3 python3-minimal python3.5 python3.5-minimal samba-common samba-libs:amd64 sgml-base shared-mime-info smbclient spawn-fcgi sysbench tcl tcl8.6 tigervnc-common tigervnc-standalone-server tk tk8.6 ucf upower vnc4server wamerican x11-common x11-utils x11-xkb-utils x11-xserver-utils x11vnc x11vnc-data xarchiver xauth xbitmaps xcompmgr xdg-user-dirs xfonts-base xfonts-encodings xfonts-utils xinit xml-core xserver-common xserver-xorg xserver-xorg-core xserver-xorg-input-all xserver-xorg-input-libinput xserver-xorg-video-all xserver-xorg-video-amdgpu xserver-xorg-video-ati xserver-xorg-video-fbdev xserver-xorg-video-nouveau xserver-xorg-video-radeon xserver-xorg-video-vesa xserver-xorg-video-vmware xterm xz-utils zerofree
```

Fourdee, 2017-12-14:
Buster notes:

RPI:
```
W: The repository 'https://archive.raspberrypi.org/debian buster Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/main/binary-armhf/Packages  404  Not Found
```

C2:
```
W: The repository 'http://ftp.debian.org/debian buster-backports Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ftp.debian.org/debian/dists/buster-backports/main/binary-arm64/Packages  404  Not Found [IP: 130.89.148.12 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Fourdee, 2017-12-14:
Note to self:
- Check over:
https://github.com/Fourdee/DietPi/issues/1219#issuecomment-347734540

Fourdee, 2017-12-14:
Ready for early internal testing to fix bugs and improve. 

Not stable enough for public users, yet.

MichaIng, 2017-12-14:
@Fourdee 
> Buster notes:
> 
> I've disabled option for now.
- Yes, the raspberrypi.org repo was always behind: https://archive.raspberrypi.org/debian/dists/ I remember stretch distro to appear there just around it became release for raspbian, long after it became release for debian and long after I switched already to stretch with my RPi. Jessie distro worked fine there, as it mainly provides just firmware/kernel/bootloader. I will upgrade my RPi to buster after doing some tests on my buster VM to then tell more reliably if buster + raspberrypi.org stretch repos work together. We need to further implement buster for this into DietPi, but let's first get the automated script ready for public testing.
- About buster debian repo: Of course no buster-backport exists, as it is already the testing repo with up-to-date packages 😆: http://ftp.debian.org/debian/dists/ Okay sid also exists, but so far this looks just like a total experimental developer playground, with also many old (pre-)jessie packages (like php5...) inside. So for this, we can just comment out buster-backports and choose the main repo always inside our software scripts.
- About debian repo generally: Found this: https://deb.debian.org/
It is already default on new debian installations and seems to be kind of mirror director. Works well, even is reachable via https, as the link shows. Only on Jessie I got the following errors on apt update, if https was used. With http everything works fine:
```
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-updates_main_binary-amd64_Packages.diff_Index is not what the server reported 9376 345
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-updates_non-free_binary-amd64_Packages.diff_Index is not what the server reported 736 349
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-updates_main_i18n_Translation-en.diff_Index is not what the server reported 3688 343
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-updates_non-free_i18n_Translation-en.diff_Index is not what the server reported 736 347
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-backports_main_binary-amd64_Packages.diff_Index is not what the server reported 27796 347
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-backports_contrib_binary-amd64_Packages.diff_Index is not what the server reported 24964 15799
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-backports_non-free_binary-amd64_Packages.diff_Index is not what the server reported 22858 351
W: Size of file /var/lib/apt/lists/partial/deb.debian.org_debian_dists_jessie-backports_main_i18n_Translation-en.diff_Index is not what the server reported 27796 345
```
- Found just a few entries about this in g***le, seems to be known as error with apt+https on Jessie: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=802617)
- I suggest we use this also as default source, with https for stretch (and buster)? As can be seen, even security updates are inside: https://deb.debian.org/debian-security/dists/

Fourdee, 2017-12-15:
@MichaIng 

Legend 👍 

In regards to mirror:

```
root@DietPi:~# cat /etc/apt/sources.list
deb http://deb.debian.org/debian stretch main contrib non-free
deb http://deb.debian.org/security stretch main contrib non-free
deb http://ftp.debian.org/debian/ stretch-backports main contrib non-free
```

1st run
```
Ign:28 http://deb.debian.org/security stretch/non-free Translation-en
Err:14 http://deb.debian.org/security stretch/main amd64 Packages
  404  Not Found [IP: 151.101.0.204 80]
Ign:15 http://deb.debian.org/security stretch/main all Packages
```

2nd run
```
Ign:20 http://deb.debian.org/security stretch/non-free Translation-en_GB
Err:6 http://deb.debian.org/security stretch/main i386 Packages
  404  Not Found [IP: 151.101.0.204 80]
```

This is why I dislike using mirror services. They are unreliable.

Fourdee, 2017-12-15:
Ref: https://github.com/Fourdee/DietPi/pull/1280#issuecomment-351887322

Fourdee, 2017-12-15:
Moved:

@Fourdee 
About the dependency list:
- I just wrote a little script that shows installed dependants of all installed packages. With this we can further reduce our dependency list to the final packages that are really directly used by dietpi / usual system operations. Here your dependency list where I added `🈯️ > dependant`, if one within our anyway installed packages exist:
		'acl' 🈺 > systemd on Jessie, but not on Stretch+Buster, so we really need this, `getfacl` or `setfacl`
		'adduser' 🈯️ > systemd+apt+cron+udev+several other packages, depending on distro
		'apt' 🈯️ > apt-utils+apt-transport-https
		'apt-transport-https'
		'apt-utils'
		'base-files' 🈯️ > bash
		'base-passwd' 🈯️ essential package will be never autoremoved
		'bash' 🈯️ > bash-completion + essential
		'bash-completion'
		'bc'
		'bsdmainutils' 🈺 Do we need this? Was not in the list before!
		'bsdutils' 🈯️ essential
		'bzip2'
		'ca-certificates'
		'console-common' 🈺 `console-setup` supersedes or depends on other console related packages
		'console-data' 🈺 `console-setup` supersedes or depends on other console related packages
		'console-setup'
		'console-setup-linux' 🈺 `console-setup` supersedes or depends on other console related packages
		'coreutils' 🈯️ essential
		'cpio' 🈯️ (> initramfs-tools-core) > initramfs-tools
		'crda'
		'cron'
		'curl'
		'dash' 🈯️ > bash > bash-completion
		'dbus' 
		'debconf' 🈯️ > debconf-utils
		'debian-archive-keyring' 🈯️ > apt and wrong on Raspbian, where raspbian-archive-keyring takes place!
		'debianutils' 🈯️ > cron+bash+dash+some others, depending on distro
		'diffutils' 🈯️ essential
		'dmidecode' 🈺 Do we need this? Was not in the list before!
		'dmsetup' 🈯️ > > > parted
		'dosfstools' 🈺 Could be left out on VMs, as we do not have fat boot partition there.
		'dphys-swapfile'
		'dpkg' 🈯️ > cron+bash-completion+others+anyway essential
		'e2fslibs' 🈯️ > e2fsprogs
		'e2fsprogs' 🈯️ essential
		'ethtool'
		'fake-hwclock'
		'fbset'
		'findutils' 🈯️ essential
		'firmware-atheros' 🈺 was x86 only before?
		'firmware-brcm80211' 🈺 was x86 only before?
		'firmware-ralink' 🈺 was x86 only before?
		'firmware-realtek' 🈺 was x86 only before?
		'firmware-misc-nonfree' 🈺 was x86 only before?
		'fuse' 🈺 Didn't see this before, do we need it? https://packages.debian.org/de/stretch/fuse
		'gnupg' 🈺 We used this before, but `gpgv` seems to be a lightweight version of it.
		'gpgv' 🈺 Superseded by `gnupg`, but more lightweight, maybe we need to switch?
		'grep' 🈯️ essential
		'groff-base' 🈺 Do we need this? Was not in the list before!
		'gzip' 🈯️ essential
		'hdparm'
		'hfsplus'
		'hostname' 🈯️ essential
		'htop'
		'ifupdown' 🈯️ > resolvconf
		'init' 🈯️ essential
		'init-system-helpers' 🈯️ > init and essential
		'initramfs-tools'
		'initscripts' 🈺 Was dependency of systemd on Jessie, but since Stretch not anymore. I guess systemd now supersedes that package.
		'insserv' 🈺 same than initscripts, don't have any of them on my systems (besides Jessie VM) and absolutely no issues.
		'iproute2' 🈯️ > ifupdown > resolvconf
		'iputils-ping' 🈺 Helpful indeed sometimes, but not necessary? Also superseded by curl, wget etc?
		'isc-dhcp-client'
		'isc-dhcp-common' 🈯️ This is just man pages!
		'iw'
		'kbd' 🈺 `console-setup` supersedes or depends on other console related packages
		'keyboard-configuration' 🈺 `console-setup` supersedes or depends on other console related packages
		'klibc-utils' 🈯️ initramfs-tools-core > initramfs-tools
		'kmod' 🈯️ (> initramfs-tools-core) > initramfs-tools
		'less' 🈺 Helpful indeed sometimes, but not necessary?
		'locales'
		'login' 🈯️ essential
		'lsb-base' 🈯️ > cron+udev+swapfile+resolvconf+others
		'mawk' 🈯️ > dash > bash
		'mount' 🈯️ > systemd
		'multiarch-support' 🈯️ > hundrets of libs, also e2fslibs > e2fsprogs, which is essential
		'nano' not necessary, but yeah I use this several times a day, everybody needs a simple editor!
		'ncurses-base' 🈯️ essential
		'ncurses-bin' 🈯️ essential
		'net-tools'
		'ntfs-3g'
		'ntp'
		'p7zip-full'
		'parted'
		'passwd' 🈯️ > adduser
		'perl-base' 🈯️ > init-system-helpers+debconf > debconf-utils
		'procps' 🈯️ > udev+systemd
		'psmisc'
		'readline-common' 🈯️ > parted+gnu(pg)
		'resolvconf'
		'rfkill' 			#Used by some onboard WiFi chipsets
		'rsync' 			🈯️ Not necessary here, as it will be installed on demand by dietpi-backup/sync
		'sed' 🈯️ essential
		'sensible-utils' 🈺 Dependency of debianutils until buster. Just wondered that it is not autoremoved on my buster VM (marked as auto), but no dependency anymore and not essential. Just purged it on my Buster VM without any warnings or additional removed packages. Let's see what happens 😄.
		'startpar' 🈺 Was dependency of sysv-rc > systemd on Jessie, but since Stretch, systemd does not depend on sysv-rc anymore. Seems to solve the tasks differently now (same as insserv)?
		'sudo'
		'systemd' 🈯️ > systemd-sysv > init
		'systemd-sysv' 🈯️ > init
		'sysvinit-utils' 🈯️ essential
		'tar' 🈯️ essential
		'tzdata'
		'udev' 🈯️ > initramfs-tools-core > initramfs-tools
		'unzip'
		'usbutils'
		'util-linux' 🈯️ > udev+systemd+others+essential
		'wget'
		'whiptail'
		'wireless-regdb' 🈯️ > crda
		'wireless-tools'
		'wpasupplicant'
		'wput'
		'zip'

By removing the green ones from the list, there should be no additional packages removed. By removing also some of the orange ones, you get the list I used in my PR: https://github.com/Fourdee/DietPi/pull/1266/files#diff-e9e0c6c64b4937739e13ffac58f4a888R74

MichaIng, 2017-12-15:
@Fourdee 
> Err:14 http://deb.debian.org/security stretch/main amd64 Packages
>   404  Not Found [IP: 151.101.0.204 80]

Hehe this is just the wrong URL, use: `deb https://deb.debian.org/debian-security/ <distro>/updates main contrib non-free`

Fourdee, 2017-12-15:
@MichaIng 

Epic list https://github.com/Fourdee/DietPi/issues/1285#issuecomment-351888843, many thanks for that 👍 

---

**In regards to relying on deps to pull in packages we manually listed (eg: ```sed```)**
In general, there is no harm is us manually listing additional packages that DietPi requires to function, instead of relying on other packages to pull it in. 

Benefits of listing (example ```sed```:
- We can see what packages DietPi fully requires/uses to function
- By not relying on deps, if Debian were to change ```sed``` ```essential``` deps overnight, no risk of breaking script.

---

**Firmware:**

Covers us for all WiFi chipsets etc across all devices. We really should have it in for all images.

```firmware-misc-nonfree``` can be x86_64 only, mostly intel/nvidia stuff

Fourdee, 2017-12-15:
with changes, quick test on DietPi RPi with only ```openssh-server``` installed:
```
The following packages will be REMOVED:
  bluetooth* bluez* bluez-firmware* bsdmainutils* busybox* console-common*
  console-data* cryptsetup* cryptsetup-bin* debconf-utils* dmidecode* ed*
  file* firmware-libertas* firmware-misc-nonfree* gdb* gettext-base*
  groff-base* hardlink* info* iptables* isc-dhcp-common* less*
  libbabeltrace-ctf1* libbabeltrace1* libboost-filesystem1.62.0*
  libboost-system1.62.0* libc6-dbg* libcwidget3v5* libdrm2* libdw1* libestr0*
  libfastjson4* libgdbm3* libglib2.0-0* libglib2.0-data* libident* libip6tc0*
  libiptc0* liblogging-stdlog0* liblognorm5* libluajit-5.1-common*
  libmagic-mgc* libmagic1* libmpdec2* libnetfilter-conntrack3* libnfnetlink0*
  libnih-dbus1* libnih1* libpam-chksshpwd* libpam-systemd* libperl5.24*
  libpipeline1* libpython3.5* libpython3.5-minimal* libpython3.5-stdlib*
  libsasl2-modules* libsigc++-2.0-0v5* libsysfs2* libudev0* libusb-0.1-4*
  libwrap0* libxapian30* libxtables12* luajit* mime-support* openssh-client*
  openssh-server* openssh-sftp-server* paxctld* perl* perl-modules-5.24*
  pi-bluetooth* rename* ucf*
0 upgraded, 0 newly installed, 75 to remove and 4 not upgraded.
After this operation, 138 MB disk space will be freed.

```

@MichaIng 

Legend 👍 Much lighter install =))

Fourdee, 2017-12-15:
dietpi-drive_manager not setting fstab correctly, will sort it.

Fourdee, 2017-12-15:
NTS:
- RPi boot loop. install stage is stuck on -3

MichaIng, 2017-12-15:
Offer some adjustments and further ideas here: https://github.com/Fourdee/DietPi/pull/1307

Fourdee, 2017-12-16:
Setting this back to High Priority. This will vastly benefit image creation, and, offer users another installation method.

Fourdee, 2017-12-16:
Jessie > Stretch:

need to check ordering of autoremove
```
 [Ok] Step 8: Finalise system for first run of DietPi:

 [Info] Installing Dropbear by default

Reading package lists... Done
Building dependency tree
Reading state information... Done
dropbear is already the newest version (2016.74-5).
The following packages were automatically installed and are no longer required:
  acl busybox-initramfs isc-dhcp-common libapt-inst1.5 libapt-pkg4.12
  libcap2-bin libdns-export100 libgnutls-deb0-28 libhogweed2 libicu52
  libirs-export91 libisc-export95 libisccfg-export90 libnettle4 libprocps3
  libpsl0 libreadline6 libssl1.0.0 libusb-0.1-4
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Fourdee, 2017-12-16:
#### Tests:
- C2 Stretch > Jessie
- RPi Stretch > Stretch

Fourdee, 2017-12-16:
@MichaIng i'll re-enable the RPi packages (eg: kernel etc) due to below:

DietPi puts them on hold, after our kernel is installed.
```
The following packages have unmet dependencies:
 libraspberrypi-bin : Depends: libraspberrypi0 (= 1.20171029-1) but 1.20170811-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2017-12-16:
@Fourdee 
Ah okay, makes sense. So we initially install kernel packages, also to at least temporarily resolve apt update messages about them, and then install our own kernel version on top of it? Otherwise we could also directly do the second and just ignore RPi packages completely. I was even thinking, if it would not be possible to completely remove the RPi kernel packages and "just" install it by ourself. This would make accidental overwriting by package and irritating available update messages impossible 😄.
______________________
Just copied from my PR for consideration:

ToDo:
- Preserve current network interfaces, otherwise internet and/or SSH access can break after reboot.
- Make WiFi related packages optional, like Bluetooth?
- Make wrapper and file system packages optional, if not explicitly needed by certain DietPi script/software?
- Choose right microcode package based on architecture?
- Recognize and preserve user SSH server choice? That way we also preserve potential key authentication, port etc. settings and other stuff, user configured on his SSH client. Otherwise SSH connection potentially fails after reboot.
- Cleanup openssh + dropbear handling in finalise script part.
- Check if we need `libpam-systemd` unit, which looks necessary for logind functionality. Otherwise logind could be just disabled: https://packages.debian.org/de/stretch/libpam-systemd
- Remove (all?) firmware packages for VM images. No real processor seems to be emulated and no other usual hardware, where firmware is needed for. Works well, tested on all distro versions on VirtualBox.

Fourdee, 2017-12-18:
Quick test after all my recent changes/commits:
- 🈯️ RPi 3 DietPi-Jessie > DietPi-Stretch

I need to setup some VM's with stock Debian and run further tests.

Fourdee, 2017-12-18:
Note to self,
Clean these up:
```
 [Info] Creating DietPi core environment

/boot/dietpi/func/dietpi-notify: line 19: /DietPi/dietpi/func/dietpi-globals: No such file or directory

 [Info] Generating DietPi /etc/fstab

/boot/dietpi/dietpi-drive_manager: line 249: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 370: /DietPi/dietpi/func/dietpi-set_userdata: No such file or directory
```

MichaIng, 2017-12-18:
Test of current script on fresh Debian Jessie, installed via mini.iso with efi partition.
- Debian disc installation correctly chose grub-efi-amd64 package by itself.
- 🈴 DietPi script chose however grub2 => **Problem found, fix on the way**
- 🈴 Several times during step 7: `/boot/dietpi/func/dietpi-notify: line 19: /DietPi/dietpi/func/dietpi-globals: No such file or directory`
- 🈴 During step 7:
```
[Info] Generating DietPi /etc/fstab
/boot/dietpi/dietpi-drive_manager: line 121: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 190: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 203: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 190: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 198: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 249: /DietPi/dietpi/func/dietpi-notify: No such file or directory
/boot/dietpi/dietpi-drive_manager: line 370: /DietPi/dietpi/func/dietpi-set_userdata: No such file or directory
```
- Step 8: Minor, but we should get rid of any expected error message:
```
 [Info] Resetting boot.ini, config.txt, cmdline.txt etc
sed: can't read /boot/cmdline.txt: No such file or directory
```
```
 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: bluetooth (disable)
 Please wait...

 [Info] Bluetooth service failed to stop. Is it installed?
 [Failed] bluetooth disable | An issue has occured
```
```
 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: preferipversion (auto)
 Please wait...

 [Ok] preferipversion auto | Completed

 [Info] Configuring kernels

rm: cannot remove '/etc/dietpi/.': Is a directory
rm: cannot remove '/etc/dietpi/..': Is a directory
```
```
 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: soundcard (none)
 Please wait...

 [Info] Installing alsa-utils. Please wait...
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package alsa-utils
 [Failed] Unable to install alsa-utils.
```
- 🈴 After reboot: grub rescue, no such partition/invalid filesystem => Will try again after grub installation fix

MichaIng, 2017-12-19:
Some fixes: https://github.com/Fourdee/DietPi/pull/1318

Reboot issue stays on EFI system:
- Keep old / + /boot/efi fstab entries => **no solution**
- Run script until end of step 5 (APT removal and installation) => `works`

Fourdee, 2017-12-19:
@MichaIng

Legend, great work 👍  

> Why do we enable i386 architecture for APT on amd64 system?

To ensure software which requires i386 pre-reqs (eg: steam I believe), is functional. Theres no harm in enabling it, i386 software should only be installed via apt with ```apt-get install xxxx:i386```.

Fourdee, 2017-12-19:
Testing VM with Debian 9 minimal:
```
./PREP_SYSTEM_FOR_DIETPI.sh: line 564: [: missing `]'
grep: ]: No such file or directory
./PREP_SYSTEM_FOR_DIETPI.sh: line 568: [: missing `]'
grep: ]: No such file or directory
```

MichaIng, 2017-12-19:
@Fourdee 
Thanks for adding+fixing. Interesting, on my Jessie VM `[ ... ]` works great. Seems to be an inconsistency in using brackets within bash, and I am still a bid confused when to use what. Reading: https://stackoverflow.com/questions/2188199/how-to-use-double-or-single-brackets-parentheses-curly-braces 😆.

Fourdee, 2017-12-19:
@MichaIng 

> Thanks for adding+fixing. Interesting, on my Jessie VM [ ... ] works great. Seems to be an inconsistency in using brackets within bash, and I am still a bid confused when to use what. Reading: 

Unsure, i'll need to review it.

I've added fixes for a few of the issue, made notes of outstanding. Will try and pick this back up tomorrow 😄

MichaIng, 2017-12-19:
@Fourdee
> Unsure, i'll need to review it
Not needed, this was mostly for myself. Your fix works!

Fourdee, 2017-12-19:
Note to self:
- ```dietpi-global``` is now loadable in PREP_SYSTEM. I'll verify tomorrow and send commit. Means we dont have to define ```dietpi-notify``` and ```AGI``` twice.

Fourdee, 2017-12-20:
@MichaIng 

Sending a large commit with vast changes to globals + this script. I'll be doing testing and resolving issues afterwards, could be very unstable, may want to avoid any commits/PR's at this stage.

Fourdee, 2017-12-20:
Re-Tests:
- 🈯️ Odroid C2, running DietPi Stretch > Stretch
- VM Debian 9

Fourdee, 2017-12-20:
@MichaIng 

Ok, global naming/use are now finalized (took me a while to find the "right" naming): https://github.com/Fourdee/DietPi/issues/1311#issuecomment-353039600

PREP_SYSTEM still needs work, however, the major script wide changes are now completed. Please feel free to continue with PR's as needed.

MichaIng, 2017-12-21:
@Fourdee 
Just used the script on DietPi VM-Stretch, upgrading to Buster:
- Jep, connection get lost in between because of interface reset.
- On first run dietpi-software connection check, it took significantly longer than 20s. Reason after investigation is, that wget by default tries to connect via IPv6, even that IPv6 is disabled on DietPi. For apt update, depending on test domain/ip, it is the same. We really need to consider, setting forced IPv4 for apt and wget by default. On first run we could however test IPv6 connectivity and remove forcing settings for apt and wget again. On enabling/disabling IPv6 within dietpi-config, this settings should also be set. I am not sure, if 'forcing' IPv6 is a good idea and will work on every access server, but forcing IPv4 is needed, as apt and wget do not check for enabled kernel module or other things.
- On applying network settings I get: `Failed to disable unit: Unit file wifi_disable_powersave.service does not exist`
- If ownCloud/Nextcloud was installed on DietPi system, their software directories stay in place on script run, but mysql/mariadb will be purged together with database files. On reinstallation of ownCloud/Nextcloud, installer will find a fully configured instance with `installed` flag in it's config file, thus will skip most steps, including mysql/mariadb database+user creation => Nextcloud/ownCloud will stay without database access. I was thinking of some smart installer checks, detecting if database settings within nc/oc config files match actual mysql/mariadb content, and in case remove installed flag from config file to allow occ terminal installation.
On the other hand, other dietpi software might be affected as well, if just packages get removed, but no manually downloaded binaries and config files. We could also, if DietPi system is detected, uninstall all dietpi software first. This actually fit's better to the idea of creating a totally fresh instance.

Fourdee, 2017-12-21:
@MichaIng 

> Jep, connection get lost in between because of interface reset.

Ok, i'll take a look, see if we can find a cure.

> On applying network settings I get: Failed to disable unit: Unit file wifi_disable_powersave.service does not exist

During PREP_SYSTEM run? Any idea what stage (eg: notify text before it)

> We really need to consider, setting forced IPv4 for apt and wget by default.

Agree, i'll see if we can apply them to the globals and script.

> On the other hand, other dietpi software might be affected as well, if just packages get removed, but no manually downloaded binaries and config files. We could also, if DietPi system is detected, [...]

The initial idea (in my mind) was for PREP_SYSTEM to achieve the following:
- Allow users with non-DietPi systems, to install it, ending up with a clean fresh installation, ready for 1st time use.
- Allow me (and/or others) to quickly update our existing DietPi ```.img``` files, again, clean fresh installation for first time use.

I personally think if we go down the path of "system upgrades" (eg: converting existing system in-use, and its installed software, across to DietPi), we will open up a huge can of worms.
I believe we should keep PREP_SYSTEM designed solely to install a fresh DietPi installation on the device, for 1st time use.

Fourdee, 2017-12-21:
Due to FX6300 on its way out, having to run VM tests on a AMD Athlon 5150, not bad, not great lol.
https://www.cpubenchmark.net/cpu.php?cpu=AMD+Athlon+5150+APU&id=2208

Fourdee, 2017-12-21:
Notes:
```
firmware-linux-nonfree can not be marked as it is not installed.
grub2 can not be marked as it is not installed.

The following packages will be REMOVED:
grub-common* grub-pc* grub-pc-bin* grub2-common*
```

Fourdee, 2017-12-21:
### Tests:
- 🈯️ VM Debian Stretch > Buster
- VM DietPi Debian Buster > Jessie
reupdate globals?
```
W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: Unable to correct problems, you have held broken packages.
 [Failed] G_AGI: An issue has occured

 [Failed] (100): Script aborted
```

MichaIng, 2017-12-21:
@Fourdee 
`🈴 2:`
Uff downgrading of cause brings several other issues and thanks god this is a scenario, no one will really need.
- Creating official images, we should always start with the same debian/raspbian image, as there will be always possible issues, obsolete files, incompatible configs etc. on distro version changes. Btw. I found the mini images to be the smallest for Debian, but of course no local repository is contained: http://ftp.nl.debian.org/debian/dists/stretch/main/installer-amd64/current/images/netboot/
- No one would want to downgrade his running production system. Most, I guess, want to stay, but indeed people are always looking for a way to upgrade, without need to start from scratch, e.g. http://dietpi.com/phpbb/viewtopic.php?f=9&t=2594 But this will always be a bid risky and outcome extremely depends on individual system, thus I agree with you, that we should not support this. But enabling a more reliable way, by first purging down a system as much as possible, would be no harm, from my point of view. We just need to make clear, that this is not the intention of the script and we will not provide any support for issues, that can easily occur.
______________________________
About the APT issues:
- This is expected, as we do not purge all packages, and the ones that stay, will not be downgraded according to priority rules. This we leave a mixture of packages from both distro versions where their versions may not fulfil each others dependencies. I guess if this should work reliably, we would need to force downgrading of all existing packages by temporary raising target repo priority to >1000: https://linux.die.net/man/5/apt_preferences
- Generally I think it would also make a sense, if we temporary overwrite existing package configs by current maintainers versions during APT step. After that we do our adjustments to fresh, repo compatible  config files and then force keeping adjusted configs again.
_________________________________
So generally about how to going on:
- Do we want to keep the option to choose distro version? => Then I would consider the mentioned steps to at least get this working on our current tests, but as mentioned would give a clear notification that there will be no support from our side (as the issues might become uncountable and hard to investigate).
- Do we want to keep the option (and handling) to run the script on already DietPi enabled system? => Then I would still consider to dietpi-software uninstall everything, before further cleaning up APT packages. I anyway want to implement some nice backup handling and restoring to ownCloud/Nextcloud, that data and database can survive uninstall/install procedures, if wanted by user.
____________________________

Just reviewed your last commits, looks very nice 👍.  Will run test runs again 😃.

MichaIng, 2017-12-22:
Optional WiFi and conditional microcode package installation: https://github.com/Fourdee/DietPi/pull/1331

The first needs on demand installation (and uninstallation?) on WiFi activation/deactivation within dietpi-config and some other scripts need to check for installed packages before throwing errors.

Fourdee, 2017-12-22:
@MichaIng 

> We just need to make clear, that this is not the intention of the script and we will not provide any support for issues, that can easily occur.

Agree, i'll make a note of no downgrade support 👍 
I'll see if we can hide distro selection, if lower than current.

---

```firmware-realtek``` also contains ethernet firmware, I'll review these, add them back in if needed (https://packages.debian.org/stretch/firmware-realtek):
https://github.com/Fourdee/DietPi/pull/1331/files#diff-e9e0c6c64b4937739e13ffac58f4a888L473

k-plan, 2017-12-31:
I want to play around as well: 😄 

[image]

#### Required Information:
- DietPi Version | v      _# #  _(note: it is v160 testing Branch updated)_ # #_
- SBC Device     | oDroid XU3/4 (armv7l) (index=11)
- Distro         | stretch (index=4)
- Command        | G_AGA:
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-PREP_SYSTEM_FOR_DIETPI

#### Expected behaviour:

- Build a new DietPi image for my Odroid HC1 from `(NR_4.x)_DietPi_v159_OdroidXU4-armv7-(Stretch).img`

#### Actual behaviour:

- ` [Error] Unable to continue, the program will now terminate.` 😢 

#### Steps to reproduce:

- set: `GIT_BRANCH=testing`
- run: `:~# ./PREP_SYSTEM_FOR_DIETPI.sh`

#### Additional logs:
```
Log file contents:
Removing libapt-inst1.5:armhf (1.0.9.8.4) ...
Removing libapt-pkg4.12:armhf (1.0.9.8.4) ...
Removing libcap2-bin (1:2.25-1) ...
Removing libirs-export91 (1:9.9.5.dfsg-9+deb8u14) ...
Removing libdns-export100 (1:9.9.5.dfsg-9+deb8u14) ...
Removing libestr0 (0.1.10-2) ...
Removing perl (5.24.1-3+deb9u2) ...
Removing libperl5.24:armhf (5.24.1-3+deb9u2) ...
Removing libgdbm3:armhf (1.8.3-14) ...
Removing libgnutls-deb0-28:armhf (3.3.8-6+deb8u7) ...
Removing libgpm2:armhf (1.20.4-6.2+b1) ...
Removing libhogweed2:armhf (2.7.1-5+deb8u2) ...
Removing libpsl0:armhf (0.5.1-1) ...
Removing libicu52:armhf (52.1-8+deb8u6) ...
Removing libisccfg-export90 (1:9.9.5.dfsg-9+deb8u14) ...
Removing libisc-export95 (1:9.9.5.dfsg-9+deb8u14) ...
Removing liblogging-stdlog0:armhf (1.0.5-2+b2) ...
Removing nfacct (1.0.1-1.1+b1) ...
Removing libnetfilter-acct1:armhf (1.0.2-1.1) ...
Removing libnettle4:armhf (2.7.1-5+deb8u2) ...
Removing libnfnetlink0:armhf (1.0.1-3) ...
Removing libpipeline1:armhf (1.4.1-2) ...
Removing libprocps3:armhf (2:3.3.9-9) ...
Removing libreadline6:armhf (6.3-8+b3) ...
Removing libssl1.0.0:armhf (1.0.1t-1+deb8u7) ...
Removing libusb-0.1-4:armhf (2:0.1.12-30) ...
Removing linux-image-4.9-armhf-odroid-xu3 (4.9.61-1) ...
Removing linux-image-4.9.61+ (4.9.61-20171118-XU3) ...
dpkg: error processing package linux-image-4.9.61+ (--remove):
 subprocess installed pre-removal script returned error exit status 1
Removing perl-modules-5.24 (5.24.1-3+deb9u2) ...
Removing xz-utils (5.2.2-1.2+b1) ...
dpkg: u-boot-tools: dependency problems, but removing anyway as you requested:
 linux-image-4.9.61+ depends on uboot-mkimage | u-boot-tools; however:
  Package uboot-mkimage is not installed.
  Package u-boot-tools is to be removed.

Removing u-boot-tools (2016.11+dfsg1-4) ...
dpkg: uboot: dependency problems, but removing anyway as you requested:
 linux-image-4.9.61+ depends on uboot; however:
  Package uboot is to be removed.
  Package uboot which provides uboot is to be removed.
 linux-image-4.9.61+ depends on uboot; however:
  Package uboot is to be removed.
  Package uboot which provides uboot is to be removed.

Removing uboot (20170824-9+deb8u1) ...
Errors were encountered while processing:
 linux-image-4.9.61+
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

k-plan, 2017-12-31:
Next run testing fun: 😃 

[image]

#### Required Information:
- DietPi Version | v
- SBC Device     | NULL (index=11)
- Distro         | stretch (index=4)
- Command        | G_AGI: apt-transport-https apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl dbus debconf dosfstools dphys-swapfile dropbear ethtool fake-hwclock fbset firmware-realtek gnupg hdparm hfsplus htop initramfs-tools iputils-ping isc-dhcp-client locales nano net-tools ntfs-3g ntp p7zip-full parted psmisc resolvconf sudo tzdata unzip usbutils wget whiptail wput zip linux-image-armhf-odroid-xu3 crda firmware-atheros firmware-brcm80211 firmware-ralink iw rfkill wireless-tools wpasupplicant
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-PREP_SYSTEM_FOR_DIETPI

#### Expected behaviour:

[Info] Non-DietPi system

- Build a new DietPi image for my Odroid HC1 from https://oph.mdrjr.net/meveric/images/Jessie/Debian-Jessie-1.1.4-20171121-XU3+XU4.img.xz

#### Actual behaviour:

- after 1h of running and updating files .... 🕛  🕐  😢  

`[Error] Unable to continue, the program will now terminate.`

- after this, system is f... up, `apt-get update` won't run any more.

#### Steps to reproduce:

- download the image, write it to sd card and after booting up run: 
`:~# ./PREP_SYSTEM_FOR_DIETPI.sh`

#### Additional logs:
```
Log file contents:
E: Method https has died unexpectedly!
E: Sub-process https received a segmentation fault.
E: Method /usr/lib/apt/methods/https did not start correctly
```

-----------------

### General Notes:

- run `apt-get update` as first step because `a_MIN_APT_PREREQS` will fail elsewhere on a fresh image

- first run and set `dpkg-reconfigure locales` to `en_GB.UTF-8` to avoid hundert message like this:
```
warning: Setting locale failed.
 warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_GB.UTF-8"
    are supported and installed on your system.
warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
```

- set: `passwd` to a known password like "`dietpi`" for user "`root`"

------------------

NB: 
- how will we handle different disk layouts? 
Not two partitions, which first is "`boot`" (FAT) / second is "`/`" (ext4), eg one big BTRFS ?

- yes, `finalize` was a PITA and very time-consuming. 
But `PREP_SYSTEM_FOR_DIETPI.sh`? For every image update? Puuuh ... I don't know.

-------------

Just like that, the fun's over for today.  😃

k-plan, 2017-12-31:
-----------------

### General Notes:

- run `apt-get update` as first step because `a_MIN_APT_PREREQS` will fail elsewhere on a fresh image

- first run and set `dpkg-reconfigure locales` to `en_GB.UTF-8` to avoid hundert message like this:
```
warning: Setting locale failed.
 warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_GB.UTF-8"
    are supported and installed on your system.
warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
```

- set: `passwd` to a known password like "`dietpi`" for user "`root`"

------------------

NB: 
- how will we handle different disk layouts? 
Not two partitions, which first is "`boot`" (FAT) / second is "`/`" (ext4), eg one big BTRFS ?

- yes, `finalize` was a PITA and very time-consuming. 
But `PREP_SYSTEM_FOR_DIETPI.sh`? For every image update? Puuuh ... I don't know.

-------------

Just like that, the fun's over for today.  😃

k-plan, 2017-12-31:
> ### _MichaIng commented:_

----------------------

@k-plan 
_€: This is about your first try ;). How did you solve the linux-image apt error for 2nd try?_

Ah just found this one: http://www.hardkernel.com/main/products/prdt_info.php?g_code=G150229074080
> Any OS for XU4 is fully compatible with the HC1.

And as I see the packages, the image seems to be based on Meverics?

- For this device we choose the default kernel meta package `linux-image-armhf-odroid-xu3`: https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh#L553
- But it depends on ancient `linux-image-3.10.106+`: https://oph.mdrjr.net/meveric/dists/all/xu3/binary-armhf/Packages Thus the existing kernel package will removed first.
- On official Debian images, `linux-image*` packages are blocked from autoremove by `/etc/apt/apt.conf.d/01autoremove(-kernels)`, which seems not the case for Meverics image.
- Removing the only installed kernel package seems to need failing, which causes the dpkg error?
- On HC1 product page they promote the use of newest 4.14 kernel on their official linux (Ubuntu based) image, so I assume `linux-image-4.14-armhf-odroid-xu4` to be stable enough also for us? Otherwise at least we should use `linux-image-4.9-armhf-odroid-xu3`.
- xu3/xu4 naming mixed together there, both are meta packages depending on linux-image-4.XX.XX packages where no device naming appears any more, thus I think they are compatible vice versa and for HC1 as well.
- **To prevent any kernel removal issues during image preparation: Install the chosen kernel package before `G_AGA`, or identify and choose already installed kernel package. I would vote for first solution, as we then keep control over which kernel is used with DietPi.**

k-plan, 2017-12-31:
@MichaIng 

> And as I see the packages, the image seems to be based on Meverics?

Yes!

> Any OS for XU4 is fully compatible with the HC1

I have write hardware comparison here: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2422&p=9583&hilit=HC1#p9583

>To prevent any kernel removal issues during image preparation: Install the chosen kernel package before G_AGA, or identify and choose already installed kernel package. I would vote for first solution, as we then keep control over which kernel is used with DietPi.

How to do this? Image was on 4.9.61+. Have tested, you can upgrade to 4.14.x, but after this, you can't go back to 4.9.61+.
https://forum.odroid.com/viewtopic.php?f=96&t=17542&p=209574#p208191
>When asked if you want to STOP the removal of the active Kernel please answer with "NO".

If you will answer "YES" kernel update stops and fail. (tested)

----------

sorry @Fourdee , no english, to time-consuming

Ja, die Odroid Welt ist etwas schwieriger ....

- Die wollen kompatibel bleiben für: XU4/XU3/HC1/MC1 und nur ein Image (+ Quellen) pflegen

> We guarantee the production of ODROID XU4/XU3/HC1/MC1 to the middle of 2020, but expect to continue production long after.

Hardkernels Images (Ubuntu) sind schon zur Zeit auf 4.14.x , da gibt es auch die Kernel source für:
https://forum.odroid.com/viewtopic.php?f=146&t=28895

Meveric bastelt aus der Kernel Source ein Debian draus und pflegt das schon seit Jahren. Sein Schwerpunkt ist die Gaming Welt ( https://forum.odroid.com/viewtopic.php?f=11&t=2684 ) daher auch die gute Unterstützung für Kodi, XBMC, Grafikausgabe über HDMI.
Der ist aber auch eine "One-Man-Show" wie Dan bisher. Daher hängt er immer etwas hinterher.
Hatte aber die gesamte Mali-T628 MP6 Hardware Grafik Unterstützung als erster laufen und lange Zeit die Nase vorn damit.

Mit dem 4.x Kernel ist es zum "Bruch" gekommen, so wie ich das bis jetzt verstanden habe.
Bis 3.x nutzen alle ODROID XU4/XU3/HC1/MC1 nur `exynos5422-odroidxu3.dtb`
Seit 4.x wir es spezifischer und man muss das richtige beim booten laden je nach verwendeten Board: 
```
exynos5422-odroidxu3-lite.dtb
exynos5422-odroidxu3.dtb
exynos5422-odroidxu4-kvm.dtb
exynos5422-odroidxu4.dtb
```

Da gibt es zwar ein paar nette Tricks für, wie man das über die `boot.ini` steuern könnte, nur haben wir das im Moment (noch) nicht (_Code Schnippsel entliehen_):

```
if test "${board_name}" = "xu4"; then setenv fdtfile "exynos5422-odroidxu4.dtb"; fi
if test "${board_name}" = "xu3"; then setenv fdtfile "exynos5422-odroidxu3.dtb"; fi
if test "${board_name}" = "xu3l"; then setenv fdtfile "exynos5422-odroidxu3-lite.dtb"; fi

if ext4load mmc 0:1 0x00000000 "/boot/.next" || fatload mmc 0:1 0x00000000 ".next"  || ext4load mmc 0:1 0x00000000 ".next"; then echo "Found mainline kernel configuration"; else setenv fdtfile "exynos5422-odroidxu3.dtb"; fi
ext4load mmc 0:1 0x44000000 /boot/dtb/${fdtfile} || fatload mmc 0:1 0x44000000 dtb/${fdtfile} || ext4load mmc 0:1 0x44000000 dtb/${fdtfile}
```

Weitere Schwierigkeit, Fourdee will es "schlank" haben und will eine FAT boot Partition, in der man einfach Konfigs editieren kann.
Beides finde ich sehr gut. Aber herzlichen Glückwunsch, wenn das Gerät von SD-Karte, eMMC, USB und HDD/SSD booten kann. Das wird dann mit der `/etc/fstab` sehr schnell unschön und die Tricksereien fangen an.

```
:~# apt-get install linux-image-
linux-image-3.10.105+                 linux-image-4.14.9+                   linux-image-4.9.35+                   linux-image-4.9.58+
linux-image-3.10.106+                 linux-image-4.14-armhf-odroid-xu4     linux-image-4.9.37+                   linux-image-4.9.60+
linux-image-3.16.0-4-armmp            linux-image-4.9.0-0.bpo.2-armmp       linux-image-4.9.38+                   linux-image-4.9.61+
linux-image-3.16.0-4-armmp-lpae       linux-image-4.9.0-0.bpo.2-armmp-lpae  linux-image-4.9.43+                   linux-image-4.9-armhf-odroid-xu3
linux-image-4.14.0+                   linux-image-4.9.0-0.bpo.4-armmp       linux-image-4.9.44+                   linux-image-armhf-odroid-xu3
linux-image-4.14.3+                   linux-image-4.9.0-0.bpo.4-armmp-lpae  linux-image-4.9.47+                   linux-image-armmp
linux-image-4.14.4+                   linux-image-4.9.30+                   linux-image-4.9.50+                   linux-image-armmp-lpae
linux-image-4.14.5+                   linux-image-4.9.33+                   linux-image-4.9.51+
linux-image-4.14.7+                   linux-image-4.9.34+                   linux-image-4.9.57+
```

Meine persönlichen "Erfahrungen" mit den verschiedenen Kernel:

- 3.x hat eine schlechtere RAM Performance, warum auch immer.

- 4.9.x ist mit dem RAM der schnellste

- 4.14.x ist etwas langsamer bei meinen Tests

Schreib/Lesegeschwindigkeit (SD-Karte vs eMMC), RAM Größe und Schreib/Lesegeschwindigkeit, machen bei den Devices, neben Prozessortakt und Kerne Anzahl, den größten Unterschied aus.

Klar haben neuere Kernel meist bessere Unterstützung für "neue Features", doch persönlich brauche ich die wenigsten davon. Würde also vorerst zu 4.9.x bei XU4 tendieren.
Da wird aber von Seiten Hardkernel erst einmal nichts mehr kommen.

Ordoid C1/C1+/C2 schenke ich mir an dieser Stelle erst einmal. Das sind noch ganze andere Nummern mit anderen Schwierigkeiten.

k-plan, 2017-12-31:
@Fourdee 
@MichaIng 

Only for clarification: 
I like your work so much. You are on a good trip and I think and hope you have finally success.
Only some difficulties arising from a new way.
Keep one's tail up boys and girls. 😃 

Or is it too early for testing?

PS: ... and the output github printing  is awesome! Make putting things on here so easy! 
Love it! 👍  💯 
Hope user will use it too!

k-plan, 2017-12-31:
> ### _MichaIng commented_
----------------------
@k-plan 
Okay, about the kernel and boot.ini modification: For me that sounds, as would it be best to keep the kernel (linux-image*) package that already exists on the image and do not mess around to much with the boot.ini, at least for now? Or stays boot.ini unaffected by kernel upgrade?
We do already the same with fstab, at least the script should just copy the drives mounted on the existing machine. Of course auto file system expansion is another topic 🙈, if the machine has that much different boot options with possibly different device name schemes.

Indeed, it might be more successful/happy ending for now, to test the preparation script on already existing devices. But on the other hand as HC1 is/should work more less the same than XU3/4, the error should have also occurred there. I guess we have to expect some more issues here and there with the script, creating images/testing for all devices. But as before it was also hand craft, the only downside is, that you can't make the script starting at a certain point, after you manually fixed the issue there. But copy/paste the commands after manually fixing is still possible 😆.
Maybe we can create some entry points, e.g. for the 8 steps of the script. Means if you run `./PREP_SYSTEM_FOR_DIETPI.sh 4` the script starts at step 4.

But hmm, we are sort of in the wrong issue here @Fourdee is it possible for you to copy the last comments over to https://github.com/Fourdee/DietPi/issues/1285?

_________________________
Another topic I found by chance: https://github.com/Fourdee/DietPi/issues/1357

k-plan, 2017-12-31:
> to test the preparation script on already existing devices. But on the other hand as HC1 is/should work more less the same than XU3/4, the error should have also occurred there.

Yes. HC1 is a XU4 without HDMI but with onboard USB3-to-SATA-Bridge, like [CloudShell for XU4](http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143599699669) before.
But anyway. Whatever device shall I test?
- C2? Fourdee did this.
- RPi? Fourdee did this and you can done by yourself.
- C1+? Puuh ... next case full of worms ....
- AllWinner (Orange/Banana/Nano Pi) ? This are crappy slow for testing ...

MichaIng, 2017-12-31:
@k-plan 
Thx for copying, so I guess there is no "moving" posts possible in github.

About image creation testing: Choice is yours, maybe orientate on the machine usage statistics, but for now the plan is anyway to get all images up to date until 6.0 release.
I will try to fix the kernel un/installation issue. Still thinking how to solve this best. The easiest way is to just "apt-get install" all device specific kernel/firmware/bootloader packages. This will also mark them as "manually" installed, thus prevents them from being autoremoved. As in case new/different versions of kernel are being installed, the older ones should be able to autoremove successfully.
The only thing is that it doesn't fit anymore into `'Step 4: APT removals:'` 😆, could be moved into `'Step 3: Distro selection / APT prep:'` or we create a new step: `APT hardware package preparation` or something like that.

But to allow further testing for now, I will create PR and fix/adjust some other things, I found yesterday.

k-plan, 2017-12-31:
Quick and dirty:
- 🈴  OrangePi One, running DietPi Jessie (3.x) > Stretch

- it's running nice and smooth 👍 

`[Ok] Completed, disk can now be saved to .img for later use, or, reboot system to start first run of DietPi:`

😃 

==>> `boot loop`

-------------------

> Thx for copying, so I guess there is no "moving" posts possible in github.

Yes, not for you and me. It's sadly not a forum.

Fourdee, 2017-12-31:
@k-plan 

> OrangePi One, running DietPi Jessie (3.x) > Stretch

All ARMbian images will fail to boot currently. We've not added the armbian APT kernel packages to the list yet.
I'll do this once v6.0 is near completion and have to go through all the images. 

Installed kernel packages should be under:
```
dpkg -l | grep '^linux'
```

Fourdee, 2017-12-31:
Maybe we can scrape the current ```linux-image``` from system, prior to dep list generation, then add it automatically. I'll take a look.

k-plan, 2017-12-31:
@Fourdee 

Hmm ... so I can stop testing for now?

Or shall go on with Odroid C1+?

Or leave you coders alone for the moment to continue your nice work?

Fourdee, 2017-12-31:
@k-plan 

Its not fully finished yet, but yep, the C1 should work (and the ones I posted above).

MichaIng, 2017-12-31:
Fix is there https://github.com/Fourdee/DietPi/pull/1362
If this generally works, we can thinking of moving the stuff somewhere else or something.

k-plan, 2017-12-31:
@Fourdee 
> the C1 should work

- 🈴  Odroid C1+ running Meveric Jessie (last Kernel 3.x) > Stretch

[image]

#### Required Information:
- DietPi Version | v.
- SBC Device     | NULL (index=10)
- Distro         | stretch (index=4)
- Command        | G_AGI: apt-transport-https apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl dbus debconf dosfstools dphys-swapfile dropbear ethtool fake-hwclock fbset firmware-realtek gnupg hdparm hfsplus htop initramfs-tools iputils-ping isc-dhcp-client locales nano net-tools ntfs-3g ntp p7zip-full parted psmisc resolvconf sudo tzdata unzip usbutils wget whiptail wput zip linux-image-armhf-odroid-c1 crda firmware-atheros firmware-brcm80211 firmware-ralink iw rfkill wireless-tools wpasupplicant
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-PREP_SYSTEM_FOR_DIETPI

#### Expected behaviour:

#### Actual behaviour:

` [Error] Unable to continue, the program will now terminate.`

#### Steps to reproduce:

#### Additional logs:
```
Log file contents:
E: Method https has died unexpectedly!
E: Sub-process https received a segmentation fault.
E: Method /usr/lib/apt/methods/https did not start correctly
```

Stop for now. Must leave the house for New Year's party.

Wish you all a Happy New Year 🚀   💥  🍸

[138 more replies]

closed 2018-01-28 by Fourdee