MichaIng/DietPi #1286: DietPi | Debian Buster

MichaIng, 2017-12-10:
As I anyway created a third VirtualBox image as fresh Stretch, I just became interested in further upgrading to Debian Buster. It is never to early to watch changes there and how we can resolve them for DietPi:

1. NTP now is a systemd service, thus removing the init service file does not disable it anymore: https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L318-L320
`dietpi-services mask ntp` works.
Other DietPi ntpd commands still work well, nothing to change here so far.

Fourdee, 2018-01-25:
@MichaIng 

I've set low priority label for info (We officially support stretch), however, don't change what you like to work on, or by how much 👍

MichaIng, 2018-01-25:
@Fourdee 
Jep, agree. Just fixed the issues I observed, as they were not too hard and I want to test/go with DietPi Buster on my RPi after v6.0 release, at least give it a try 😄.

MichaIng, 2018-02-20:
🈴 General info: On Debian Buster repo there are many many packages updates so far, and somehow the mirroring to Raspbian Buster repo somehow is not done well (see missing `libsodium23` above). At the moment there is some perl update, which would due to broken dependencies remove Apache and MariaDB (and for sure many other titels):
```
2018-02-20 00:26:39 root@micha:~# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  apache2-data apache2-utils augeas-lenses certbot dh-python galera-3 libaio1 libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap libaugeas0 libbrotli1 libgdbm3 liblua5.2-0 libmpdec2 libperl5.26 libpython3-stdlib
  libpython3.6-minimal libpython3.6-stdlib libreadline5 libwrap0 lsof mariadb-client-core-10.1 mariadb-common mariadb-server-core-10.1 perl-modules-5.26 python3 python3-acme python3-asn1crypto python3-augeas python3-certbot
  python3-certifi python3-cffi-backend python3-chardet python3-configargparse python3-configobj python3-cryptography python3-distutils python3-future python3-idna python3-josepy python3-lib2to3 python3-minimal python3-mock
  python3-openssl python3-parsedatetime python3-pbr python3-pkg-resources python3-requests python3-rfc3339 python3-six python3-tz python3-urllib3 python3-zope.component python3-zope.event python3-zope.hookable python3-zope.interface
  python3.6 python3.6-minimal socat
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  apache2 apache2-bin libapache2-mod-php libapache2-mod-php7.2 libconfig-inifiles-perl libdbi-perl mariadb-client-10.1 mariadb-server mariadb-server-10.1 mysqltuner perl python-certbot-apache python3-certbot-apache
The following packages will be upgraded:
  perl-base
1 upgraded, 0 newly installed, 13 to remove and 0 not upgraded.
Need to get 1,241 kB of archives.
After this operation, 123 MB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
2018-02-20 14:53:43 root@micha:~# apt list --upgradable
Listing... Done
perl/testing 5.26.1-4+b1 armhf [upgradable from: 5.26.1-4]
perl-base/testing 5.26.1-4+b1 armhf [upgradable from: 5.26.1-4]
```
As `dietpi-software` forces `apt-get dist-upgrade` on installation, it would just force this removal.

On Debian I didn't face such problems so far, it seems to be the transfer from Debian repo to Raspbian repo that is not done too carefully by times.

MichaIng, 2018-02-27:
Further Buster integration of PHP: https://github.com/Fourdee/DietPi/pull/1562

MichaIng, 2018-02-28:
🈯️ perl upgrade issue resolved, finally the necessary library is there:
13 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
2018-02-28 23:37:38 root@micha:/var/log# apt list --upgradable
Listing... Done
...
libperl5.26/testing 5.26.1-4+b1 armhf [upgradable from: 5.26.1-4]
...
```

MichaIng, 2018-03-29:
🈴 Deluge is not available in Debian Buster repo, thus fails to install: https://packages.debian.org/stretch/deluge
🈯️ RPi: But it is available on Raspbian Buster repo 🤔.

Possible would be installation from source, but I will not touch this until Buster is moved into stable branch.
__________________
🈴 timedatectl output changed, thus run_ntpd status fails:
```
root@VM-Buster:~# timedatectl
                      Local time: Thu 2018-03-29 21:37:32 CEST
                  Universal time: Thu 2018-03-29 19:37:32 UTC
                        RTC time: Thu 2018-03-29 19:37:33
                       Time zone: Europe/Berlin (CEST, +0200)
       System clock synchronized: yes
systemd-timesyncd.service active: yes
                 RTC in local TZ: no
```

Fourdee, 2018-08-28:
@MichaIng 

👍 Site was updated few days ago to show optional Buster experimental images. 

Very stable. Iam yet to find an issue. 💯

MichaIng, 2018-10-12:
Next dangerous Raspbian repo bug:
```
2018-10-13 00:14:03 root@micha:/var/log# APT
Hit:1 https://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
4 packages can be upgraded. Run 'apt list --upgradable' to see them.
Listing... Done
libaprutil1-dbd-sqlite3/testing 1.6.1-3+b1 armhf [upgradable from: 1.6.1-3]
libaprutil1-ldap/testing 1.6.1-3+b1 armhf [upgradable from: 1.6.1-3]
perl-base/testing 5.26.2-7+b1 armhf [upgradable from: 5.26.2-7]
perl/testing 5.26.2-7+b1 armhf [upgradable from: 5.26.2-7]
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  apache2-data apache2-utils augeas-lenses certbot galera-3 libaio1 libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap libaugeas0 libbrotli1 libgdbm-compat4 libgdbm5 libjansson4 libjemalloc1 liblua5.2-0 libperl5.26
  libpython3-stdlib libreadline5 libwrap0 lsof mariadb-client-core-10.1 mariadb-common mariadb-server-core-10.1 perl-modules-5.26 python3 python3-acme python3-asn1crypto python3-augeas python3-certbot python3-certifi
  python3-cffi-backend python3-chardet python3-configargparse python3-configobj python3-cryptography python3-distutils python3-future python3-idna python3-josepy python3-lib2to3 python3-minimal python3-mock python3-openssl
  python3-parsedatetime python3-pbr python3-pkg-resources python3-requests python3-requests-toolbelt python3-rfc3339 python3-setuptools python3-six python3-tz python3-urllib3 python3-zope.component python3-zope.event
  python3-zope.hookable python3-zope.interface python3.6 python3.6-minimal socat
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  apache2 apache2-bin libapache2-mod-php libapache2-mod-php7.2 libconfig-inifiles-perl libdbi-perl mariadb-client-10.1 mariadb-server mariadb-server-10.1 perl python-certbot-apache python3-certbot-apache
The following packages will be upgraded:
  perl-base
1 upgraded, 0 newly installed, 12 to remove and 0 not upgraded.
Need to get 1,251 kB of archives.
After this operation, 105 MB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
2018-10-13 00:14:25 root@micha:/var/log# apt install perl
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 perl : Depends: libperl5.26 (= 5.26.2-7+b1) but 5.26.2-7 is to be installed
E: Unable to correct problems, you have held broken packages.
2018-10-13 00:14:38 root@micha:/var/log# apt install perl-base
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  apache2-data apache2-utils augeas-lenses certbot galera-3 libaio1 libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap libaugeas0 libbrotli1 libgdbm-compat4 libgdbm5 libjansson4 libjemalloc1 liblua5.2-0 libperl5.26
  libpython3-stdlib libreadline5 libwrap0 lsof mariadb-client-core-10.1 mariadb-common mariadb-server-core-10.1 perl-modules-5.26 python3 python3-acme python3-asn1crypto python3-augeas python3-certbot python3-certifi
  python3-cffi-backend python3-chardet python3-configargparse python3-configobj python3-cryptography python3-distutils python3-future python3-idna python3-josepy python3-lib2to3 python3-minimal python3-mock python3-openssl
  python3-parsedatetime python3-pbr python3-pkg-resources python3-requests python3-requests-toolbelt python3-rfc3339 python3-setuptools python3-six python3-tz python3-urllib3 python3-zope.component python3-zope.event
  python3-zope.hookable python3-zope.interface python3.6 python3.6-minimal socat
Use 'apt autoremove' to remove them.
Suggested packages:
  perl
The following packages will be REMOVED:
  apache2 apache2-bin libapache2-mod-php libapache2-mod-php7.2 libconfig-inifiles-perl libdbi-perl mariadb-client-10.1 mariadb-server mariadb-server-10.1 perl python-certbot-apache python3-certbot-apache
The following packages will be upgraded:
  perl-base
1 upgraded, 0 newly installed, 12 to remove and 2 not upgraded.
Need to get 1,251 kB of archives.
After this operation, 105 MB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
2018-10-13 00:14:53 root@micha:/var/log# apt install libaprutil1-ldap
[... 6 lines cut ...]
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libaprutil1-ldap : Depends: libaprutil1 (= 1.6.1-3+b1) but 1.6.1-3 is to be installed
E: Unable to correct problems, you have held broken packages.
2018-10-13 00:15:05 root@micha:/var/log# apt install libaprutil1-dbd-sqlite3
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
[... cut ...]
- `perl` update, but required `libperl` update is missing
- `perl-base` update, but dependency update of it's dependants missing => all `perl-base` dependants will be removed on unattended apt-get dist-upgrade
- `libaprutil1` child package updates, but `libaprutil1` update itself forgotten...

Again, never faced something similar on Debian, this is a Raspbian-only issue that regularly occurs with different packages by times. Quite dangerous, if one accidentally hits yes or, as said, runs the update unattended with `-y` option.

Fourdee, 2018-10-13:
@MichaIng 

From past experience, with RPi, wait for their official Buster image before we support it. They have their own repo, which they must iron out.

MichaIng, 2018-10-16:
@Fourdee 
> wait for their official Buster image before we support it.

Yes in every case. I just like being on the edge (should go with Arch Linux, lol) and would like to have Buster support fully integrated into DietPi, before it's officially released to stable. So we can do concurrent release at best 😃.

For RPi official Buster images we anyway need to wait longer, until also https://archive.raspberrypi.org has Buster branch. If I remember right, this took again some time after Debian Buster release.
____________
So basically those ongoing Raspbian repo dependency issues are of course non-critical, since it's officially testing, but it makes life for testers unnecessary (?) hard for my impression. But I am complaining here all the time, now it's time to address this to the repo maintainers, being constructive 😄.

MichaIng, 2018-12-19:
VirtualBox: TTY1 terminal resolution always resets to ~800x600 or something like this, after TTY is taken over from boot/kernel.
When commenting the `GRUB_GFXPAYLOAD_LINUX=keep` setting, then the resolution goes down to VBox default 640xsomething in between, so this is still correct/works. But when TTY1 service takes over the terminal, there seem to be an additional setting somewhere that applies 800x600. This is done regardless of grub settings, so has nothing to do with grub at all.. So far couldn't find something about it.

MichaIng, 2018-12-19:
Last update wave on Debian+Raspbian Buster, including systemd + many libraries, breaks Dropbear. Once logged out, connecting via SSH fails. Restart of Dropbear service required.

Damn, I logged out myself from my RPi with this 🤣.

MichaIng, 2018-12-24:
🈴 Current ownCloud 10.0.10 does not support PHP7.3 => Disabled on Buster for now: https://github.com/Fourdee/DietPi/pull/2361/commits/a454034c0060f815e88deee8048bb89404b43ac2

MichaIng, 2019-01-15:
Debian Buster release soon. Feature/Package freeze schedule has started: https://release.debian.org/testing/freeze_policy.html
> 2019-01-12 - Transition freeze
> Starting 2019-01-12, new transitions and large/disruptive changes are no longer acceptable for buster.
> 2019-02-12 - Soft-freeze
> 2019-03-12 - Full-freeze

With v6.21 I will put some effort in testing remaining software titles on Buster. Especially some 3rd party repos don't have a Buster branch yet, so we need to force Stretch there.

Raspbian Buster release will be later and according to the many above critical repo issues, this is required.

MichaIng, 2019-02-08:
Damn, the soft freeze did not reduce the amount of APT package updates, but increased it, in some cases with critical major version updates, e.g. mariadb v10.1 => v10.3.

One single day and 43 updated packages (and I have ONLY installed Nextcloud), including systemd, udev, APT itself, pathon, PHP...
```
43 packages can be upgraded. Run 'apt list --upgradable' to see them.
Listing... Done
apt-transport-https/testing 1.8.0~rc2 all [upgradable from: 1.8.0~beta1]
apt-utils/testing 1.8.0~rc2 armhf [upgradable from: 1.8.0~beta1]
apt/testing 1.8.0~rc2 armhf [upgradable from: 1.8.0~beta1]
ca-certificates/testing 20190110 all [upgradable from: 20170717]
libapache2-mod-php7.3/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
libapt-inst2.0/testing 1.8.0~rc2 armhf [upgradable from: 1.8.0~beta1]
libapt-pkg5.0/testing 1.8.0~rc2 armhf [upgradable from: 1.8.0~beta1]
libargon2-1/testing 0~20171227-0.2 armhf [upgradable from: 0~20171227-0.1]
libdb5.3/testing 5.3.28+dfsg1-0.3 armhf [upgradable from: 5.3.28+dfsg1-0.2]
libgcrypt20/testing 1.8.4-5 armhf [upgradable from: 1.8.4-4]
libgnutls30/testing 3.6.6-2 armhf [upgradable from: 3.6.5-2+rpi1]
libhogweed4/testing 3.4.1-1 armhf [upgradable from: 3.4.1~rc1-1]
libjson-c3/testing 0.12.1+ds-2 armhf [upgradable from: 0.12.1-1.3]
liblzma5/testing 5.2.4-1 armhf [upgradable from: 5.2.2-1.3]
libmpfr6/testing 4.0.2-1 armhf [upgradable from: 4.0.2~rc1-1]
libnettle6/testing 3.4.1-1 armhf [upgradable from: 3.4.1~rc1-1]
libnghttp2-14/testing 1.36.0-2 armhf [upgradable from: 1.36.0-1]
libparted2/testing 3.2-24 armhf [upgradable from: 3.2-23]
libpcre2-8-0/testing 10.32-4 armhf [upgradable from: 10.32-3]
libpng16-16/testing 1.6.36-5 armhf [upgradable from: 1.6.36-2]
libpopt0/testing 1.16-12 armhf [upgradable from: 1.16-11]
libslang2/testing 2.3.2-2 armhf [upgradable from: 2.3.2-1+b1]
libsystemd0/testing 240-5+rpi1 armhf [upgradable from: 240-4+rpi1]
libudev1/testing 240-5+rpi1 armhf [upgradable from: 240-4+rpi1]
parted/testing 3.2-24 armhf [upgradable from: 3.2-23]
php7.3-cli/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-common/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-curl/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-gd/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-intl/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-json/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-mbstring/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-mysql/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-opcache/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-readline/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-xml/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
php7.3-zip/testing 7.3.1-3 armhf [upgradable from: 7.3.1-1]
python3-pkg-resources/testing 40.7.1-1 all [upgradable from: 40.6.3-1]
python3-setuptools/testing 40.7.1-1 all [upgradable from: 40.6.3-1]
systemd-sysv/testing 240-5+rpi1 armhf [upgradable from: 240-4+rpi1]
systemd/testing 240-5+rpi1 armhf [upgradable from: 240-4+rpi1]
udev/testing 240-5+rpi1 armhf [upgradable from: 240-4+rpi1]
xz-utils/testing 5.2.4-1 armhf [upgradable from: 5.2.2-1.3]
```

To all who run DietPi on Buster:
- Carefully watch the update process, better run `apt dist-upgrade` more often then less, to not have hundreds of packages upgraded in a single run.
- Report any issues you face, e.g.: https://github.com/Fourdee/DietPi/pull/2490
- We will assist to get those fixed quickly.

MichaIng, 2019-02-15:
Yey, archive.raspberrypi.org repo already has a Buster branch: https://archive.raspberrypi.org/debian/dists/
It simply does not provide `raspi-copies-and-fills` anymore, which was incompatible with Buster udev version (see OT).

Besides that: No kernel or other package update when switching from the Stretch repo currently.
Will remove the s/buster/stretch/g sed when applying the repo and patch on v6.22 für RPi Buster systems (although I doubt there are much or any, besides mine 😄.

MichaIng, 2019-02-20:
🈴 RPi `i-sable-2km` module build might fail due to `gcc` version mismatch: https://github.com/Fourdee/DietPi/issues/2547#issuecomment-463831296

MichaIng, 2019-06-16:
Okay I mark this issue as closed now. We will gradually roll out Buster images for additional devices, at first as experimental/beta images, after official Debian (respectively Raspbian) Buster release and some user feedback, as new official image.

There are still some issues opened but those can be handled individually: https://github.com/MichaIng/DietPi/issues?q=is%3Aopen+is%3Aissue+label%3A"Debian+Buster"

closed 2019-06-16 by MichaIng